rockchip-linux/mpp #5: Hardware based scaler

sat8, 2017-03-27:
Hi

I wonder if a hardware based scaler is available in mpp. My use case is to decode a HD stream and then run some analysis on down scaled raw frame. I see that there is an abandoned ffmpeg repo here. Did it use hardware resizer in any way? 

Thnx

hizukiayaka, 2017-04-01:
Only JPEG supports it now.

closed 2017-04-01 by hizukiayaka

HermanChen, 2017-04-04:
About this question we plan to add post-process module into mpp. It will come later this year.

sat8, 2017-04-04:
thnx for the update.

zillevdr, 2017-05-04:
> About this question we plan to add post-process module into mpp. It will come later this year.__

Include this an deinterlacer?

WainDing, 2017-05-04:
yes， post-process abort deinterlace will come later.

Adam-Langley, 2018-12-14:
Hi,

How is porting the hardware scaler to the HD/H264 decoder coming along?

Thanks,