rockchip-linux/mpp #14: Getting assert when using DRM buffers

mcerveny, 2017-03-31:
If DRM buffer type is used the program getting asserts on ION buffer service.
https://github.com/rockchip-linux/mpp/blob/develop/mpp/base/mpp_buffer_impl.cpp#L641
The code should return for **all other** types than MPP_BUFFER_TYPE_ION.

hizukiayaka, 2017-04-01:
It is a known bug, that you must make fake ION buffer as I did in for_linux branch.

HermanChen, 2017-04-04:
We can trying implement an auto detect method to fix it while keeping ION/DRM type binding interface.

mcerveny, 2017-04-04:
Please test the solution with "Mode 3: Pure external mode" and "Commit flow group" - see http://tinkerboarding.co.uk/forum/showthread.php?tid=51
Thanks, Martin
[mpi_dec_test.zip](https://github.com/rockchip-linux/mpp/files/892428/mpi_dec_test.zip)

hizukiayaka, 2017-04-06:
The mode 3 is developing for Gstreamer rockchip.

closed 2017-06-01 by hizukiayaka