rockchip-linux/mpp #16: hal_h264d_api: hal_h264d_init hard mode error, value=0

zillevdr, 2017-04-05:
Hello,

todays git branch for_linux got an error:
[h264_rkmpp @ 0xa96025d0] Initializing RKMPP decoder.
mpi: mpp version: 0e51a98 author: Randy Li [build]: update the android build rules
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:118
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:165

I can't find why and need help. 

cat /proc/device-tree/compatible
rockchip,rk3288-miniarmrockchip,rk3288

ls /dev/*service
/dev/hevc-service  /dev/vpu-service

Regards zillevdr

hizukiayaka, 2017-04-06:
Which distribution do you use? I may offer you a binary version.

zillevdr, 2017-04-06:
I use Gentoo. There is a binary version difficult. Need cmake or make a special option?

hizukiayaka, 2017-04-06:
Could you fetch the latest and rebuild once? Maybe you are using the broken code I am testing in the morning.

HermanChen, 2017-04-06:
Looks like a platform detection error?

hizukiayaka, 2017-04-06:
Yes, with this environment variable mpp_debug=0x10 and try again.

HermanChen, 2017-04-06:
Please print the log of:
strings /proc/device-tree/compatible

zillevdr, 2017-04-06:
/proc/device-tree/compatible are in my first comment. mpp_debug I will test in the evening (Europe time).
I have also tested the repository from LongChair at https://github.com/LongChair/mpp/tree/for_linux. The same result.

zillevdr, 2017-04-07:
It was my fail. -DRKPLATFORM=ON was missed. Thank you for your help. This can be closed.

closed 2017-04-07 by zillevdr