rockchip-linux/mpp #17: decoded frames are in wrong order

zillevdr, 2017-04-12:
Hello,
the decoded frames are in wrong order. PresentationTimeStamp are ordered but the frames not. The picture on the screen jumps forward and backward. The packet from stream (decoder in) are not in the order for presentation. There is an [tutorial](http://dranger.com/ffmpeg/tutorial05.html) that explain that.
This are the pts in and out decoder:

> CodecVideoDecode: pts decoder in 1476218481 pts decoder out 1476186081
CodecVideoDecode: pts decoder in 1476211281 pts decoder out 1476187881
CodecVideoDecode: pts decoder in 1476205881 pts decoder out 1476189681
CodecVideoDecode: pts decoder in 1476207681 pts decoder out 1476191481
CodecVideoDecode: pts decoder in 1476209481 pts decoder out 1476193281
CodecVideoDecode: pts decoder in 1476213081 pts decoder out 1476195081
CodecVideoDecode: pts decoder in 1476214881 pts decoder out 1476196881
CodecVideoDecode: pts decoder in 1476216681 pts decoder out 1476198681
CodecVideoDecode: pts decoder in 1476232881 pts decoder out 1476200481
CodecVideoDecode: pts decoder in 1476225681 pts decoder out 1476202281
CodecVideoDecode: pts decoder in 1476220281 pts decoder out 1476204081
CodecVideoDecode: pts decoder in 1476222081 pts decoder out 1476205881

The frames should be sorted by pts not sorted the pts to the frames in incoming order. Or leave the pts to the decoded frame. Then can sorting in user space.

Best regards zillevdr

timkingh, 2017-04-13:
Hi, the decoded frames are in display order, and sorting can be done in MPP only, but not user space.

closed 2017-04-13 by hizukiayaka

zillevdr, 2017-04-13:
Why jumps the picture forward and backward? It looks like the decoded frames are in wrong order.

XiaoleoDing, 2017-04-13:
hi, zillevdr
 As you said, "PresentationTimeStamp are ordered but the frames not. The picture on the screen jumps forward and backward." Can you give me the bitstream for my analysis?

zillevdr, 2017-04-13:
Hi XiaoleoDing,
yes, when I come home. It's a DVB-S TV stream H264 decoded.

zillevdr, 2017-04-13:
Hi XiaoleoDing,
you can download the stream from there
https://c.gmx.net/@324833219872561205/NfuYVQAdRDG9ZmpmUeCIYA
Regards zillevdr

hizukiayaka, 2017-04-14:
I have downloaded your video sample, it works fine with Gstreamer rockchip at RK3288 ASUS Tinker.

hizukiayaka, 2017-04-14:
Also there are some error in your stream, it usually happens when you are using a not well designed parser when you are packing your stream.
To handle this kind of issue, please reference to the implementation of Gstreamer Rockchip.
https://github.com/rockchip-linux/gstreamer-rockchip/blob/develop/gst/rockchipmpp/gstmppdecbufferpool.c#L244

XiaoleoDing, 2017-04-14:
hi， zillevdr
    As hizukiayaka said, when mpp_frame_get_errorinfo(frame) !=0, current frame output must drop and not to dispaly. The reason is that, a fault tolerance mechanism is adopted, when find poc is less than the first I frame. In some demand, we need to show immediately, can not to wait. So, dispaly the first I frame immediately and mark poc less than the first I frame error.
like that:
	Line 1214: 01-02 08:16:43.710 I/vpu_api_legacy( 3818): get one frame pts 88544427566, fd 0x30, poc 2, errinfo 0, discard 0, eos 0, verr 0
	Line 1278: 01-02 08:16:43.710 I/vpu_api_legacy( 3818): get one frame pts 88544367566, fd 0x31, poc -4, errinfo 1, discard 0, eos 0, verr 1
	Line 1324: 01-02 08:16:43.730 I/vpu_api_legacy( 3818): get one frame pts 88544307566, fd 0x32, poc -10, errinfo 0, discard 1, eos 0, verr 1
	Line 1328: 01-02 08:16:43.730 I/vpu_api_legacy( 3818): get one frame pts 88544327566, fd 0x33, poc -8, errinfo 0, discard 1, eos 0, verr 1
	Line 1332: 01-02 08:16:43.730 I/vpu_api_legacy( 3818): get one frame pts 88544347566, fd 0x34, poc -6, errinfo 0, discard 1, eos 0, verr 1
	Line 1336: 01-02 08:16:43.730 I/vpu_api_legacy( 3818): get one frame pts 88544407566, fd 0x36, poc 0, errinfo 0, discard 1, eos 0, verr 1
	Line 1340: 01-02 08:16:43.730 I/vpu_api_legacy( 3818): get one frame pts 88544387566, fd 0x35, poc -2, errinfo 0, discard 1, eos 0, verr 1
	Line 1382: 01-02 08:16:43.770 I/vpu_api_legacy( 3818): get one frame pts 88544447566, fd 0x39, poc 4, errinfo 0, discard 0, eos 0, verr 0
	Line 1384: 01-02 08:16:43.770 I/vpu_api_legacy( 3818): get one frame pts 88544467566, fd 0x3a, poc 6, errinfo 0, discard 0, eos 0, verr 0
	Line 1386: 01-02 08:16:43.770 I/vpu_api_legacy( 3818): get one frame pts 88544487566, fd 0x38, poc 8, errinfo 0, discard 0, eos 0, verr 0
	Line 1392: 01-02 08:16:43.770 I/vpu_api_legacy( 3818): get one frame pts 88544507566, fd 0x3b, poc 10, errinfo 0, discard 0, eos 0, verr 0
	Line 1398: 01-02 08:16:43.770 I/vpu_api_legacy( 3818): get one frame pts 88544527566, fd 0x3c, poc 12, errinfo 0, discard 0, eos 0, verr 0
	Line 1400: 01-02 08:16:43.770 I/vpu_api_legacy( 3818): get one frame pts 88544547566, fd 0x3d, poc 14, errinfo 0, discard 0, eos 0, verr 0
	Line 1402: 01-02 08:16:43.780 I/vpu_api_legacy( 3818): get one frame pts 88544567566, fd 0x3e, poc 16, errinfo 0, discard 0, eos 0, verr 0
	Line 1422: 01-02 08:16:43.780 I/vpu_api_legacy( 3818): get one frame pts 88544587566, fd 0x37, poc 18, errinfo 0, discard 0, eos 0, verr 0

zillevdr, 2017-04-16:
Hello,
the issue was in my app. I had freed the buffer after page flip. The DRM driver need 10 ms to render the frame on screen. In this time had the decoder over written the frame. Sorry for the trouble.
Thank you for your help! Best regards 
zillevdr

hizukiayaka, 2017-04-17:
You should have a look on the Gstreamer rockchip to know how to release and reference a frame buffer correctly.