rockchip-linux/mpp #24: DRM allocator appears to use 8X more memory than it should

OtherCrashOverride, 2017-07-26:
The following line appears to incorrectly calculate the size:
https://github.com/rockchip-linux/mpp/blob/release/osal/allocator/allocator_drm.c#L189

```dmcb.size = dmcb.width * dmcb.bpp;```
should probably be:
```dmcb.size = dmcb.width * (dmcb.bpp / 8);```

bpp=bits per pixel. there are 8bits in a byte. size is in bytes.

hizukiayaka, 2017-08-01:
Thanks, it actually doesn't matter, since that field is filled by kernel, that line doesn't have any effect.

OtherCrashOverride, 2017-08-02:
Coincidentally, while debugging a project, I came to the same conclusion.  The line should be removed to since the field is an output and not an input to the IOCTL.

hizukiayaka, 2017-08-02:
So this line is just an useless sentence, I will remove it in the next release version.

closed 2017-08-02 by hizukiayaka