rockchip-linux/mpp #25: RK3328 ioctl error

sueshieh, 2017-08-21:
When running mpi_enc_test h264 test with linux kernel release-4.4 (03d6a38), mpp_device_init() ioctl error. RK3328 MPP (unsigned long) / Kernel (u32) not match.

- MPP: mpp/hal/worker/mpp_device/mpp_device.c

#define VPU_IOC_MAGIC                       'l'
#define VPU_IOC_SET_CLIENT_TYPE             _IOW(VPU_IOC_MAGIC, 1, unsigned long)
#define VPU_IOC_GET_HW_FUSE_STATUS          _IOW(VPU_IOC_MAGIC, 2, unsigned long)
#define VPU_IOC_SET_REG                     _IOW(VPU_IOC_MAGIC, 3, unsigned long)
#define VPU_IOC_GET_REG                     _IOW(VPU_IOC_MAGIC, 4, unsigned long)
#define VPU_IOC_PROBE_IOMMU_STATUS          _IOR(VPU_IOC_MAGIC, 5, unsigned long)

- Kernel: drivers/video/rockchip/vpu/mpp_dev_common.h

#define MPP_IOC_MAGIC			'l'
#define MPP_IOC_SET_CLIENT_TYPE		_IOW(MPP_IOC_MAGIC, 1, u32)
#define MPP_IOC_GET_HW_FUSE_STATUS	_IOW(MPP_IOC_MAGIC, 2, u32)
#define MPP_IOC_SET_REG			_IOW(MPP_IOC_MAGIC, 3, u32)
#define MPP_IOC_GET_REG			_IOW(MPP_IOC_MAGIC, 4, u32)
#define MPP_IOC_PROBE_IOMMU_STATUS	_IOR(MPP_IOC_MAGIC, 5, u32)

hizukiayaka, 2017-08-29:
Confirm, will be fixed in the next release

closed 2017-09-06 by hizukiayaka

dalmatele, 2017-09-21:
@sueshieh Can you give me the example video? 
Because every input file I use, mpi_enc_test created unused file encode.

sueshieh, 2017-09-22:
@dalmatele I got kernel crash when I modified kernel driver drivers/video/rockchip/vpu/mpp_dev_common.h ioctl to "unsigned long". So I have no idea how to fix this issue.

If you don't change kernel driver ioctl definition, mpp ioctl just return error and do nothing. So the output encoded file is dummy zero.

sueshieh, 2017-09-22:
@hizukiayaka Did this fix commit to kernel source? Will it causes kernel crash? thanks.

dalmatele, 2017-09-22:
@sueshieh in my test, I saw that if I put the data in to the frame buffer, the packet I got from output always has zero value. But MPP does not output any error.
This is my kernel version: Linux rock64 4.4.70-rockchip-ayufan-85 #1 SMP Fri Aug 4 08:38:41 UTC 2017 aarch64 GNU/Linux
And I use RK3288 chip.

hizukiayaka, 2017-09-22:
@sueshieh 
I have submitted a patch for the kernel a few weeks ago, but it is still under review.

dalmatele, 2017-09-22:
@ sueshieh I've check ioctl function in mpp/hal/worker/mpp_device/mpp_device.c and see that the return is 0.
I think no error in this.
Maybe my problem is difference with you.

dalmatele, 2017-09-25:
And now I see this in kernel's log:
mpp_dev_ioctl:545: unknown mpp ioctl cmd 40086c03

dalmatele, 2017-09-26:
@sueshieh change unsigned log to __u32 will make kernel crash.

sueshieh, 2017-09-29:
Yes. After fix the ioctl number, kernel will crash.

dalmatele, 2017-10-08:
@hizukiayaka can you send me the patch to test?

dalmatele, 2017-11-01:
@sueshieh: I think I knew the error.
In 64 bit system, unsigned long will has size is 64 bit, but in kernel it used 32 bit size.
I've tried unsigned int, kernel has recognized ioctl command but it still has hang.

Climax777, 2018-05-10:
this is happening to me too on rock64

Andrewt12, 2018-06-05:
in mpp_device.c I changed to look like this

`#define VPU_IOC_SET_CLIENT_TYPE             _IOW(VPU_IOC_MAGIC, 1, uint32_t)
#define VPU_IOC_SET_REG                     _IOW(VPU_IOC_MAGIC, 3, uint64_t)
#define VPU_IOC_GET_REG                     _IOW(VPU_IOC_MAGIC, 4, uint64_t)

#define VPU_IOC_SET_CLIENT_TYPE_U32         _IOW(VPU_IOC_MAGIC, 1, uint32_t)
`
I no longer get ioctl error. Still doesn't work though, but I get a black screen instead of white. One step closer.

dalmatele, 2018-06-07:
@Andrewt12: I've changed these too but no luck because RK3328 has some bug with encode function.
I've talked with developers but they said that they didn't have time to fix.