rockchip-linux/mpp #28: video/rk_vpu_service.h: No such file or directory when build

dalmatele, 2017-10-17:
When we built mpp with new code (release branch), we had this error:
video/rk_vpu_service.h: No such file or directory

hizukiayaka, 2017-10-18:
Install the kernel header file.

closed 2017-10-18 by hizukiayaka

dalmatele, 2017-10-18:
@hizukiayaka but not header file with this name rk_vpu_service.h

hizukiayaka, 2017-10-18:
It looks the kernel is out of data. Ask the maintainer of the kernel to update.

dalmatele, 2017-10-18:
@hizukiayaka As I see, the new kernel (the last code in github: https://github.com/rockchip-linux/kernel.git does not has this header.
Or you use another source?

dbermond, 2017-11-09:
@dalmatele Could you find any solution for this? I cannot find the header `rk_vpu_service.h` anywhere.

dalmatele, 2017-11-11:
@dbermond No, I don't have any solution. I don't know why he said that I used old kernel then closed this issue.

dbermond, 2017-11-12:
@dalmatele Thank you for the reply. Then, we are stuck in this situation.

Kwiboo, 2017-11-12:
See https://github.com/rockchip-linux/kernel/blob/release-4.4/include/uapi/video/rk_vpu_service.h for the uapi header, was added in one of the recent kernel updates.

dbermond, 2017-11-13:
@Kwiboo Thank you for providing the information!

rockchipp-mpp 20171107 now builds fine for me when using this rockchip kernel header.

dalmatele, 2017-11-14:
@Kwiboo good support.
@dbermond have you test if mpp can encode/decode by hardware on RK3328 rockchip with new kernel.
In old kernel, It can not.

Kwiboo, 2017-11-14:
@dalmatele decoding on RK3328 works for me with old and new kernel (I have never tested encoding), you may need some patches: https://github.com/Kwiboo/mpp/commit/e34511ef9354282ff1edf87391a9ea1a3a321292 if you use 64-bit kernel and 32-bit userspace and also the two patches added in https://github.com/LibreELEC/LibreELEC.tv/commit/61ad185c74ed719dc82656073ba510aa5cc45020 to fix dma issues

dalmatele, 2017-11-14:
@Kwiboo yes, it has problem with encode, maybe wrong ioctl command. But I'll check this new kernel to see if it works.

dalmatele, 2017-11-20:
I think we have some wrong with ioctl command.
You can see this issue: https://github.com/rockchip-linux/mpp/issues/30
When I've debuged, I've see that client sent 40086c03 command, but kernel used 40046c03 command.
It's maybe we use 64bit chip, right? Although I've used the lastest kernel.

dalmatele, 2018-01-07:
@Kwiboo can you help me testing encode process?
Until now, I've not encoded on RK3328: unknow mpp ioctl cmd 40086c03 and 40086c04