rockchip-linux/mpp #32: Decoding video with mpp, and showing NOT in fullscreen

nikitos1550, 2017-11-26:
Hello!

I have rk3328 system, I built all related sources from rockchip-linux repos, rockchip kernel. 
I can decode video via gst-launch-1.0 util using kmssink (my system is without X/Wayland).
But video is overlaying in fullscreen mode.

What is proper way to scale video to some smaller size and show on the part of screen?
Also I built rk-qt-video repo and testing player example, that somehow suppose to show video as I want.
But it is not working properly.

```
# ./player /root/test2.mp4
mpi: mpp version: Without VCS, under bleeding
QMetaObject::indexOfSignal: signal setFullScreen(bool) from QVideoWidget redefined in VideoWidget
mpp_buffer: mpp_group 0x7f68009a40 tag unknown caller gst_mpp_allocator_drm_buf mode external type ion deinit with 4177920 bytes not released
mpp_buffer:
dumping buffer group 0x7f68009a40 id 3
mpp_buffer: mode external
mpp_buffer: type ion
mpp_buffer: limit size 0 count 0
mpp_buffer: used buffer count 1
mpp_buffer: buffer 0x7f6800c460 fd   74 size    4177920 ref_count   1 discard 0 caller gst_mpp_allocator_drm_buf
mpp_buffer: unused buffer count 0
[ 2109.317484] rk-vcodec ff360000.rkvdec: closed
Error: "Internal data stream error."
QMetaObject::indexOfSignal: signal setFullScreen(bool) from QVideoWidget redefined in VideoWidget
mpi: mpp version: Without VCS, under bleeding
QMetaObject::indexOfSignal: signal setFullScreen(bool) from QVideoWidget redefined in VideoWidget
[ 2114.159702] rockchip-pm-domain ff100000.syscon:power-controller: failed to get ack on domain 'pd_video', val=0x10040
^C[ 2116.173425] rk-vcodec ff360000.rkvdec: closed
# ./player /root/test.mp4
mpi: mpp version: Without VCS, under bleeding
QMetaObject::indexOfSignal: signal setFullScreen(bool) from QVideoWidget redefined in VideoWidget
Warning: "A lot of buffers are being dropped."
Warning: "A lot of buffers are being dropped."
[ 2138.896773] rk-vcodec ff360000.rkvdec: closed
QMetaObject::indexOfSignal: signal setFullScreen(bool) from QVideoWidget redefined in VideoWidget
[ 2142.783737] rockchip-pm-domain ff100000.syscon:power-controller: failed to get ack on domain 'pd_video', val=0x10040
```

Video is not playing, I can see only first frame in all combinations FullScreen EGL/DRM or not FullScreen EGL/DRM modes.

nikitos1550, 2017-11-26:
I also built Xserver with armsoc, and QT5 with xcb support. Player demo now working, but video can be show only in fullscreen mode, if not fullscreen that black square on video place. Sametime if I will start gst-launch with rkximagesink it can be resized as window and moved (very slow).

hizukiayaka, 2017-11-28:
Ask rk-qt-video, not here.

closed 2017-11-28 by hizukiayaka