rockchip-linux/mpp #40: Mpi_test failed

Antonavt, 2018-04-05:
Hello,

I have RK3399
I built mpp using by commands:
> mpp/build/linux/aarch64/make-Makefiles.bash
make
sudo make install`
and try to start build/linux/aarch64/test/mpi_test

I receive output:

> mpi_test: mpi_test start
mpi_test: mpi_test decoder test start
mpi: mpp version: b186d04 author: Herman Chen [cmake]: Change to shorter RKPLATFORM macro
mpp: unable to create unsupported type 0 coding 0
mpi: mpp_init leave ret -1
mpi_test: mpp_init failed
mpi: mpp_destroy enter ctx 0x2b2d2140
mpi: mpp_destroy leave ret 0
mpi_test: mpi_test failed

If I'm not mistaken it means that something works incorrect. How can i find problem of this error?
What are the reasons for this behavior?

hizukiayaka, 2018-04-06:
read the help!

closed 2018-04-06 by hizukiayaka

Antonavt, 2018-04-09:
Which part of help do i have to read for understanding "Why mpi_test is failed"? Could you give a link?

noah538, 2018-09-10:
I had the same problem. I thought is was mpp_init, so i changed MPP_VIDEO_CodingUnused => to MPP_VIDEO_CodingAVC, but it will still fail at mpi_control() afterward. would you please share the the fixes thanks.

rageworx, 2019-02-13:
Me same error, so what help need to be read ?
something kind of rude ?