rockchip-linux/mpp #41: Can't build MPP on Tinkerboard (Armbian OS)

JyuHo, 2018-05-10:
Since the last additions on the git, it is impossible for me to compile mpp and I can not understand the error.

My operating system is Armbian Debian Stretch Kernel 4.4.120-rockchip.

Here is the error:

>Scanning dependencies of target vproc_rga
[ 75%] Building CXX object mpp/vproc/rga/CMakeFiles/vproc_rga.dir/rga.cpp.o
/home/tinker/mpp/mpp/vproc/rga/rga.cpp: In function ‘MPP_RET rga_deinit(RgaCtx)’:
/home/tinker/mpp/mpp/vproc/rga/rga.cpp:131:27: error: ‘close’ was not declared in this scope
         close(impl->rga_fd);
                           ^
mpp/vproc/rga/CMakeFiles/vproc_rga.dir/build.make:62: recipe for target 'mpp/vproc/rga/CMakeFiles/vproc_rga.dir/rga.cpp.o' failed
make[2]: *** [mpp/vproc/rga/CMakeFiles/vproc_rga.dir/rga.cpp.o] Error 1
CMakeFiles/Makefile2:1997: recipe for target 'mpp/vproc/rga/CMakeFiles/vproc_rga.dir/all' failed
make[1]: *** [mpp/vproc/rga/CMakeFiles/vproc_rga.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

AngryManMLS, 2018-05-10:
Getting the same issue as you only at about 53% in.

JyuHo, 2018-05-10:
I tried with a kernel mainline fair by Myy with the integration of the VPU / RGA and it does not change anything

JyuHo, 2018-05-10:
To be able to do the compilation I'm going back on a previous commits like this:

	$ git clone https://github.com/rockchip-linux/mpp.git
	$ cd mpp
        $ git checkout -b test-branch fb73f23f53
	$ cmake -DRKPLATFORM=ON -DHAVE_DRM=ON
	$ make -j4
	$ sudo make install

AngryManMLS, 2018-05-10:
Worked for me JyuHo.

HermanChen, 2018-05-11:
The latest commit e855cb171a2 fix this issue. Thanks for report.

closed 2018-05-21 by HermanChen