rockchip-linux/mpp #42: mpph264enc in Kernel 3.0.36 in RK3188 problem.

chandong83, 2018-05-19:
Hello,

I am porting MPP Framework in linux kernel 3.0.36.
and my AP is RK 3188.
I know MPP does not support linux kernel 3.0.36.
but, I made some changes and it works a little bit.

when i encode H.264, I see an unknown lines in the video file.
please check the link below.
[screenshot](https://github.com/chandong83/rk/blob/master/img/scr1.png)

and, when the image is not move, it is fine.
[screenshot](https://github.com/chandong83/rk/blob/master/img/3.png)

The link below is actually encoded video.
[movie](https://youtu.be/QAgZbYGFgTs)

I don't know where to look in this case.

Please give me a hint.

Thank you.

chandong83, 2018-05-21:
Thank you for your reply.
I will check that.

My work is as follows:

I modified the memory addressing approach to physical addressing with reference to libvpu of Walle-hardware source code.
(I added ION_CUSTOM_GET_PHYS ioctl Command)

And I created a function called mpp_buffer_get_phy that gets the physical address.

And I modified it to mpp_buffer_get_phy instead of mpp_buffer_get_fd.

And I Tested with mpi_enc_test.

Is this correct?

Thank you.

HermanChen, 2018-05-21:
Yes, that is correct. Nice work.
But I suggest to convert fd to phycial address in kernel for better compatibility.

The mpi_enc_test use CPU to write the buffer. If the ion buffer is cachable then it will have cache issue.
Do you use CPU to copy from camera buffer to MppBuffer?
It is better to send buffer from camera directly to mpp with fd or phycial address. Then hardware can accept the phycial address directly. This may remove the flashing line.

chandong83, 2018-05-21:
Thank you so much.

I have a one more question.
do you mean "mpi_enc_test use CPU to write the buffer", "it used fread in read_yuv_image function"?
right?
so if I encode a sequence of YUV files, do I force it to clear The Cache of Buffer.(like MppBuffer)?

I mean, How do I handle the cache when I encode a file?

Thank you.

HermanChen, 2018-05-21:
Yes, Not only the "fread in read_yuv_image" it is read from file and write to DDR memory. The "writing to DDR memory" is operated by CPU.

The old version ion has a flush/invalidate/clean function. Use "flush" or "clean" can handle the cache issue.

chandong83, 2018-05-21:
Thank you for your good teaching.

I will try what you have taught me.

Thank you.

HermanChen, 2018-05-21:
You are welcome : )

chandong83, 2018-05-22:
Hello @HermanChen 
Thank you so very much!! 
I solved this problem with your help.
you were right.
This was a output buffer cache issue.
I solved the problem by adding function to clear the output buffer cache.

Thank you.
(I've been through this problem for a month.)

The link below is a successful video.
[Successful videos](https://youtu.be/YL5TK19KKHg)

HermanChen, 2018-05-22:
Glad to hear this : )

closed 2018-05-22 by HermanChen