rockchip-linux/mpp #44: Please do that $#^T

Cyper77, 2018-06-01:
i think imx219 still not coded for tinkerboard :( very well thank you... we cant use openCV too.
did you think, implement this driver for tinkerboard gpu mpp (like rpi MMAL) ???
we can use mmal for accelerated camera image processing on Rpi !!!
https://github.com/allwinner-zh/linux-3.4-sunxi/blob/master/drivers/media/video/sunxi-vfe/device/imx219.c
WHY csi port is un usable????????????

HermanChen, 2018-06-02:
What is your question?
I do not think camera support or motion detection is related to mpp. Mpp is for video encoding / decoding so far. It has no relation to camera or gpu.

Cyper77, 2018-06-02:
ok mpp only video decode api for gpu, but openMAX(raspi MMAL) works with full gpu-csi-dsi at low level.
which one is better? any one can do that for better tinkerboard usage? even openmax we havent get a proper imx219 driver yet...

eddiecailinux, 2018-06-04:
here is the driver we use for tinker for your reference: https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/media/i2c/imx219.c

HermanChen, 2018-06-04:
mpp decode video by VPU(Video Process Unit) not GPU.
If OpenMax component can work at gpu-csi-dsi mode then it must be several components to work together. Mpp can also work as OpenMax component and compose such a component link.

OMX Implement with standalone component or with full decoding-display component have their own advantage and disadvantage.
Standalone decoding component is more flexible and less limitation while loss some latency. full decoding-display component has best latency performance but hard to make modification for user requirements.

Cyper77, 2018-06-04:
thanks, you talk about latency timing and performance a lot, but we havent got any tinkerboard mipi CSI camera API ?? with gstreamer very very slow and buggy. where is TinkerB latency performance for camera module? raspicam api has 120fps at low level C++ with MMAL. and WORKING. Where ASUS or Rockchip prestige??? peeeh. We bought a garbage :( i am so sad for that...

eddiecailinux, 2018-06-05:
we have already fix the camera issue with Asus. Please ask Asus for a update.

closed 2018-06-05 by HermanChen