rockchip-linux/mpp #46: are there functions in mpp  to  convert yuv  to RGB?

hwd8868, 2018-07-05:


WainDing, 2018-09-10:
No such function is currently available in mpp.

closed 2019-02-18 by HermanChen

codomania, 2019-02-20:
Any tips on how this can be done are greatly appreciated. I am working on a home project in which I will decode H.264 and compose the output with a separate ARGB buffer. Hence need to convert the YUV -> ARGB.

qvoid, 2019-02-25:
> Any tips on how this can be done are greatly appreciated. I am working on a home project in which I will decode H.264 and compose the output with a separate ARGB buffer. Hence need to convert the YUV -> ARGB.

You can get NV12 frames by calling *Mpp::get_frame* (check out *mpp.cpp*), and then convert these yuv frames  to frames in other color format. This may help you -- [stackoverflow](https://stackoverflow.com/questions/17892346/how-to-convert-rgb-yuv-rgb-both-ways)

codomania, 2019-02-25:
Yes, I understand that there is a need to convert NV12 -> ARGB but the question is mainly around, is there a hardware block which I use. I don't want to use SW based color conversion will it will not able to keep up with 1080/2K @ 30fps. I saw linux-rga project on rockchip, it seems to provide HW acceleration color conversion support but I am not able to get it going. I don't see it included in Ubuntu release I downloaded from the NanoPC website and also buildroot does not contain it. If you provide some sample example which I can use for color conversion using HW module then it will be great help. thanks

HermanChen, 2019-02-26:
There are simple informal rga test case in mpp:
https://github.com/rockchip-linux/mpp/blob/develop/mpp/vproc/rga/test/rga_test.cpp

yuxiatongzhi, 2019-07-12:
> There are simple informal rga test case in mpp:
> https://github.com/rockchip-linux/mpp/blob/develop/mpp/vproc/rga/test/rga_test.cpp

编译你上面链接的  test 可执行文件，在设备上执行
`./rga_test -i /sdcard/mpp/nv12.yuv -o /sdcard/mpp/yuv2rgb.rgb -w 1080 -h 1920 -f 0 -dst_w 1080 -dst_h 1920 -dst_fmt 65546`
[image]

[image]

`dump_mpp_frame_to_file`的输出为什么全是YUV？

HermanChen, 2019-07-15:
可以自动修改加上支持，然后给我们 patch 合并上