rockchip-linux/mpp #48: mpp can't finalize installation

JyuHo, 2018-08-06:
Hello,
I have this erros with mpp compilation on a Tinkerboard

Scanning dependencies of target rockchip_mpp_static
[ 71%] Building CXX object mpp/CMakeFiles/rockchip_mpp_static.dir/mpp_impl.cpp.o
[ 71%] Building CXX object mpp/CMakeFiles/rockchip_mpp_static.dir/mpp.cpp.o
[ 71%] Building CXX object mpp/CMakeFiles/rockchip_mpp_static.dir/mpp_info.cpp.o
[ 72%] Building CXX object mpp/CMakeFiles/rockchip_mpp_static.dir/mpi.cpp.o
/home/tinker/mpp/mpp/mpp_impl.cpp: In function ‘MPP_RET mpp_dump_init(MppDumpInf             o*, MppCtxType)’:
/home/tinker/mpp/mpp/mpp_impl.cpp:40:34: error: ‘memset’ was not declared in thi             s scope
     memset(info, 0, sizeof(*info));
                                  ^
mpp/CMakeFiles/rockchip_mpp_static.dir/build.make:110: recipe for target 'mpp/CM             akeFiles/rockchip_mpp_static.dir/mpp_impl.cpp.o' failed
make[2]: *** [mpp/CMakeFiles/rockchip_mpp_static.dir/mpp_impl.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:772: recipe for target 'mpp/CMakeFiles/rockchip_mpp_static.             dir/all' failed
make[1]: *** [mpp/CMakeFiles/rockchip_mpp_static.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

HermanChen, 2018-08-07:
Thanks for reporting : )
The following patch will fix the issue. [b81b3e5](https://github.com/rockchip-linux/mpp/commit/b81b3e52cdaf1fa8387c7fcadbd5b20c8de988c6)

JyuHo, 2018-08-08:
yep now work like a charm.

thx

closed 2018-08-09 by HermanChen