rockchip-linux/mpp #49: Ubuntu 18.04.1 LTS (GNU/Linux 4.4.132-1075-rockchip-ayufan-ga83beded8524 aarch64

Martossien, 2018-08-12:
My device is a SCISHION V88 Mini III ( RK3328 ) ( procedure  :  https://forum.armbian.com/topic/6658-rk3328-scishion-v88-piano-and-v88-mini-iii-tv-boxes/ ) so 4.4.132 Ubuntu 18.04.1 LTS
I have the vpu device
root@rock64:~# ls -l /dev/ | grep vpu
crw-rw-rw- 1 root   video   247,   0 Jul 26 09:08 vpu_service

With this version i have compiled last ffmpeg
When i launch ffmpeg with h264_rkmpp i have allways this error (all video file)

Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scaler_0'
Error reinitializing filters!﻿
Failed to inject frame into filter network: Function not implemented 

how i have compiled ::
sudo apt-get update -qq && sudo apt-get -y install autoconf automake build-essential cmake git-core libass-dev libfreetype6-dev libsdl2-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev pkg-config texinfo wget zlib1g-dev yasm libx264-dev libx265-dev libnuma-dev libvpx-dev libfdk-aac-dev libmp3lame-dev libopus-dev 
of course :
apt install librockchip-mpp-dev
compiling ffmpeg
cd ~/ffmpeg_sources && \
wget -O ffmpeg-snapshot.tar.bz2 https://ffmpeg.org/releases/ffmpeg-snapshot.tar.bz2 && \
tar xjvf ffmpeg-snapshot.tar.bz2 && \
cd ffmpeg && \
PATH="$HOME/bin:$PATH" PKG_CONFIG_PATH="$HOME/ffmpeg_build/lib/pkgconfig" ./configure \
  --prefix="$HOME/ffmpeg_build" \
  --pkg-config-flags="--static" \
  --extra-cflags="-I$HOME/ffmpeg_build/include" \
  --extra-ldflags="-L$HOME/ffmpeg_build/lib" \
  --extra-libs="-lpthread -lm" \
  --bindir="$HOME/bin" \
  --enable-hardcoded-tables --enable-gpl --enable-libaom --enable-libass --enable-libfdk-aac --enable-libfreetype --enable-libmp3lame --enable-libopus --enable-libvorbis --enable-libvpx --enable-libx264  --enable-libx265 \
  --enable-rkmpp \
  --enable-version3 \
  --enable-libdrm \
  --enable-nonfree && \
PATH="$HOME/bin:$PATH" make -j4 && \
make install && \

Martossien, 2018-08-12:
Source file ::
https://download.blender.org/durian/movies/Sintel.2010.720p.mkv
works with ::
ffmpeg -hwaccel drm -hwaccel_device /dev/dri/card0 -c:v h264_rkmpp -i Sintel_Trailer.720p.DivX_Plus_HD.mkv  -f null - -benchmark
don't work with ::
ffmpeg -hwaccel drm -hwaccel_device /dev/dri/card0 -c:v h264_rkmpp -i Sintel_Trailer.720p.DivX_Plus_HD.mkv /tmp/test.mkv
error:

[h264_rkmpp @ 0x55b3d47560] Decoder noticed an info change (1280x544), format=0
[h264_rkmpp @ 0x55b3d47560] Received a frame.
[graph 0 input from stream 0:0 @ 0x55b3dd7190] Setting 'video_size' to value '1280x544'
[graph 0 input from stream 0:0 @ 0x55b3dd7190] Setting 'pix_fmt' to value '181'
[graph 0 input from stream 0:0 @ 0x55b3dd7190] Setting 'time_base' to value '1/1000'
[graph 0 input from stream 0:0 @ 0x55b3dd7190] Setting 'pixel_aspect' to value '1/1'
[graph 0 input from stream 0:0 @ 0x55b3dd7190] Setting 'sws_param' to value 'flags=2'
[graph 0 input from stream 0:0 @ 0x55b3dd7190] Setting 'frame_rate' to value '24/1'
[graph 0 input from stream 0:0 @ 0x55b3dd7190] w:1280 h:544 pixfmt:drm_prime tb:1/1000 fr:24/1 sar:1/1 sws_param:flags=2
[format @ 0x55b3ea6c00] Setting 'pix_fmts' to value 'yuv420p|yuvj420p|yuv422p|yuvj422p|yuv444p|yuvj444p|nv12|nv16|nv21'
[auto_scaler_0 @ 0x55b3de8940] Setting 'flags' to value 'bicubic'
[auto_scaler_0 @ 0x55b3de8940] w:iw h:ih flags:'bicubic' interl:0
[format @ 0x55b3ea6c00] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_null_0' and the filter 'format'
Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scaler_0'
Error reinitializing filters!
Failed to inject frame into filter network: Function not implemented
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x55b3d3ba40] Statistics: 0 seeks, 0 writeouts
[libvorbis @ 0x55b3ea7f00] 39 frames left in the queue on closing
[AVIOContext @ 0x55b3d08120] Statistics: 66889 bytes read, 4 seeks

Martossien, 2018-08-12:
and works with -vcodec copy, so no transcode
ffmpeg -hwaccel drm -hwaccel_device /dev/dri/card0 -c:v h264_rkmpp -i Sintel_Trailer.720p.DivX_Plus_HD.mkv -vcodec copy /tmp/test.mkv

hizukiayaka, 2018-08-13:
The drm prime doesn't support software processing, you may contact LibreELEC and ffmpeg guys to add a hardware data path for libswscale.
Rockchip mpp has nothing to do here.

Martossien, 2018-08-13:
_you may contact LibreELEC_
LibreELEC ? i use UBUNTU not libreELEC
_ffmpeg guys to add a hardware data path for libswscale._
I am shure of this : if i open a ticket on ffmpeg it was clode quickly because ::
ffmpeg -c:v h264 -i Sintel_Trailer.720p.DivX_Plus_HD.mkv  /tmp/test.mkv WORKS
FFmpeg guy say to me , ask to the dev of hardware decoder rockchip mpp ( h264_rkmpp )

hizukiayaka, 2018-08-13:
It is not us to add the support for ffmpeg.

Martossien, 2018-08-13:
ok i open a ticket on ffmpeg support

closed 2023-08-03 by HermanChen