rockchip-linux/mpp #51: Building error (RK1108 SDK in Ubuntu 14.04)

oggcgg, 2018-08-29:
I was trying to build mpp for RK1108, my arm.linux.cross.cmake just like following
_cmake_minimum_required( VERSION 2.6.3 )                                                         
                                                                                                 
 SET(CMAKE_SYSTEM_NAME Linux)                                                                    
 SET(CMAKE_C_COMPILER "arm-linux-gcc")                                                           
 SET(CMAKE_CXX_COMPILER "arm-linux-g++")                                                         
 SET(CMAKE_SYSTEM_PROCESSOR "armv7-a")                                                           
                                                                                                                                                
 add_definitions(-fPIC)                                                                          
 add_definitions(-DARMLINUX)                                                                     
 add_definitions(-D__gnu_linux__)_ 

when make the source code I came across errors like below...Did I do something wrong?

[ 70%] Building CXX object mpp/CMakeFiles/rockchip_mpp.dir/mpi.cpp.o
Linking CXX shared library librockchip_mpp.so
arm-rkcvr-linux-uclibcgnueabihf/bin/ld: error: **librockchip_mpp.so.0 uses VFP register arguments, CMakeFiles/rockchip_mpp.dir/mpp_info.cpp.o does not**
arm-rkcvr-linux-uclibcgnueabihf/bin/ld: failed to merge target specific data of file CMakeFiles/rockchip_mpp.dir/mpp_info.cpp.o
arm-rkcvr-linux-uclibcgnueabihf/bin/ld: error: **librockchip_mpp.so.0 uses VFP register arguments, CMakeFiles/rockchip_mpp.dir/mpp.cpp.o does not**
arm-rkcvr-linux-uclibcgnueabihf/bin/ld: failed to merge target specific data of file CMakeFiles/rockchip_mpp.dir/mpp.cpp.o
                 :
                 :
                 :
arm-rkcvr-linux-uclibcgnueabihf/bin/ld: error: librockchip_mpp.so.0 uses VFP register arguments, vproc/iep/libvproc_iep.a(iep.cpp.o) does not
arm-rkcvr-linux-uclibcgnueabihf/bin/ld: failed to merge target specific data of file vproc/iep/libvproc_iep.a(iep.cpp.o)

oggcgg, 2018-08-30:
Thanks, Herman!
But I don't get it. All the CMake files were generated by "make-Makefiles.bash" and I did not change anything. Why will be conflict on old compile objects? Could you help to explain more?

HermanChen, 2018-08-30:
Cmake will generate all compile objects under the build directory.
Try this:
rm -rf CMakeFiles/ CMakeCache.txt cmake_install.cmake mpp/ osal/ test/ utils/ Makefile CTestTestfile.cmake
This will remove the compile files generated by Cmake.

oggcgg, 2018-08-30:
Herman, 
The reason why I came across these errors was I used the wrong flag, seems like rockchip supports "hardfp" rather then "softfp". So, I modify  my "arm.linux.cross.cmake" as following
 **SET(CMAKE_SYSTEM_PROCESSOR "armv7-a_hardfp")** 

And now, I could finally build mpp without errors. Thanks for your help, Herman!!
Here is my reference: https://stackoverflow.com/questions/9753749/arm-compilation-error-vfp-registered-used-by-executable-not-object-file

closed 2018-08-30 by oggcgg

HermanChen, 2018-08-30:
Thanks for report : )