rockchip-linux/mpp #52: RK3288开发板 linux下 运行mpi_dec_test失败

kenchiii, 2018-08-31:
根据 MPP开发参考_v0.3 编译及运行：
[root@rockchip:~]# mpi_dec_test -t 7 -i /root/Tennis1080p.h264 -n 10
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: /root/Tennis1080p.h264
mpi_dec_test: output file name: 
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 10
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 10786528
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: Without VCS info
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:119
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:169
mpp_device: mpp_device_init failed to find device for coding 7 type 0
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Segmentation fault

kenchiii, 2018-08-31:
[root@rockchip:~]# cat /proc/device-tree/compatible
rockchip,rk3288-ema-EVMproject-act8846rockchip,rk3288

HermanChen, 2018-08-31:
找不到内核设备，运行 mpp_platform_test 发一下结果。

kenchiii, 2018-08-31:
[root@rockchip:/udisk/test]# ./mpp_platform_test
mpp_plat_test: chip name: rockchip,rk3288-ema-EVMproject-act8846 rockchip,rk3288 
mpp_plat_test: 
mpp_plat_test: chip vcodec type 00000000
mpp_plat_test: 
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder: (null)
mpp_plat_test: H.265 decoder: (null)
mpp_plat_test: MJPEG decoder: (null)
mpp_plat_test: VP9   decoder: (null)
mpp_plat_test: avs   decoder: (null)
mpp_plat_test: 
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder: (null)
mpp_plat_test: H.265 encoder: (null)
mpp_plat_test: MJPEG encoder: (null)
mpp_plat_test: mpp platform test done

HermanChen, 2018-08-31:
应该是内核设备没找到，内核可能有点问题，看一下 /dev/ 目录下有没有 vpu_service，或者 rkhevc 这样的设备

kenchiii, 2018-08-31:
两个都没有，那应该是内核配置的问题吧？请问怎么解决呢，内核要怎么配置？

kenchiii, 2018-08-31:
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In 请问这个也是内核配置的问题吗

HermanChen, 2018-08-31:
后面这个 log 是正常的，没有问题

kenchiii, 2018-08-31:
请问想彻底解决这个问题，是要重新配置内核，打开vpu_service，或者 rkhevc 这样的设备吗？第一次用mpp，不太懂

HermanChen, 2018-08-31:
是的，估计是内核dts里设备没配置上，可以去dts里搜一下vpu_service/vpu/rkhevc 这些的关键字。
mpp 需要依赖这边内核设备来运行。

HermanChen, 2018-08-31:
或者是用的开源的 linux 内核，没带上 vcodec_service 这个驱动。

HermanChen, 2018-09-05:
如果解决了这个问题请关闭一下，谢谢：）

HermanChen, 2018-09-05:
多谢，如果 dts 修正之后还有 mpp 相关问题的话，欢迎提出，再次感谢~

kenchiii, 2018-10-16:
你好，为什么我已经在内核中打开了vpu-service 在dev下也看到了vpu-service 但运行vpu_api_test时仍提示以下信息？
[image]

kenchiii, 2018-10-16:
[image]

HermanChen, 2018-10-16:
运行 mpp_platform_test 看一下打印什么出来

HermanChen, 2018-10-16:
内核设备看起来正常，平台检测也是ok的。
用 mpi_dec_test 测试看一下，vpu_api_test 已经很久没维护了，出错估计是正常的……

kenchiii, 2018-10-16:
mpi_dec_test -t 7 -i /root/Tennis1080p.h264 -n 10
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: /root/Tennis1080p.h264
mpi_dec_test: output file name: 
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 10
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 10786528
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: Without VCS info
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: found ion allocator
mpp_rt: found drm allocator
mpp_rt: found drm allocator in dts                                      
mpi_dec_test: decode_get_frame get info changed found                   
mpi_dec_test: decoder require buffer w:h [1920:1080] stride [1920:1088]8
mpi_dec_test: decode_get_frame get frame 1                              
mpi_dec_test: decode_get_frame get frame 2                              
mpi_dec_test: decode_get_frame get frame 3                              
mpi_dec_test: decode_get_frame get frame 4                              
mpi_dec_test: decode_get_frame get frame 5                              
mpi_dec_test: decode_get_frame get frame 6                              
mpi_dec_test: decode_get_frame get frame 7                              
mpi_dec_test: decode_get_frame get frame 8                              
mpi_dec_test: decode_get_frame get frame 9                              
mpi_dec_test: decode_get_frame get frame 10                             
mpi_dec_test: reach max frame number 10                                 
[ 3060.165757] rk-vcodec ff9a0000.vpu-service: closed                   
mpi_dec_test: test success max memory 14.34 MB 
mpi_dec_test运行正常吧？

HermanChen, 2018-10-16:
正常

kenchiii, 2018-10-16:
那可能是demo代码的问题了？最后提示一个Segmentation fault

HermanChen, 2018-10-16:
嗯，vpu_api_test 已经好久不维护了

kenchiii, 2018-10-16:
好的。那请问一下你们是否会安排维护呢？谢谢。

HermanChen, 2018-10-16:
因为没什么人用，基本不会维护……
现在主要是用 mpi_dec_test 和 mpi_enc_test

kenchiii, 2018-10-16:
好的。谢谢你。我一开始还以为是内核又漏了哪里没配置。

kenchiii, 2018-10-16:
你好，现在是可以解码了。但内存泄漏比较严重，每秒吃内存1.5M。你们有什么改进的建议嘛？谢谢

HermanChen, 2018-10-16:
估计是解码出来的 buffer 没有释放掉，参考 mpi_dec_test 的做法

kenchiii, 2018-10-16:
特别是在调用decode_getframe时，容易出现内存泄漏

HermanChen, 2018-10-16:
get_frame 出来的 frame 要 deinit 掉

kenchiii, 2018-10-16:
我现在是每次开头分配，末尾释放。不这样做的话，程序就会崩溃。

HermanChen, 2018-10-16:
还有，看一下会不会到一定程度就稳定住，因为解码器是在尽力解码，会出现解得太多太快的情况

HermanChen, 2018-10-16:
尽量照 demo 的流程来

kenchiii, 2018-10-17:
你好，不好意思，打扰了。请问vpu_api_test是不是不能跑多线程？我现在解码demo没有问题，但是我把他做成多线程，无法取得解码后的数据，读取的数据字节总是为0。

acquo, 2018-10-19:
不好意思, 打擾, 我一樣在關注這個 issue, 也在這邊發問
我是用 firefly RK3288 linux, 一樣遇到 mpi_dec_test decode 失敗的問題
但錯誤訊息是 mpi_dec_test: decoder_get_frame get err info:1 discard:0. 

$ ./mpp_platform_test
mpp_plat_test: chip name: firefly,rk3288-firefly rockchip,rk3288
mpp_plat_test: 
mpp_plat_test: chip vcodec type 00000101
mpp_plat_test: found vpu1 codec
mpp_plat_test: found RK hevc decoder
mpp_plat_test: 
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder: /dev/vpu-service
mpp_plat_test: H.265 decoder: /dev/hevc-service
mpp_plat_test: MJPEG decoder: /dev/vpu-service
mpp_plat_test: VP9   decoder: /dev/vpu-service
mpp_plat_test: avs   decoder: /dev/vpu-service
mpp_plat_test: 
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder: /dev/vpu-service
mpp_plat_test: H.265 encoder: /dev/vpu-service
mpp_plat_test: MJPEG encoder: /dev/vpu-service
mpp_plat_test: mpp platform test done

$ ./mpi_dec_test -t 7 -i ./Tennis1080p.h264 -n 10
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: ./Tennis1080p.h264
mpi_dec_test: output file name: 
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 10
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 10786528
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: Without VCS info
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_dec_test: decode_get_frame get info changed found
mpi_dec_test: decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 1
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 2
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 3
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 4
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 5
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 6
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 7
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 8
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 9
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 10
mpi_dec_test: reach max frame number 10
mpi_dec_test: test success max memory 14.34 MB

WainDing, 2018-10-19:
开一下调试开关  export rkv_h264d_debug=0xFFFF，然后log发一下

acquo, 2018-10-19:
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: ./Tennis1080p.h264
mpi_dec_test: output file name: 
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 10
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 10786528
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: Without VCS info
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
[... 690 lines cut ...]
h264d_parse: SliceSTATE_ParseNalu
h264d_init: [DPB_malloc] g_framecnt=9, com_flag=0, mark_idx=1, slot_idx=1, slice_type=0, struct=3, lay_id=0
h264d_init: [DPB_INFO] cur_slot_idx=1
h264d_init: [DPB_INFO] ref_slot_idx[0]=2
h264d_init: [DPB_INFO] ref_slot_idx[1]=3
h264d_init: [DPB_INFO] ref_slot_idx[2]=0
h264d_init: [DPB_MARK_INFO] slot_idx=1, top_used=1, bot_used=1, out_flag=1, fd=0xff
h264d_init: [DPB_MARK_INFO] ---------- cur_slot=1 --------------------
h264d_init: find short reference, slot_idx=0.
h264d_init: find short reference, slot_idx=0.
h264d_init: find short reference, slot_idx=3.
[... cut ...]
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=607, pkt_no=10
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=10
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 10
mpi_dec_test: reach max frame number 10
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=10
h264d_dpb: warning: max_dec_frame_buffering(3) is less than dpb_size(4) calculated from Profile/Level.
mpi_dec_test: test success max memory 14.34 MB

WainDing, 2018-10-19:
从log看是硬件报了错误。能否码流上传一下，我看一下，这边能不能复现。

acquo, 2018-10-19:
我是直接用這個
https://github.com/sliver-chen/mpp_linux_cpp/blob/master/res/Tennis1080p.h264

acquo, 2018-10-23:
請問在你們的機器上可以復現嗎? 我還需要提供什麼資訊?

acquo, 2018-10-23:
我改用 https://www.dropbox.com/s/3n5pdp9y35j3kxo/test.h264?dl=0
這個碼流用 ffmpeg -i test.h264 rec.yuv  沒有報錯

但是 ./mpi_dec_test -t 7 -i ./test.h264 -n 10
還是會出現錯誤 mpi_dec_test: decoder_get_frame get err info:1 discard:0.
log: https://www.dropbox.com/s/sad8sg3pii11rtm/log.log?dl=0

WainDing, 2018-10-25:
不好意思，www.dropbox.com我这里登入不了

acquo, 2018-10-25:
抱歉, 沒注意到, 我先放在 github 上面
https://github.com/acquo/rklog/blob/master/log.log
https://github.com/acquo/rklog/blob/master/test.h264

我改用  release_20171218 就能正常 decode 了
我本來是用 Branch: release commit: 1a388b0a94fd8ccb92da2f4f5e3b76161fc599dd

zhigaowu, 2019-06-26:
> 好的。谢谢你。我一开始还以为是内核又漏了哪里没配置。

本人新人一枚，咨询一下这个您是怎么解决的呢？

closed 2023-08-03 by HermanChen