rockchip-linux/mpp #53: mpi_enc_test在3328-CC上無法編碼

boris0604, 2018-09-03:
在3328-CC進行測試
mpi_dec_test test.264->test.raw正常
mpi_enc_test test.raw->new.264一直會報length 0

是什麼問題？

HermanChen, 2018-09-03:
有完整log么？

boris0604, 2018-09-03:
#mpi_enc_test -i raw.yuv -o test.264 -t 7 -w 1280 -h 720 -d 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: raw.yuv
mpi_enc_test: output file name: test.264
mpi_enc_test: width      : 1280
mpi_enc_test: height     : 720
mpi_enc_test: format     : 0
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : 1
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 1280 h 720 type 7
mpi: mpp_create enter ctx 0x558a5780a8 mpi 0x558a5780b0
mpi: mpp version: 4968080 author: Herman Chen [mpp_impl]: Add tid to dump file
mpi: mpp_create leave ret 0 ctx 0x558a578b60 mpi 0x7f83dd05f0
mpi: mpp_init enter ctx 0x558a578b60 type 1 coding 7
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x558a578b60 cmd 320003 parm 0x558a5780b8
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 3456000 fps 30 gop 60
mpi: mpi_control enter ctx 0x558a578b60 cmd 320005 parm 0x558a578104
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 3456000 [3240000 : 3672000]
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x558a578b60 cmd 320007 parm 0x558a57813c
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x558a578b60 cmd 32000f parm 0x558a578278
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x558a578b60 cmd 32000e parm 0x7fdc606db8
mpi: mpi_control leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x558a578b60 frame 0x558a57f4c0
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x558a578b60 packet 0x7fdc606da8
mpi: mpi_encode_get_packet leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x558a578b60 frame 0x558a57f640
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x558a578b60 packet 0x7fdc606da8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 0 size 0
mpi: mpi_encode_put_frame enter ctx 0x558a578b60 frame 0x558a57f7c0
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x558a578b60 packet 0x7fdc606da8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 1 size 0
mpi: mpi_encode_put_frame enter ctx 0x558a578b60 frame 0x558a57f940
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x558a578b60 packet 0x7fdc606da8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 2 size 0
mpi: mpi_encode_put_frame enter ctx 0x558a578b60 frame 0x558a57fac0
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x558a578b60 packet 0x7fdc606da8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 3 size 0
mpi: mpi_encode_put_frame enter ctx 0x558a578b60 frame 0x558a57fc40
mpi: mpi_encode_put_frame leave ret 0

boris0604, 2018-09-03:
利用gst＠3328-CC進行測試也沒辦法編碼

gst-launch-1.0 videotestsrc num-buffers=512 ! video/x-raw,format=NV12,width=640,height=480,framerate=30/1 ! queue ! mpph264enc ! queue ! h264parse ! mpegtsmux ! filesink location=/home/firefly/h264.ts

HermanChen, 2018-09-05:
编码器的命令行里少了对图像格式的指定，加一个 -f 4 看看。4 是 YUV420SP 格式。
我这边的测试如下：
mpi_enc_test -w 720 -h 480 -t 7 -f 4 -n 60 -i /sdcard/soccer_720x480_30fps.yuv -o /sdcard/out.h264
打印如下：
04-11 08:20:48.478 I/mpi_enc_test( 1216): cmd parse result:
04-11 08:20:48.478 I/mpi_enc_test( 1216): input  file name: /sdcard/soccer_720x480_30fps.yuv
04-11 08:20:48.478 I/mpi_enc_test( 1216): output file name: /sdcard/out.h264
04-11 08:20:48.478 I/mpi_enc_test( 1216): width      : 720
04-11 08:20:48.478 I/mpi_enc_test( 1216): height     : 480
04-11 08:20:48.478 I/mpi_enc_test( 1216): format     : 4
04-11 08:20:48.478 I/mpi_enc_test( 1216): type       : 7
04-11 08:20:48.478 I/mpi_enc_test( 1216): debug flag : 0
04-11 08:20:48.479 I/mpi_enc_test( 1216): mpi_enc_test start
04-11 08:20:48.479 I/mpp_rt  ( 1216): NOT found ion allocator
04-11 08:20:48.479 I/mpp_rt  ( 1216): found drm allocator
04-11 08:20:48.481 I/mpi_enc_test( 1216): mpi_enc_test encoder test start w 720 h 480 type 7
04-11 08:20:48.481 I/mpi     ( 1216): mpp version: 84bc2e3 author: sayon.chen [vepu] vepu ratecontrol modify
04-11 08:20:48.483 I/mpi_enc_test( 1216): mpi_enc_test bps 1296000 fps 30 gop 60
04-11 08:20:48.483 I/h264e_api( 1216): h264e_config MPP_ENC_SET_RC_CFG bps 1296000 [1215000 : 1377000]
04-11 08:20:48.496 I/mpi_enc_test( 1216): test_mpp_run encoded frame 0 size 50191
04-11 08:20:48.511 I/mpi_enc_test( 1216): test_mpp_run encoded frame 1 size 12751
04-11 08:20:48.519 I/mpi_enc_test( 1216): test_mpp_run encoded frame 2 size 4911
04-11 08:20:48.527 I/mpi_enc_test( 1216): test_mpp_run encoded frame 3 size 4678
04-11 08:20:48.535 I/mpi_enc_test( 1216): test_mpp_run encoded frame 4 size 3860
是正常的。

请再尝试一下。

noah538, 2018-09-11:
剛試過可以正常解出out.264,
想請問解出的 out.264 如何輸入給 mpi_dec_test 測試?

HermanChen, 2018-09-12:
mpi_dec_test -t 7 -i out.264 -o out.264
mpi_dec_test 直接运行有说明

noah538, 2018-09-12:
@HermanChen 

1. 使用 /usr/local/bin 的 mpi_dec_test  可以正常解碼
**root@linaro-alip:/home/linaro# which mpi_dec_test
/usr/local/bin/mpi_dec_test**

root@linaro-alip:/home/linaro# mpi_dec_test -t 7 -i out.h264 -o out.h264 -d 1
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: out.h264
mpi_dec_test: output file name: out.h264
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 1
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 0
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: 598cae3 author: Jacob Chen DEBIAN: update rules for release_20171218-2
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi: mpp_init leave ret 0
mpi_dec_test: found last packet
mpi: mpi_decode_put_packet enter ctx 0x80b33740 packet 0x80b336c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x80b33740 frame 0xbef012ac
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x80b33740 frame 0xbef012ac
mpi: mpi_decode_get_frame leave ret 0
mpi_dec_test: decode_get_frame get frame 0
mpi_dec_test: found last frame
mpi: mpi_reset enter ctx 0x80b33740
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x80b33740
mpi: mpp_destroy leave ret 0
mpi_dec_test: test success

2. 同一個 out.h264 使用 local build 出來的 mpi_dec_test  會有錯誤
**root@linaro-alip:/home/linaro# which ./mpi_dec_test
./mpi_dec_test**

root@linaro-alip:/home/linaro# ./mpi_dec_test -t 7 -i out.h264 -o out.h264 -d 1
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: out.h264
mpi_dec_test: output file name: out.h264
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 1
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 0
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: 598cae3 author: Jacob Chen DEBIAN: update rules for release_20171218-2
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi: mpp_init leave ret 0
mpi_dec_test: found last packet
mpi: mpi_decode_put_packet enter ctx 0x8cd740 packet 0x8cd6c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x8cd740 frame 0xbea572c0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x8cd740 frame 0xbea572c0
mpi: mpi_decode_get_frame leave ret 0
mpi_dec_test: decode_get_frame get frame 1
mpi_dec_test: found last frame
mpi: mpi_reset enter ctx 0x8cd740
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x8cd740
mpi: mpp_destroy leave ret 0
*** Error in `./mpi_dec_test': free(): invalid pointer: 0x00016c98 ***
Aborted

HermanChen, 2018-09-12:
后一个解码过程也是没问题的，只是在输入销毁资源的时候有问题
应该对比下代码就可以发现问题。
另外，这两个库的版本信息为啥是一样的？没有从这个库里编译出来么？

HermanChen, 2018-09-18:
> 
> 
> 我的環境ROC-3328-CC , Debian9-arch64, H.264還是不能encode，只有sps / pps / sei ，frame都是空的
> mpp在cmake的時候出現
> CMAKE_SYSTEM_PROCESSOR value `aarch64` is unknown
> 這有影響嗎？

这个应该 toolchain 有问题吧？

HermanChen, 2018-09-18:
> 
> 
> 后一个解码过程也是没问题的，只是在最后销毁资源的时候有问题
> 应该对比下代码就可以发现问题。
> 另外，这两个库的版本信息为啥是一样的？没有从这个库里编译出来么？

这种错误感觉像是 C 库不匹配？估计也是运行库环境的问题……

boris0604, 2018-09-18:
> > 我的環境ROC-3328-CC , Debian9-arch64, H.264還是不能encode，只有sps / pps / sei ，frame都是空的
> > mpp在cmake的時候出現
> > CMAKE_SYSTEM_PROCESSOR value `aarch64` is unknown
> > 這有影響嗎？
> 
> 这个应该 toolchain 有问题吧？

感覺不是，decoder運行是正常的，只有h.264 encode運行有問題

HermanChen, 2018-09-18:
试试不指定输入 yuv 数据，使用 mpp 绘制的图像看看：
mpi_enc_test -w 1280 -h 720 -t 7 -f 4 这样编码看看有没有出错，不指定输入和输出。

HermanChen, 2018-09-19:
硬件输出的长度固定，这肯定不对……
是否drm开了cache，只看得到软件的数据，看不到硬件的输出？
更新下 mpp 的版本看看

boris0604, 2018-09-21:
問題解了
用32bit toolchain編譯，然後套用rkmpp-0001-fix-32-bit-mmap-issue-on-64-bit-kernels

HermanChen, 2018-09-25:
好的，多谢反馈，我关闭问题了。
那个 patch 也可以发看一是是什么问题。

closed 2018-09-25 by HermanChen

Rereflyer, 2018-10-19:
> 剛試過可以正常解出out.264,
> 想請問解出的 out.264 如何輸入給 mpi_dec_test 測試?

你好，请问“mpi_enc_test test.raw->new.264一直會報length 0”最后是如何解决的？我们也遇到了相同的问题，以下是mpp_platform_test的输出：
mpp_plat_test: chip name: firefly,roc-rk3328-cc rockchip,rk3328
mpp_plat_test: 
mpp_plat_test: chip vcodec type 01021202
mpp_plat_test: found vpu2 codec
mpp_plat_test: found rkvdec decoder
mpp_plat_test: found avs+ decoder
mpp_plat_test: found vpu2 encoder
mpp_plat_test: found h265 stand-alone encoder
mpp_plat_test: 
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp_plat_test: avs   decoder: /dev/vpu_service
mpp_plat_test: 
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder: /dev/vepu
mpp_plat_test: H.265 encoder: /dev/h265e
mpp_plat_test: MJPEG encoder: /dev/vepu
mpp_plat_test: mpp platform test done

ethancwchen, 2018-11-21:
irefly@firefly:~/mpp-release/build/linux/aarch64/test$ sudo mpi_enc_test -i t.yuv -w 640 -h 272 -t 7 -o t.h264 -f 4
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: t.yuv
mpi_enc_test: output file name: t.h264
mpi_enc_test: width      : 640
mpi_enc_test: height     : 272
mpi_enc_test: format     : 4
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 640 h 272 type 7
mpi: mpp version: Without VCS info
mpi_enc_test: mpi_enc_test bps 652800 fps 30 gop 60
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 652800 [612000 : 693600]
mpi_enc_test: test_mpp_run encoded frame 0 size 0
mpi_enc_test: test_mpp_run encoded frame 1 size 0
mpi_enc_test: test_mpp_run encoded frame 2 size 0
mpi_enc_test: test_mpp_run encoded frame 3 size 0
mpi_enc_test: test_mpp_run encoded frame 4 size 0
mpi_enc_test: test_mpp_run encoded frame 5 size 0
mpi_enc_test: test_mpp_run encoded frame 6 size 0
mpi_enc_test: test_mpp_run encoded frame 7 size 0
mpi_enc_test: test_mpp_run encoded frame 8 size 0
mpi_enc_test: test_mpp_run encoded frame 9 size 0
mpi_enc_test: test_mpp_run encoded frame 10 size 0
mpi_enc_test: test_mpp_run encoded frame 11 size 0
mpi_enc_test: test_mpp_run encoded frame 12 size 0
mpi_enc_test: test_mpp_run encoded frame 13 size 0
mpi_enc_test: test_mpp_run encoded frame 14 size 0
mpi_enc_test: test_mpp_run encoded frame 15 size 0
mpi_enc_test: test_mpp_run encoded frame 16 size 0
mpi_enc_test: test_mpp_run encoded frame 17 size 0

ethancwchen, 2018-11-21:
>           在3328-CC進行測試
> mpi_dec_test test.264->test.raw正常
> mpi_enc_test test.raw->new.264一直會報length 0
> 是什麼問題？

HermanChen, 2018-11-21:
上面的那位说是 toolchain 相关的问题

ethancwchen, 2018-11-21:
>           問題解了
> 用32bit toolchain編譯，然後套用rkmpp-0001-fix-32-bit-mmap-issue-on-64-bit-kernels
firefly@firefly:~/mpp-release/build/linux/arm/test$ ./mpi_test
-bash: ./mpi_test: No such file or directory
firefly@firefly:~/mpp-release/build/linux/arm/test$ ./mpi_enc_test
-bash: ./mpi_enc_test: No such file or directory
firefly@firefly:~/mpp-release/build/linux/arm/test$

ethancwchen, 2018-11-22:
我是在3328上做编辑，請問要如何修正呢

從我的 ASUS 傳送

-------- 原始郵件 --------
寄件者：Herman Chen <notifications@github.com>
傳送日期：Wed, 21 Nov 2018 10:32:23 +0800
收件者：rockchip-linux/mpp <mpp@noreply.github.com>
副本：ethancwchen <ethancwchen@gmail.com>,Comment <comment@noreply.github.com>
主旨:Re: [rockchip-linux/mpp] mpi_enc_test在3328-CC上無法編碼 (#53)

ethancwchen, 2018-11-22:
> 用32bit toolchain編譯，然後套用rkmpp-0001-fix-32-bit-mmap-issue-on-64-bit-kernels
那兒可以找到呢?

boris0604, 2018-11-23:
> > 用32bit toolchain編譯，然後套用rkmpp-0001-fix-32-bit-mmap-issue-on-64-bit-kernels
> > 那兒可以找到呢?

https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages/multimedia/rkmpp/patches