rockchip-linux/mpp #54: Crash in mpp_buf_slot_get_prop

Kwiboo, 2018-09-03:
LibreELEC/Kodi users are reporting crashes when trying to open some Tvheadend streams.

I was able to capture the following backtrace using a debug build:
```
Thread 1 (Thread 0xdbe86370 (LWP 823)):
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0xf55b9150 in __GI_abort () at abort.c:79
#2  0xf494405c in mpp_buf_slot_get_prop (slots=0xf03f4c60, index=index@entry=-1, type=type@entry=SLOT_FRAME_PTR, val=0xdbe85b8c, val@entry=0xdbe85b84) at ../mpp/base/mpp_buf_slot.cpp:875
#3  0xf48f4b14 in check_ref_dbp_err (p_Dec=p_Dec@entry=0xf1add5c0, pref=<optimized out>, active_refs=active_refs@entry=2) at ../mpp/codec/dec/h264/h264d_init.c:1391
#4  0xf48f6ecc in check_refer_picture_lists (currSlice=0xf03f8270) at ../mpp/codec/dec/h264/h264d_init.c:1422
#5  init_picture (currSlice=0xf03f8270) at ../mpp/codec/dec/h264/h264d_init.c:1970
#6  0xf48ea69c in parse_loop (p_Dec=p_Dec@entry=0xf1add5c0) at ../mpp/codec/dec/h264/h264d_parse.c:907
#7  0xf48e8050 in h264d_parse (decoder=0xf1add5c0, in_task=0xdbe85cf0) at ../mpp/codec/dec/h264/h264d_api.c:599
#8  0xf48e150c in try_proc_dec_task (task=0xdbe85ccc, mpp=0xf0362e88) at ../mpp/codec/mpp_dec.cpp:608
#9  mpp_dec_parser_thread (data=0xf0362e88) at ../mpp/codec/mpp_dec.cpp:783
#10 0xf75c4314 in start_thread (arg=0xdbe86370) at pthread_create.c:486
#11 0xf5670498 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /usr/lib/libc.so.6
```

Looks like `index=index@entry=-1` is the culprit.

Using the following patch seems to mitigate the crash in mpp, please validate this workaround.

```
diff --git a/mpp/codec/dec/h264/h264d_init.c b/mpp/codec/dec/h264/h264d_init.c
--- a/mpp/codec/dec/h264/h264d_init.c
+++ b/mpp/codec/dec/h264/h264d_init.c
@@ -1387,7 +1387,11 @@ static RK_U32 check_ref_dbp_err(H264_DecCtx_t *p_Dec, H264_RefPicInfo_t *pref, R
     for (i = 0; i < MAX_REF_SIZE; i++) {
         if (pref[i].valid) {
             MppFrame mframe = NULL;
-            RK_U32 slot_idx = p_Dec->dpb_info[pref[i].dpb_idx].slot_index;
+            RK_S32 slot_idx = p_Dec->dpb_info[pref[i].dpb_idx].slot_index;
+            if (slot_idx < 0) {
+                dpb_error_flag |= 1;
+                break;
+            }
             mpp_buf_slot_get_prop(p_Dec->frame_slots, slot_idx, SLOT_FRAME_PTR, &mframe);
             if (mframe) {
                 if (i < active_refs) {
```

WainDing, 2018-09-03:
Could you tell me the android verison ? And, please recode the stream which can be reappearance。

Kwiboo, 2018-09-03:
This was on linux running on a Rock64 and not Android, sorry for missing enviorenment details.
Latest rockchip 4.4 kernel and u-boot is used.

mpp is latest release branch on github + 3 patches at https://github.com/Kwiboo/mpp/commits/libreelec (h264 fast decode mode reverted to fix "jumpy" playback on rk3288/rk3399).

Test images with this issue can be found at http://cvh.libreelec.tv/test/RK3328/

WainDing, 2018-09-04:
From the log point of view, there should be an exception when applying for slot.
Please  do these:
export rkv_h264d_debug=0xFFFF
and show me the details.
Thanks.

closed 2021-05-19 by HermanChen