rockchip-linux/mpp #57: 10-Bit H264 broken in Kodi after April 2018 changes

mo123, 2018-09-08:
Hi

It's not possible to play 10-bit H264 in Kodi 18 on Nougat anymore if using mpp after April 2018 on RK3328, RK3229.

mo123, 2018-09-18:
Hope it can be fixed, last worked in around April 2018 code but that doesn't have deinterlacing support which is useful for h264/mpeg2 1080i live tv video playback in Kodi which is in the latest code.

WainDing, 2018-09-18:
Can recoder a short media for me ? And tell me before which commit can worke?

mo123, 2018-09-18:
Worked here, maybe a few commits after too.
5db93dc author: Herman Chen [utils]: Add iniparser for ini file parsing
April 11 2018 commit

The 10-bit H264 Level 5.1/5.2 can still work in MediaCenter and Kodi 18(modified version with 10-bit H264 support) with above mpp.
So sometime after that the 10-bit H264 support broke in Android.

Modified Kodi 18 with 10-bit h264 support, can use for testing or MediaCenter v3.2.3
https://mega.nz/#!1Q1ADYJL!7LA-b4D-I6gLtPjjMvcnMNZSiTyUaocjFuwKsaKJTNc

Firmware tested was Nougat v1.24 on RK3328, 3.10 kernel and also contained 10-bit patch below.
Without patch, 10-bit H264 would also not work.
Subject: [PATCH] add H264/H265 4K p60 10bit support
Date: 30 Mar 2018

Sample videos
They would freeze or play with strange colors(only 8-bit)
https://www.koi-sama.net/files/hi10/
https://cloud.mail.ru/public/J2qe/PU6j9Z8Bc

On LibreELEC Linux they all play fine, maybe because of updated 4.4 kernel, no MediaCodec problems as on Android or other changes.

420584113, 2018-09-19:
@mo123 , Can we provide logcat with problems? This helps us to view the problem version.

420584113, 2018-09-19:
please exec cmd:  setprop use_mpp_mode 1 when play H264 10bit video before.

It can enable H264 10bit support.

mo123, 2018-09-20:
@rimonxu 

I tried new patch but still have the same problem.
Kodi 18(with 10-bit h264 support) would still only play 10-bit videos in 8-bit color with videos having purple, green color problems.
Any other solution?

rimonxu, 2018-09-20:
`diff --git a/mpp/legacy/vpu_api.cpp b/mpp/legacy/vpu_api.cpp
index cff6533..c0a9845 100644
--- a/mpp/legacy/vpu_api.cpp
+++ b/mpp/legacy/vpu_api.cpp
@@ -314,8 +314,8 @@ RK_S32 vpu_open_context(VpuCodecContext **ctx)
use_mpp = 0;
} else {
if (s->videoCoding == OMX_RK_VIDEO_CodingAVC

-       && s->codecType == CODEC_DECODER && s->width <= 1920
-       && s->height <= 1088 && !s->extra_cfg.mpp_mode
+      && s->codecType == CODEC_DECODER
+      && !s->extra_cfg.mpp_mode
       && strncmp(mpp_get_soc_name(), "rk3399", 6)) {
       /* H.264 smaller than 1080p use original vpuapi library for better error process */
       // NOTE: rk3399 need better performance`

please patch up this patch. If there are any more questions, please provide a log.

rimonxu, 2018-09-20:
Please give me your email, i provice new patch to you .

closed 2023-08-03 by HermanChen