rockchip-linux/mpp #58: mpi_dec_test doesn't output raw yuv data as expected

noah538, 2018-09-11:
issue: mpi_dec_test doesn't output raw yuv data as expected

chip: RK3288 armv71
os: linux
decoder:  mpi_dec_test doesn't output raw yuv data as expected
input file:  1.out.h264  was encoded by mpi_enc_test type 7  (only 1 frame)

root@linaro-alip:/home/linaro# ./mpi_dec_test -i 1.out.h264 -o 1.out.yuv -w 1920 -h 1080 -t 7 -d 1
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: 1.out.h264
mpi_dec_test: output file name: 1.out.yuv
mpi_dec_test: width      : 1920
mpi_dec_test: height     : 1080
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 1
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 102072
mpi_dec_test: mpi_dec_test decoder test start w 1920 h 1080 type 7
mpi: mpp version: 598cae3 author: Jacob Chen DEBIAN: update rules for release_20171218-2
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi: mpp_init leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi_dec_test: found last packet
mpi: mpi_decode_put_packet enter ctx 0x672320 packet 0x6722c0
mpi: mpi_decode_put_packet leave ret 0
mpi: mpi_decode_get_frame enter ctx 0x672320 frame 0xbeb6b8b0
mpi: mpi_decode_get_frame leave ret 0
mpi_dec_test: decode_get_frame get frame 1
mpi_dec_test: found last frame
mpi: mpi_reset enter ctx 0x672320
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x672320
[75800.270282] rk-vcodec ff9a0000.vpu-service: closed
mpi: mpp_destroy leave ret 0
*** Error in `./mpi_dec_test': free(): invalid pointer: 0x00016c98 ***
Aborted
root@linaro-alip:/home/linaro#

HermanChen, 2018-09-11:
Can you provide the test stream?

johlim, 2018-09-11:
me too.

 now I amd using tinker board (RK3288)

1) modify CMakelist.txt to build for 
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 7455475..a1e5dc8 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -6,6 +6,12 @@
 # ----------------------------------------------------------------------------

 # vim: syntax=cmake
+set(CMAKE_C_COMPILER "/usr/bin/arm-linux-gnueabihf-gcc" )
+set(CMAKE_CXX_COMPILER "/usr/bin/arm-linux-gnueabihf-g++" )
+
+include_directories( "/usr/local/include/rockchip")
+include_directories( "/usr/include/libdrm")
+
 if(NOT CMAKE_BUILD_TYPE)
     # default to Release build for GCC builds
     set(CMAKE_BUILD_TYPE Debug CACHE STRING
@@ -102,6 +108,11 @@ elseif(${SYSPROC} STREQUAL "armv7-a_hardfp")
     set(ARM true)
     set(ARMEABI_V7A_HARDFP true)
     add_definitions(-DARCH_ARM=1 -DDHAVE_ARMV7=1)
+elseif(${SYSPROC} STREQUAL "armv7l")
+    message(STATUS "Detected ARMv7 system processor")
+    set(ARM true)
+    set(ARMEABI_V7A_HARDFP true)
+    add_definitions(-DARCH_ARM=1 -DDHAVE_ARMV7=1)

2. sudo make install 
3. sudo ldconfig /usr/local/lib
4. run /usr/local/bin/mpp_dec_test
mpi_dec_test -i /home/linaro/h264.ts -o 1.out.yuv -w 640 -h 480 -t 7 -d 1

h264.ts file made by "gst-launch-1.0 videotestsrc num-buffers=512 ! video/x-raw,format=NV12,width=640,height=480,framerate=30/1 ! queue ! mpph264enc ! queue ! h264parse ! mpegtsmux ! filesink location=/home/linaro/h264.ts"

5. mpp_dec_test made kernel panic like this

[ 6143.853455] rk_vcodec: reg_init:1184: error: translate reg address failed, dumping regs
[ 6143.862348] Unhandled fault: page domain fault (0x01b) at 0x814d1e80
[ 6143.869368] pgd = de814000
[ 6143.872354] [814d1e80] *pgd=24587835, *pte=625e875f, *ppte=625e8c7f
[ 6143.879284] Internal error: : 1b [#3] SMP ARM
[ 6143.884095] Modules linked in: 8723bs
[ 6143.888141] CPU: 3 PID: 10988 Comm: mpp_dec_parser Tainted: G      D         4.4.103+ #1
[ 6143.897080] Hardware name: Rockchip (Device Tree)
[ 6143.902274] task: e542f400 task.stack: e68d6000
[ 6143.907281] PC is at reg_init+0x73c/0x834
[ 6143.911710] LR is at irq_work_queue+0x4c/0x84
[ 6143.916520] pc : [<c0554450>]    lr : [<c01dc22c>]    psr: 80070013
[ 6143.916520] sp : e68d7d88  ip : e68d7c68  fp : e68d7eb4
[ 6143.929205] r10: ec2de580  r9 : 00000051  r8 : 00000194
[ 6143.934974] r7 : c0e28535  r6 : 00000000  r5 : 00000065  r4 : e81a3800
[ 6143.942183] r3 : 814d1e80  r2 : 00000000  r1 : c0e28535  r0 : edccd418
[ 6143.949393] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[ 6143.957276] Control: 10c5387d  Table: 1e81406a  DAC: 00000051
[ 6143.963613]
[ 6143.963613] PC: 0xc05543d0:
[ 6143.968316] 43d0  e5d332ac e3530000 0a000003 e59f0154 eb0d34ce e3500000 0a00001d e59f30fc
[ 6143.977381] 43f0  e5933000 e3130020 0a000003 e59f10f0 e30024b2 e59f0130 ebf25ed0 e1a00004
[ 6143.986445] 4410  ea000033 e51b0100 e1a05128 eb1a28e7 e3a02e4a e59f10c8 e3a06000 e59f010c
[ 6143.995510] 4430  ebf25ec6 e59f7108 e1560005 0affffc5 e51b30fc e1a02006 e51b00e0 e1a01007
[ 6144.004575] 4450  e7933106 e2866001 e59002f0 eb0368a2 eafffff4 e5941000 e3c13002 e3530001
[ 6144.013640] 4470  1a00000d e5942038 e592300c e1b00e23 1a000005 e5923010 e1a03ba3 e35300c8
[ 6144.022707] 4490  c3a03005 ca000003 ea000003 e3130502 0a000001 e3a03003 e5843004 e5952088
[ 6144.031767] 44b0  e3063867 e1520003 1a000005 e5943038 e5932020 e30e3a60 e1520003 c3a03003
[ 6144.040822]
[... 94 lines cut ...]
[ 6144.774625] 7da0: 00000014 e81a3820 e81a3818 00000002 00000002 edccd624 814d1e80 c0bdd130
[ 6144.783659] 7dc0: 00000000 c12759bc 00000001 00000000 c0c5999c edccd418 00000000 c019bc70
[ 6144.792693] 7de0: 00000000 00000000 e68d7ecc e68d7df8 c019bc70 c019b7ac e68d7e14 c016ef88
[ 6144.801726] 7e00: 00000001 00000000 e68d7e5c e68d7e18 c0148044 eeaceec0 00000064 e68d7e1c
[ 6144.810760] 7e20: e68d7e1c e68d7e24 e68d7e24 e542f400 00000000 814d2000 ed468e00 c019b064
[ 6144.819793] 7e40: e68d7e84 e68d7e50 c019b064 c019af7c c0159434 c0113350 c01480b8 00000001
[ 6144.828827] 7e60: 00000001 e68d7e94 00000081 c019bb0c 00000000 eeacc200 e68d7ecc e68d7e88
[ 6144.837860] 7e80: c019bb0c c019b7ac 814d2000 ec2de580 b6569cb4 edccd418 c135de3c edc7be18
[ 6144.846893] 7ea0: 00000051 00000000 e68d7f14 e68d7eb8 c0555980 c0553d20 00000001 00000000
[ 6144.855927] 7ec0: e68d7f54 e68d7ed0 c019d514 c019b9dc ffffffff 814d1e80 00000194 e68d7ee8
[ 6144.864962] 7ee0: c01445bc c0be0428 e5424058 b6569cb4 edc03838 ebe7a000 40046c03 b6569cb4
[ 6144.873995] 7f00: 00000005 00000000 e68d7f7c e68d7f18 c0245414 c0555698 e68d7f6c e68d7f28
[ 6144.883028] 7f20: c0bdcbb8 c0144460 c0206934 00000001 00000001 c024f520 b6ee0000 b6ee0000
[ 6144.892062] 7f40: 00000001 40046c03 b6569cb4 00000005 e68d7f6c 00000000 ebe7a000 ebe7a001
[ 6144.901096] 7f60: 40046c03 b6569cb4 00000005 00000000 e68d7fa4 e68d7f80 c0245550 c0244e70
[ 6144.910129] 7f80: b6ee0000 b6ee0000 00000001 00000036 c0107484 e68d6000 00000000 e68d7fa8
[ 6144.919163] 7fa0: c01072c0 c0245500 b6ee0000 b6ee0000 00000005 40046c03 b6569cb4 00000005
[ 6144.928197] 7fc0: b6ee0000 b6ee0000 00000001 00000036 beabe130 00000000 00000000 00000000
[ 6144.937230] 7fe0: b6ee04a8 b6569c94 b6e8826d b6bca716 20070030 00000005 00000000 00000000
[ 6144.946272] [<c0554450>] (reg_init) from [<c0555980>] (vpu_service_ioctl+0x2f4/0x848)
[ 6144.954926] [<c0555980>] (vpu_service_ioctl) from [<c0245414>] (do_vfs_ioctl+0x5b0/0x690)
[ 6144.963962] [<c0245414>] (do_vfs_ioctl) from [<c0245550>] (SyS_ioctl+0x5c/0x84)
[ 6144.972038] [<c0245550>] (SyS_ioctl) from [<c01072c0>] (ret_fast_syscall+0x0/0x3c)
[ 6144.980400] Code: e51b30fc e1a02006 e51b00e0 e1a01007 (e7933106)
[ 6144.987130] ---[ end trace 2d4f8e2d155224d9 ]---

hizukiayaka, 2018-09-19:
I don't where that rockchip mpp come from, anyway just use the latest rockchip mpp.
If you have no idea on cross building, I would suggest that you had better start with the buildroot SDK from the rockchip. The rockchip mpp and Gstreamer are included in the buildroot based SDK and the default configure file would enable them.

closed 2023-08-03 by HermanChen