rockchip-linux/mpp #65: Provide samples for decoding in "pure external mode".

acquo, 2018-10-29:
I want to implement a zero-copy EGL/GLES h264 player, it looks like I should use pure external mode.
Could your team provide a sample for demonstrating how to use "pure external mode"?
Thanks.

HermanChen, 2018-10-29:
It just like Android way. Surfaceflinger create GrallocBuffer with ion/drm file handle. There will be 16 buffers or more. Then commit these buffer to a external mode MppBufferGroup. And set this buffer group to mpp decoder. The mpp decoder will automatically decode image to selected buffer and return to user by output MppFrame.

HermanChen, 2023-11-24:
Try this commit 431058ad9ded69e9ff95b99040131526344dc798 for pure external mode with option -bufmode e

closed 2023-11-29 by HermanChen