rockchip-linux/mpp #71: 切换视频分辨率mpp会崩溃。

liyungithub, 2018-11-22:
通过gstreamer插件rtspsrc取视频流，mppvideodec插件解码，原来输入的分辨是704x576，切换回352x288后，mpp崩溃
[image]

HermanChen, 2018-11-23:
有没有完整log

liyungithub, 2018-11-23:
h264d_init: [DPB_malloc] g_framecnt=1754, com_flag=0, mark_idx=3, slot_idx=3, slice_type=0, struct=3, lay_id=0
h264d_init: [DPB_INFO] cur_slot_idx=3
h264d_init: [DPB_INFO] ref_slot_idx[0]=2
h264d_init: [DPB_MARK_INFO] slot_idx=0, top_used=1, bot_used=1, out_flag=1, fd=0xf8
h264d_init: [DPB_MARK_INFO] slot_idx=1, top_used=1, bot_used=1, out_flag=1, fd=0xf9
h264d_init: [DPB_MARK_INFO] slot_idx=2, top_used=1, bot_used=1, out_flag=1, fd=0xfa
h264d_init: [DPB_MARK_INFO] slot_idx=3, top_used=1, bot_used=1, out_flag=1, fd=0xff
h264d_init: [DPB_MARK_INFO] ---------- cur_slot=3 --------------------
h264d_init: [DPB_REF_ERR] slot_idx=2, dpb_err[0]=0
h264d_init: list0 dpb: cur_err_flag=0, pps_refs=1, over_flag=0, num_ref_l0=1
h264d_parse: SliceSTATE_InitPicture
h264d_parse: SliceSTATE_GetSliceData
h264d_parse: SliceSTATE_RegisterOneFrame
h264d_dpb: [DPB_size] p_Dpb->used_size=4
h264d_api: [CALLBACK] g_no=1755, out_idx=3, dpberr=0, harderr=0, ref_flag=1, errinfo=0, discard=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=1204, pkt_no=1756
h264d_parse: SliceSTATE_ResetSlice
h264d_parse: SliceSTATE_ReadNalu
h264d_parse: nalu_type=SLICE.
h264d_sps: [DPB_size] dpb_size[0]=6, mvc_flag=0, dpb_size[1]=0
h264d_slice: [SLICE_HEAD] type=0, layer_id=0,sps_id=0, pps_id=0, struct=3, frame_num=4
h264d_parse: SliceSTATE_ParseNalu
h264d_init: [DPB_malloc] g_framecnt=1756, com_flag=0, mark_idx=4, slot_idx=4, slice_type=0, struct=3, lay_id=0
h264d_init: [DPB_INFO] cur_slot_idx=4
h264d_init: [DPB_INFO] ref_slot_idx[0]=3
h264d_init: [DPB_MARK_INFO] slot_idx=0, top_used=1, bot_used=1, out_flag=1, fd=0xb4
h264d_init: [DPB_MARK_INFO] slot_idx=1, top_used=1, bot_used=1, out_flag=1, fd=0xb3
h264d_init: [DPB_MARK_INFO] slot_idx=2, top_used=1, bot_used=1, out_flag=1, fd=0xb5
h264d_init: [DPB_MARK_INFO] slot_idx=3, top_used=1, bot_used=1, out_flag=1, fd=0xb6
h264d_init: [DPB_MARK_INFO] slot_idx=4, top_used=1, bot_used=1, out_flag=1, fd=0xff
h264d_init: [DPB_MARK_INFO] ---------- cur_slot=4 --------------------
h264d_init: [DPB_REF_ERR] slot_idx=3, dpb_err[0]=0
h264d_init: list0 dpb: cur_err_flag=0, pps_refs=1, over_flag=0, num_ref_l0=1
h264d_parse: SliceSTATE_InitPicture
h264d_parse: SliceSTATE_GetSliceData
h264d_parse: SliceSTATE_RegisterOneFrame
h264d_dpb: [DPB_size] p_Dpb->used_size=5
h264d_api: [CALLBACK] g_no=1757, out_idx=4, dpberr=0, harderr=0, ref_flag=1, errinfo=0, discard=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=1177, pkt_no=1756
h264d_parse: SliceSTATE_ResetSlice
h264d_parse: SliceSTATE_ReadNalu
h264d_parse: nalu_type=SLICE.
h264d_sps: [DPB_size] dpb_size[0]=6, mvc_flag=0, dpb_size[1]=0
h264d_slice: [SLICE_HEAD] type=0, layer_id=0,sps_id=0, pps_id=0, struct=3, frame_num=4
h264d_parse: SliceSTATE_ParseNalu
h264d_init: [DPB_malloc] g_framecnt=1756, com_flag=0, mark_idx=4, slot_idx=4, slice_type=0, struct=3, lay_id=0
h264d_init: [DPB_INFO] cur_slot_idx=4
h264d_init: [DPB_INFO] ref_slot_idx[0]=3
h264d_init: [DPB_MARK_INFO] slot_idx=0, top_used=1, bot_used=1, out_flag=1, fd=0xba
h264d_init: [DPB_MARK_INFO] slot_idx=1, top_used=1, bot_used=1, out_flag=1, fd=0xb9
h264d_init: [DPB_MARK_INFO] slot_idx=2, top_used=1, bot_used=1, out_flag=1, fd=0xbb
h264d_init: [DPB_MARK_INFO] slot_idx=3, top_used=1, bot_used=1, out_flag=1, fd=0xbc
h264d_init: [DPB_MARK_INFO] slot_idx=4, top_used=1, bot_used=1, out_flag=1, fd=0xff
h264d_init: [DPB_MARK_INFO] ---------- cur_slot=4 --------------------
h264d_init: [DPB_REF_ERR] slot_idx=3, dpb_err[0]=0
h264d_init: list0 dpb: cur_err_flag=0, pps_refs=1, over_flag=0, num_ref_l0=1
h264d_parse: SliceSTATE_InitPicture
h264d_parse: SliceSTATE_GetSliceData
h264d_parse: SliceSTATE_RegisterOneFrame
h264d_dpb: [DPB_size] p_Dpb->used_size=5
h264d_api: [CALLBACK] g_no=1757, out_idx=4, dpberr=0, harderr=0, ref_flag=1, errinfo=0, discard=0
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=687, pkt_no=1755
h264d_parse: SliceSTATE_ResetSlice
h264d_parse: SliceSTATE_ReadNalu
h264d_parse: nalu_type=SLICE.
h264d_sps: [DPB_size] dpb_size[0]=5, mvc_flag=0, dpb_size[1]=0
h264d_slice: [SLICE_HEAD] type=0, layer_id=0,sps_id=0, pps_id=0, struct=3, frame_num=4
h264d_parse: SliceSTATE_ParseNalu
h264d_init: [DPB_malloc] g_framecnt=1755, com_flag=0, mark_idx=4, slot_idx=4, slice_type=0, struct=3, lay_id=0
h264d_init: [DPB_INFO] cur_slot_idx=4
[... 185 lines cut ...]
h264d_parse: nalu_type=SLICE.
h264d_sps: [DPB_size] dpb_size[0]=6, mvc_flag=0, dpb_size[1]=0
h264d_slice: [SLICE_HEAD] type=0, layer_id=0,sps_id=0, pps_id=0, struct=3, frame_num=6
h264d_parse: SliceSTATE_ParseNalu
h264d_init: [DPB_malloc] g_framecnt=1758, com_flag=0, mark_idx=6, slot_idx=6, slice_type=0, struct=3, lay_id=0
h264d_init: [DPB_INFO] cur_slot_idx=6
h264d_init: [DPB_INFO] ref_slot_idx[0]=5
h264d_init: [DPB_MARK_INFO] slot_idx=0, top_used=1, bot_used=1, out_flag=1, fd=0xb4
h264d_init: [DPB_MARK_INFO] slot_idx=1, top_used=1, bot_used=1, out_flag=1, fd=0xb3
h264d_init: [DPB_MARK_INFO] slot_idx=2, top_used=1, bot_used=1, out_flag=1, fd=0xb5
h264d_init: [DPB_MARK_INFO] slot_idx=3, top_used=1, bot_used=1, out_flag=1, fd=0xb6
h264d_init: [DPB_MARK_INFO] slot_idx=4, top_used=1, bot_used=1, out_flag=1, fd=0xb7
h264d_init: [DPB_MARK_INFO] slot_idx=5, top_used=1, bot_used=1, out_flag=1, fd=0xb8
h264d_init: [DPB_MARK_INFO] slot_idx=6, top_used=1, bot_used=1, out_flag=1, fd=0xff
h264d_init: [DPB_MARK_INFO] ---------- cur_slot=6 --------------------
h264d_init: [DPB_REF_ERR] slot_idx=5, dpb_err[0]=0
h264d_init: list0 dpb: cur_err_flag=0, pps_refs=1, over_flag=0, num_ref_l0=1
h264d_parse: SliceSTATE_InitPicture
h264d_parse: SliceSTATE_GetSliceData
h264d_parse: SliceSTATE_RegisterOneFrame
h264d_dpb: [DPB_size] p_Dpb->used_size=6
h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=158, pkt_no=9
h264d_api: [CALLBACK] g_no=9, out_idx=7, dpberr=0, harderr=0, ref_flag=1, errinfo=0, discard=0
h264d_api: [CALLBACK] g_no=1759, out_idx=6, dpberr=0, harderr=0, ref_flag=1, errinfo=0, discard=0
Segmentation fault

HermanChen, 2018-11-26:
没有版本信息，这样的log看不出问题来，只能依赖你们用 gdb 或者其他工作跟一下了。

closed 2019-02-18 by HermanChen