rockchip-linux/mpp #72: 源码中有块代码疑似存在bug

giwhy, 2018-11-23:
你好，
在 h264d_dpb.c   ，direct_output 这个函数里面
    if (p->structure == FRAME) {
        //!< we have a frame (or complementary field pair), so output it directly
        FUN_CHECK(ret = flush_direct_output(p_Vid));
        write_picture(p, p_Vid);
	        
        free_storable_picture(p_Vid->p_Dec, p);
        p_Dpb->last_picture = NULL;
        p_Vid->out_buffer.is_used = 0;
        p_Vid->out_buffer.is_directout = 0;
		p_Dpb->last_output_poc = p->poc;

        goto __RETURN;
    }

这里p_Dpb->last_output_poc 获取到的值 是错乱的，想必 p->poc 在free_storable_picture(p_Vid->p_Dec, p); 之后被释放了？   我不清楚设计上是这样的还是 写错了  ，请确认一下。

WainDing, 2018-11-23:
这个是设计用的，记录每一次最后的输出p->poc。

giwhy, 2018-11-26:
好的

closed 2018-11-26 by giwhy