rockchip-linux/mpp #73: H265硬编码已经彻底支持了吗？

zhliji2, 2018-11-26:
请问已经支持H265硬编码了吗？

HermanChen, 2018-11-26:
已经支持，参见提交：3e5ad4fed304bcce11d11058accf4f716ca7f076

zhliji2, 2018-11-26:
RK3399是硬解？还是软解？

HermanChen, 2018-12-05:
H265 硬编码目前只存在于 RK3328 芯片上，RK3399 上没有 H265 硬编码。

HermanChen, 2018-12-05:
解码的话，都是硬解码

boris0604, 2018-12-06:
我再ROC-3328-CC上執行HEVC編碼測試，出現以下錯誤。Kernel的部份有需要調適嗎？

mpp_device: mpp_device_init failed to open device /dev/h265e, errno 12, error msg: Cannot allocate memory
mpi_enc_test: mpi_enc_test bps 3456000 fps 30 gop 60
mpp_device: mpp_device_send_reg_with_id ioctl VPU_IOC_WRITE failed ret -1 errno 9 Bad file descriptor
hal_h265e_vepu22: vepu22_set_cfg error: hal_h265e_set_param 0x9 fail
mpp: command 320007 param 0xab3d146c ret -1
mpi_enc_test: mpi control enc set codec cfg failed ret -1
mpi_enc_test: mpi_enc_test test mpp setup failed ret -1

danog, 2019-02-06:
Hello, I have encountered a similar error while trying the mpi_enc_test executable with h265, mind suggesting a solution?

HermanChen, 2019-02-12:
H.265 encoding is only supported on RK3328H Android platform. Linux platform is not formally supported yet.

jbloomfield2008, 2019-05-01:
Is Linux support planned and if so, is there a timeline when support will be added?

HermanChen, 2019-05-07:
The Linux kernel is the same to Andoird kernel. But RK3328 Linux kernel is not formally released.
The mpp part is the same.

jbloomfield2008, 2019-05-08:
Understood. I dropped the monet.bin firmware into the Linux kernel, but it still threw errors, so will have to wait until the Linux kernel is formally released.

closed 2023-08-03 by HermanChen