rockchip-linux/mpp #79: 用mpi_dec_test解码jpg图片失败

niujiabenben, 2019-01-07:
我想用mpp解码单个jpg图片, 操作过程为:
git clone -b release https://github.com/rockchip-linux/mpp.git
cmake -DRKPLATFORM=ON -DHAVE_DRM=ON && make
./test/mpi_dec_test -i test.jpg -w 1280 -h 720 -t 8

报错:
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: test.jpg
mpi_dec_test: output file name:
mpi_dec_test: config file name:
mpi_dec_test: width      : 1280
mpi_dec_test: height     :  720
mpi_dec_test: type       : 8
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 290872
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_dec_test: mpi_dec_test decoder test start w 1280 h 720 type 8
mpi: mpp version: 10c9a2d author: Herman Chen [jpegd]: Add vpu jpegd stream patch
mpp_log: can not found match soc name: rockchip,android 9tripod,x3399-development-board rockchip,rk339
mpp_dec: mpp_dec_advanced_thread slot size 1843200 is not equal to buffer size 3686400
mpp_dec: Assertion slot_size == buffer_size failed at mpp_dec_advanced_thread:1086

我所用的系统: 
Linux X3399 4.4.83 #7 SMP PREEMPT Wed Nov 21 17:49:40 CST 2018 aarch64 aarch64 aarch64 GNU/Linux

HermanChen, 2019-01-07:
麻烦提供一下测试源

HermanChen, 2019-01-09:
在命令行指定下 -f 4 的输出格式可以正常，具体问题还在处理中。

HermanChen, 2019-01-09:
[tmp.zip](https://github.com/rockchip-linux/mpp/files/2739269/tmp.zip)
试一下这个补丁，如果还有问题的话提供下log，可能是内核相关的问题

HermanChen, 2019-01-09:
加了 patch 之后，在新版本内核上解码正常，估计是内核版本问题

eddiecailinux, 2019-01-09:
麻烦更新最新的kernel测试

niujiabenben, 2019-01-12:
目前我们用的板子是定制的, 更新kernel有一些麻烦. 在当前的系统下, 解码jpeg图片还是有问题.

我往前找了几个commit, 比如 9月5号的这个commit: 15e357ffd0c98e3099be81eb91b4a48cf8db78a7 
没有上述的 "Assertion slot_size == buffer_size failed " 问题了, 但是程序卡在下面的语句出不去了:
   ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);

请问您能提供一个简单的在我们当前系统下能运行的demo么? 或者给一些建议我试着自己实现以下?
十分感谢!

niujiabenben, 2019-01-16:
还是不行, 和上面一样卡在这一句出不去了:
ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);

HermanChen, 2019-01-16:
我在手上的 3228 平台（和 3399 一样的 vpu硬件）上做测试，会出现解码花的情况，但是不会卡死，现象有不同啊

HermanChen, 2019-01-16:
export jpegd_debug=255，然后再跑一下看看

niujiabenben, 2019-01-17:
如果您愿意的话, 我们可以寄一块板子过去给您测试.

HermanChen, 2019-01-18:
产品支持最好走公司的 redmine 来提交问题单，在 github 不合适做正规的客户支持

alanlilyan, 2019-01-18:
mpp解码jpeg图像，结果提示可能是png图像。请问这个是什么问题？

:~/mpp$ ./test/mpi_dec_test -i ./1080.jpg -t 8 -w 1920 -h 1080
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: ./1080.jpg
mpi_dec_test: output file name:
mpi_dec_test: config file name:
mpi_dec_test: width      : 1920
mpi_dec_test: height     : 1080
mpi_dec_test: type       : 8
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 340928
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_dec_test: mpi_dec_test decoder test start w 1920 h 1080 type 8
mpi: mpp version: 930a2737 author: Ding Wei [h264d]: fix bug: fast mode buffer malloc
**jpegd_parser: input img maybe png format,check it
jpegd_parser: input img maybe png format,check it**
mpp_dec: mpp_dec_advanced_thread slot size 4177920 is not equal to buffer size 8355840
mpp_dec: Assertion slot_size == buffer_size failed at mpp_dec_advanced_thread:1086
mpi_dec_test: decoded frame 1
mpi_dec_test: found last packet
mpi_dec_test: decoded frame 2
mpi_dec_test: found eos frame
mpi_dec_test: test success max memory 0.00 MB

HermanChen, 2019-01-18:
只是 warning，没事，有客户把 png 当 jpeg 送进来过，于是加了这个打印

alanlilyan, 2019-01-18:
但是，后面size大小 assert信息，包括 test success max memory 0.00 MB也是正常的嘛？

HermanChen, 2019-01-18:
正常，jpeg 解码使用的是 mpp 外部提供的 buffer，mpp 内部本身不分配新 buffer，所以使用量为 0

alanlilyan, 2019-01-18:
额、我再结文档看看代码，mpp buffer的三种管理模式不是清楚。谢谢

HermanChen, 2019-01-21:
[MPP 开发参考_v0.3.pdf](https://github.com/rockchip-linux/mpp/files/2777342/MPP._v0.3.pdf)
文档供参考

yuanouwen, 2019-02-21:
我看mpp_dec_test.cpp测试对MJPEG的解码单独做了处理，这个原因是啥，还有我用此测试程序解码MJPEG,发现耗时达60ms，720p像素，请解答下

HermanChen, 2019-02-21:
mjpeg 解码是单次解码，buffer 是外部应用提供的，与一般的解码器不同
你看到的时间估计包括了码流拷贝，buffer 分配等的软件时间，硬件时间会比较短，多次解码情况下，流水处理时间会短

yuanouwen, 2019-03-08:
发现 ####mpi_dec time7 将近40ms。####sencond memcpy time 为20ms。所以导致硬解差不多60ms。请帮忙分析下。代码如下：
static int decode_advanced(MpiDecLoopData *data,void* imageData,int imageLen,void* returnData,int* returnLen)
 {
     RK_U32 pkt_eos  = 0;
     MPP_RET ret = MPP_OK;
     MppCtx ctx  = data->ctx;
     MppApi *mpi = data->mpi;
     char   *buf = data->buf;
     MppPacket packet = data->packet;
     MppFrame  frame  = data->frame;
     MppTask task = NULL;
#if 0
    FILE *fpp=fopen("/home/firefly/fw/color3.mjpeg", "rb");
    fprintf(stderr,"###start....data->packet_size is %d\n",data->packet_size);
    size_t read_size = fread(buf, 1, data->packet_size, fpp);
    fprintf(stderr,"####encode1 color image [%0x] [%0x] [%0x] [%0x] [%0x] [%0x]\n",*buf,*(buf+1),*(buf+2),*(buf+3),*(buf+4),*(buf+5));

     if (read_size != data->packet_size || feof(fpp)) {
         mpp_log("found last packet\n");
 
         // setup eos flag
         data->eos = pkt_eos = 1;
     }
 #else
    size_t read_size = 1843789;
    data->eos = pkt_eos = 1;
    pkt_eos = 1;
    /*if (read_size != data->packet_size ) {
         mpp_log("found last packet\n");
 
         // setup eos flag
         data->eos = pkt_eos = 1;
     }*/
 #endif
#if 1
     unsigned long long ts = ts_ms();
     memcpy(buf,imageData,imageLen);
     fprintf(stderr, "####first memcpy time is %d\n", (int)(ts_ms()-ts));
     #if 0
     fprintf(stderr,"####encode2 color image [%0x] [%0x] [%0x] [%0x] [%0x] [%0x]\n",*buf,*(buf+1),*(buf+2),*(buf+3),*(buf+4),*(buf+5));
     #endif
     // reset pos
     ts = ts_ms();
     mpp_packet_set_pos(packet, buf);
     mpp_packet_set_length(packet, imageLen);
     fprintf(stderr, "####mpi_dec time1 is %d\n", (int)(ts_ms()-ts));
     // setup eos flag
     if (pkt_eos)
         mpp_packet_set_eos(packet);
#endif

            ts = ts_ms();
         ret = mpi->poll(ctx, MPP_PORT_INPUT, MPP_POLL_BLOCK);
         if (ret) {
             mpp_err("mpp input poll failed\n");
             return ret;
         }
         fprintf(stderr, "####mpi_dec time2 is %d\n", (int)(ts_ms()-ts));
         
     

     ts = ts_ms();
     ret = mpi->dequeue(ctx, MPP_PORT_INPUT, &task);  /* input queue */
     if (ret) {
         mpp_err("mpp task input dequeue failed\n");
         return ret;
     }
     fprintf(stderr, "####mpi_dec time3 is %d\n", (int)(ts_ms()-ts));

     ts = ts_ms();
     mpp_assert(task);
     fprintf(stderr, "####mpi_dec time4 is %d\n", (int)(ts_ms()-ts));

     ts = ts_ms();
     mpp_task_meta_set_packet(task, KEY_INPUT_PACKET, packet);
     mpp_task_meta_set_frame (task, KEY_OUTPUT_FRAME,  frame);
     fprintf(stderr, "####mpi_dec time5 is %d\n", (int)(ts_ms()-ts));

         ts = ts_ms();
     ret = mpi->enqueue(ctx, MPP_PORT_INPUT, task);  /* input queue */
     if (ret) {
         mpp_err("mpp task input enqueue failed\n");
         return ret;
     }
     fprintf(stderr, "####mpi_dec time6 is %d\n", (int)(ts_ms()-ts));
     
     

        /* poll and wait here */
         ts = ts_ms();
         ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);
         if (ret) {
             mpp_err("mpp output poll failed\n");
             return ret;
         }
         fprintf(stderr, "####mpi_dec time7 is %d\n", (int)(ts_ms()-ts));

         ts = ts_ms();
         ret = mpi->dequeue(ctx, MPP_PORT_OUTPUT, &task); /* output queue */
         if (ret) {
             mpp_err("mpp task output dequeue failed\n");
             return ret;
         }
         fprintf(stderr, "####mpi_dec time8 is %d\n", (int)(ts_ms()-ts));

     
     
     mpp_assert(task);
 
     if (task) {
         MppFrame frame_out = NULL;
         mpp_task_meta_get_frame(task, KEY_OUTPUT_FRAME, &frame_out);
         //mpp_assert(packet_out == packet);
 
         if (frame) {
             /* write frame to file here */
             MppBuffer buf_out = mpp_frame_get_buffer(frame_out);
 
             if (buf_out) {
                 void *ptr = mpp_buffer_get_ptr(buf_out);
                 size_t len  = mpp_buffer_get_size(buf_out);
                 ts = ts_ms();
                 memcpy(returnData,ptr,len);
                 fprintf(stderr, "####sencond memcpy time is %d\n", (int)(ts_ms()-ts));
                 //returnData = ptr;
                 *returnLen = len;
#if 0
                 if (data->fp_output)
                     fwrite(ptr, 1, len, data->fp_output);
#endif
                #if 0
                mpp_log("####decoded frame %d size %d\n", data->frame_count, len);
                #endif
             }
             
             if (mpp_frame_get_eos(frame_out))
                 mpp_log("found eos frame\n");
             
         }
 
         /* output queue */
         
            #if 1
         ret = mpi->enqueue(ctx, MPP_PORT_OUTPUT, task);
         if (ret)
             mpp_err("mpp task output enqueue failed\n");
         #endif
         
         //firstrun = 0;
     }
 
     return ret;
 }

MinZhaoLin, 2019-05-31:
@HermanChen 你好，请问如何做流水处理？简单地加一个while循环就可以了？我需要进行解码再编码，做了流水后，发现解码和编码的耗时都增加了，难道不能同时进行解码和编码吗？还是需要注意哪里代码的使用？

lwdddddddd, 2019-08-03:
你好，我在rk3288平台4.4.55内核上面运行最新的mpp解码，也会遇到上面那个问题，卡在以下部分：
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_gen_regs enter
HAL_JPEGD_COMMON: jpegd_setup_output_fmt enter
HAL_JPEGD_COMMON: jpegd_setup_output_fmt exit
HAL_JPEGD_VDPU1: jpegd_gen_regs enter
HAL_JPEGD_VDPU1: jpegd_write_code_word_number enter
HAL_JPEGD_VDPU1: jpegd_write_code_word_number exit
HAL_JPEGD_COMMON: jpegd_write_qp_ac_dc_table enter
HAL_JPEGD_COMMON: jpegd_write_qp_ac_dc_table exit
HAL_JPEGD_VDPU1: jpegd_set_chroma_table_id enter
HAL_JPEGD_VDPU1: jpegd_set_chroma_table_id exit
HAL_JPEGD_VDPU1: jpegd_set_stream_offset enter
HAL_JPEGD_VDPU1: jpegd_set_stream_offset exit
HAL_JPEGD_VDPU1: jpegd_setup_pp enter
HAL_JPEGD_VDPU1: output_frame_fd:0, reg14:7f800000
HAL_JPEGD_VDPU1: jpegd_setup_pp exit
HAL_JPEGD_VDPU1: jpegd_gen_regs exit
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_gen_regs exit
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_start enter
而我之前在目前最新版本内核4.4.154上面运行mpp解码则会报如下错误：
mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_start mpp_device_send_reg Failed!!!
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_wait enter
mpp_device: mpp_device_wait_reg ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
HAL_JPEGD_VDPU1: frame_00 output YUV(1920*1088) saving to /tmp/output00.yuv
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_wait exit
error: fl_hw_decode failed.
mpp_task_impl: mpp_task_queue_deinit idx 0 task 0x7fd5d440 status 3 meta size 0
mpp_task_impl: Assertion p->tasks[i].status == MPP_INPUT_PORT || p->tasks[i].status == MPP_INPUT_HOLD failed at mpp_task_queue_deinit:445
jpegd_parser: jpegd_deinit enter
jpegd_parser: jpegd_deinit exit
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_deinit enter
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_deinit exit
请问你们有测试过的可以用当前版本mpp正常解码的内核版本有哪些？

HermanChen, 2019-08-05:
> 
> 
> @HermanChen 你好，请问如何做流水处理？简单地加一个while循环就可以了？我需要进行解码再编码，做了流水后，发现解码和编码的耗时都增加了，难道不能同时进行解码和编码吗？还是需要注意哪里代码的使用？

流水是说做多线程处理。不过，jpeg 解码和编码使用的的同一个硬件，多线程处理没什么收益。

HermanChen, 2019-08-05:
> 你好，我在rk3288平台4.4.55内核上面运行最新的mpp解码，也会遇到上面那个问题，卡在以下部分：
> HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_gen_regs enter
> HAL_JPEGD_COMMON: jpegd_setup_output_fmt enter
> HAL_JPEGD_COMMON: jpegd_setup_output_fmt exit [...]

遇到这种情况，先运行下 mpp_platform_test 和 mpp_runtime_test 看一下。

lwdddddddd, 2019-08-05:
> 遇到这种情况，先运行下 mpp_platform_test 和 mpp_runtime_test 看一下。

我在4.4.55和4.4.154上面都运行了，结果是正常的，下面是打印输出：
root@linaro-alip:/home/vvwork/mpp/mpp-release/osal/test# ./mpp_platform_test 
mpp_plat_test: chip name: rockchip,rk3288w
mpp_plat_test: 
mpp_plat_test: chip vcodec type 00000101
mpp_plat_test: found vpu1 codec
mpp_plat_test: found RK hevc decoder
mpp_plat_test: 
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder: /dev/vpu-service
mpp_plat_test: H.265 decoder: /dev/hevc-service
mpp_plat_test: MJPEG decoder: /dev/vpu-service
mpp_plat_test: VP9   decoder: /dev/vpu-service
mpp_plat_test: avs   decoder: /dev/vpu-service
mpp_plat_test: 
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder: /dev/vpu-service
mpp_plat_test: H.265 encoder: /dev/vpu-service
mpp_plat_test: MJPEG encoder: /dev/vpu-service
mpp_plat_test: mpp platform test done
root@linaro-alip:/home/vvwork/mpp/mpp-release/osal/test# ./mpp_runtime_test 
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpp_rt_test: mpp found ion buffer is invalid
mpp_rt_test: mpp found drm buffer is valid

lwdddddddd, 2019-08-05:
@HermanChen 你好，我安装了一下firefly-rk3288最新的ubuntu16.04镜像，内核版本也是4.4.154。我发现系统自带的mpp是可以正常解码的，不会报如上VPU错误，另外循环解码也没有问题。系统自带的mpp版本是：mpi: mpp version: 10c9a2d6 author: Herman Chen [jpegd]: Add vpu jpegd stream patch。而会报错的版本是：mpi: mpp version: Without VCS info，这个是我下载的最新的mpp编译出来的。

HermanChen, 2019-08-05:
Without VCS info 这个说明没有版本管理信息，可能是非标版本？

HermanChen, 2019-08-05:
下最新 mpp 的话，应该是会带有 git 信息的

lwdddddddd, 2019-08-05:
> 下最新 mpp 的话，应该是会带有 git 信息的

我重新下载编译了一下，还是会报错，版本信息是：mpi: mpp version: 3d353987 author: Johnson Ding [jpege]: Fix jpeg encoder stride problem

lwdddddddd, 2019-08-06:
@HermanChen 还能提供一下“10c9a2d”这个版本或其他版本的链接吗？因为buffer拷贝速度慢的问题需要修改一下源码。

Archer112, 2020-08-26:
> 下最新mpp的话，应该是会带有git信息的

你好 我使用这个demo 做循环解码jpg图片 运行不到一分钟出现了这个错误 请问是什么原因
pi_dec_test start
input file size 960000
mpi_dec_test decoder test start w 1280 h 720 type 8
mpi: mpp version: 0b701fe author: Lawrence-Tang DEBIAN: prepare for release_20171218-3
mpp_device: mpp_device_init failed to open device /dev/vpu_service, errno 24, error msg: Too many open files
HAL_JPEG_VDPU2: hal_jpegd_vdpu2_init get vpu_socket(-1) <= 0, failed.
mpp_hal: mpp_hal_init hal jpegd init failed ret -2
mpp_hal: mpp_hal_init could not found coding type 8
mpp_dec: mpp_dec_init could not init hal
mpp: error found on mpp initialization
mpp_dec: mpp_dec_control found NULL input dec (nil)
mpp: command 31000a param 0x7fce15fc04 ret -3
mpp input poll failed
found last packet
mpp input poll failed
mpi->reset failed