rockchip-linux/mpp #82: 根据mpi_dec_test demo修改替换mpp task方案异常

alanlilyan, 2019-01-19:
解码jpeg文件，根据test目录下mpi_dec_test.c源码修改。将decode_advanced()函数改成自己的do_decode，运行之后在packet deinit或者 frame deinit时，都会报错。请问如何正确使用decode_put_packet和decode_get_frame()??

~/mpp$ ./test/mpi_my_dec_test
== file size is 340928
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi: mpp version: 930a2737 author: Ding Wei [h264d]: fix bug: fast mode buffer malloc
read_size is 340928
mpi_my_dec_test: found last packet
pkt_eos is 1
decode get frame deinit 44444
== alan: mpp decode time  1.37 ms ==
mpp_packet: check_is_mpp_packet pointer (nil) failed on check
段错误

源码文件：
[mpi_my_dec_test.c](https://github.com/rockchip-linux/mpp/files/2775342/mpi_my_dec_test.txt)

```c
static MPP_RET do_decode(MppDecData *data)
{
    RK_U32 pkt_eos  = 0;
    MPP_RET ret = MPP_OK;
    MppCtx ctx  = data->ctx;
    MppApi *mpi = data->mpi;
    char   *buf = data->buf;
    MppPacket packet = data->packet;
    MppFrame  frame  = data->frame;

    size_t read_size = fread(buf, 1, data->packet_size, data->fp_input);

    printf("read_size is %d\n", read_size);
    if (read_size == data->packet_size || feof(data->fp_input)) {
        mpp_log("found last packet\n");

        // setup eos flag
        data->eos = pkt_eos = 1;
    }
    printf("pkt_eos is %d\n", pkt_eos);

    struct timeval st,et;
    gettimeofday(&st, NULL);
    // reset pos
    mpp_packet_set_pos(packet, buf);
    mpp_packet_set_length(packet, read_size);
    // setup eos flag
    if (pkt_eos) {
        mpp_packet_set_eos(packet);
    }

        ret = mpi->decode_put_packet(ctx, packet);
        if (ret == MPP_OK) {
            ret = mpi->decode_get_frame(ctx, &frame);
            if (MPP_ERR_TIMEOUT == ret) {
                mpp_err("decode_get_frame failed too much time\n");
            }

            if (MPP_OK != ret) {
                mpp_err("decode_get_frame failed ret %d\n", ret);
            } else {
#if 0
	    printf("decode get frame eos 3-1\n");
            if (mpp_frame_get_eos(frame)) {
                mpp_log("frame get eos\n");
            }
	    printf("decode get frame eos 3-2\n");
#endif
            //mpp_frame_deinit(&frame);
	    printf("decode get frame deinit 44444\n");
	    }
        }

    gettimeofday(&et, NULL);
    printf("== alan: mpp decode time %5.2f ms ==\n", ((double)((et.tv_sec - st.tv_sec) * 1000000 + (et.tv_usec - st.tv_usec)) / 1000));
    return ret;
}
```

HermanChen, 2019-01-21:
请看一下 block 格式的配置，估计是 non-block 模式，get_frame 出来的 frame 为空

HermanChen, 2019-01-21:
目前 jpeg 解码需要使用 task 路径，为的是指定输出 buffer 的要求，put_packet / get_frame 接口验证不是很充分。

alanlilyan, 2019-01-21:
设置block模式，测试发现卡在 Mpp::get_frame() { mFramew->wait() }. 里面一直要等待pthread cond。
因为直接用Demo中MppTask解码1920x1080 RGB格式，单纯计算task 解码耗时要25-30ms。算上内存拷贝的损失。现在硬解方案似乎并不比普通libjpeg快多少。除了用vpu也没有发现其他jpeg硬解加速的方案。
请问jpeg解码现在这种速度是正常的嘛？

HermanChen, 2019-01-21:
小图像用mpp没什么优势，1080p 这样的大图的硬件解码时间会比较有优势，但是最好是不要有 cpu 访问拷贝，这样效率很低。硬件的标称能力是 90M pixel 每秒，1080p 能解 45 帧，这样 20ms 左右正常。
mpp 里有码流拷贝，内存分配等额外的时候，单帧解码时间长是正常的。
vpu 硬件主要是用于解 camera 进来的数据，解了之后去硬件录像或拍照会比较有优势，单帧使用的话，额外开销不好平摊。

lwdddddd, 2019-07-24:
你好，目前MJPEG一定要用task模式吗？

HermanChen, 2019-07-24:
目前是

lwdddddd, 2019-07-24:
那请问mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_NON_BLOCK)需要耗时18ms正常吗？

lwdddddd, 2019-07-24:
那请问mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK)需要耗时18ms正常吗？

HermanChen, 2019-07-24:
正常，解码时间差不多