rockchip-linux/mpp #83: 有rga相关开发文档吗？

fuckdk, 2019-01-22:
我按照范例rga_test.cpp代码加到mpp解码取帧后面，试图将NV12转RGB，但是调用rga_init(&ctx);就失败了：
mpp_log: rga_init open device failed
init rga context failed -5
请问这是什么错误？调用rga有啥说明文档吗？

fuckdk, 2019-01-22:
#define DEFAULT_RGA_DEV     "/dev/rga"
我在我的rk3399 ubuntu16.04上没有这个dev，是需要安装什么吗

lihuang1111, 2019-01-22:
有完整log吗。可以先看下dev/目录下是否有rga设备节点。

mpp内调用rga的说明文档我这边暂时没有，现有文档只针对librga的独立模块的。

fuckdk, 2019-01-22:
> 有完整log吗。可以先看下dev/目录下是否有rga设备节点。
> 
> mpp内调用rga的说明文档我这边暂时没有，现有文档只针对librga的独立模块的。

ls /dev/
aer_inject       loop4               rfkill  tty26  tty52    usbmon5
block            loop5               rkvdec  tty27  tty53    usbmon6
bus              loop6               rtc     tty28  tty54    v4l
char             loop7               rtc0    tty29  tty55    vcs
console          mali0               shm     tty3   tty56    vcs1
cpu_dma_latency  mapper              snd     tty30  tty57    vcs2
disk             mem                 stderr  tty31  tty58    vcs3
dri              memory_bandwidth    stdin   tty32  tty59    vcs4
fb0              mmcblk1             stdout  tty33  tty6     vcs5
fd               mmcblk1boot0        tty     tty34  tty60    vcs6
full             mmcblk1boot1        tty0    tty35  tty61    vcs7
fuse             mmcblk1p1           tty1    tty36  tty62    vcsa
hdmi_hdcp1x      mmcblk1p2           tty10   tty37  tty63    vcsa1
hwrng            mmcblk1p3           tty11   tty38  tty7     vcsa2
i2c-0            mmcblk1p4           tty12   tty39  tty8     vcsa3
i2c-1            mmcblk1p5           tty13   tty4   tty9     vcsa4
i2c-3            mmcblk1p6           tty14   tty40  ttyFIQ0  vcsa5
i2c-4            mmcblk1rpmb         tty15   tty41  ttyS0    vcsa6
i2c-9            mqueue              tty16   tty42  ttyS1    vcsa7
iio:device0      net                 tty17   tty43  ttyS2    vendor_storage
initctl          network_latency     tty18   tty44  ttyS3    vhci
input            network_throughput  tty19   tty45  uhid     video0
kmsg             null                tty2    tty46  uinput   vmdrm0
log              port                tty20   tty47  urandom  vpu_service
loop-control     ppp                 tty21   tty48  usbmon0  watchdog
loop0            ptmx                tty22   tty49  usbmon1  zero
loop1            pts                 tty23   tty5   usbmon2  zram0
loop2            ram0                tty24   tty50  usbmon3
loop3            random              tty25   tty51  usbmon4
---------
请问这个vpu_service能当rga的dev用吗

lihuang1111, 2019-01-22:
不行。你那能否编译kernel，可以的话我这边提供编译rga设备的方法。