rockchip-linux/mpp #84: RK3399 RGB888硬编码JPEG硬件错误

jinke5245, 2019-02-14:
jpege_api: jpege: hardware return error status 40

HermanChen, 2019-02-18:
请提供下编码命令或者使用代码的情况，以及完整错误 log

jinke5245, 2019-02-21:
修改了 `read_yuv_image` 方法读取 RGB888 到 buffer

```bash
$ ./mpi_enc_test -i ~/Workspace/resources/images/1080.rgb -w 1920 -h 1080 -t 8 -f 65542
```

```text
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: /home/firefly/Workspace/resources/images/1080.rgb
mpi_enc_test: output file name: 
mpi_enc_test: width      : 1920
mpi_enc_test: height     : 1080
mpi_enc_test: format     : 65542
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 1920 h 1080 type 8
mpi: mpp version: Missing VCS info
mpi_enc_test: mpi_enc_test bps 7776000 fps 30 gop 60
jpege_api: jpege: hardware return error status 40
mpi_enc_test: test_mpp_run encoded frame 0 size 259738
mpi_enc_test: test_mpp_run encode max 1 frames
mpi_enc_test: mpi_enc_test success total frame 1 bps 62337120
mpp_buffer: ~MppBufferService cleaning misc group
```

rageworx, 2019-03-06:
I have same problem with YUV ( MPP_FMT_YUV420P ) .
Here's log:
```
 ✘ linaro@linaro-alip  ~/Projects/mpp/build/test   release ● 
>  ./mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -i yuv/conv_01.yuv -o test.jpg -n 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: yuv/conv_01.yuv
mpi_enc_test: output file name: test.jpg
mpi_enc_test: width      : 2456
mpi_enc_test: height     : 2054
mpi_enc_test: format     : 4
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 0

mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp version: 2dc830f0 author: sliver.chen [vp8e]: add vpu1 and vpu2 vp8e supprt.
mpi_enc_test: mpi_enc_test bps 18917340 fps 30 gop 60
jpege_api: jpege: hardware return error status 40
mpi_enc_test: test_mpp_run encoded frame 0 size 631116
mpi_enc_test: test_mpp_run encode max 1 frames
mpi_enc_test: mpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
Any solution, or should I know what meaning of 40 ?

HermanChen, 2019-03-07:
# mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -o /sdcard/2456x2054.jpg
01-15 04:37:49.818 I/mpi_enc_test( 2937): cmd parse result:
01-15 04:37:49.819 I/mpi_enc_test( 2937): input  file name:
01-15 04:37:49.819 I/mpi_enc_test( 2937): output file name: /sdcard/2456x2054.jpg
01-15 04:37:49.819 I/mpi_enc_test( 2937): width      : 2456
01-15 04:37:49.819 I/mpi_enc_test( 2937): height     : 2054
01-15 04:37:49.819 I/mpi_enc_test( 2937): format     : 4
01-15 04:37:49.819 I/mpi_enc_test( 2937): type       : 8
01-15 04:37:49.819 I/mpi_enc_test( 2937): debug flag : 0
01-15 04:37:49.820 I/mpi_enc_test( 2937): mpi_enc_test start
01-15 04:37:49.821 I/mpp_rt  ( 2937): NOT found ion allocator
01-15 04:37:49.821 I/mpp_rt  ( 2937): found drm allocator
01-15 04:37:49.840 I/mpi_enc_test( 2937): mpi_enc_test encoder test start w 2456 h 2054 type 8
01-15 04:37:49.840 I/mpi     ( 2937): mpp version: 7337346 author: Herman Chen TESTONLY: ndk r16 and cmake 3.12 patch
01-15 04:37:49.841 I/mpi_enc_test( 2937): mpi_enc_test bps 18917340 fps 30 gop 60
01-15 04:37:49.929 I/mpi_enc_test( 2937): test_mpp_run encoded frame 0 size 1093942
01-15 04:37:49.929 I/mpi_enc_test( 2937): test_mpp_run encode max 1 frames
01-15 04:37:49.947 I/mpi_enc_test( 2937): mpi_enc_test success total frame 1 bps 262546080
01-15 04:37:49.948 I/mpp_buffer( 2937): ~MppBufferService cleaning misc group

在 3399 Android 9.0 上测试正常，可能内核有问题么？能否确认一下内核版本？

HermanChen, 2019-03-07:
可以试没有输入参数，使用默认数据的编码看看

jinke5245, 2019-03-07:
@HermanChen 我的问题可不可以给个解决方案 或者给我一个可以运行的demo(RGB888 -> JPEG)

HermanChen, 2019-03-07:
我就是用最新的 mpi_enc_test 做的测试，RGB888 应该不支持，手册上硬件支持的格式为：
YUV420P/YUV420SP/YUYV422/UYVY422/RGB565/RGB555/RGB44

rageworx, 2019-03-07:
Dear @HermanChen,
I'm using linaro aarch64, not Android.
Thanks !

HermanChen, 2019-03-08:
> 
> 
> Dear @HermanChen,
> I'm using linaro aarch64, not Android.
> Thanks !

Hi, aarch64 may have some different in kernel comparing to 32-bit version. But there should be no functional difference between them.
Can you provide vcodec_service.c in your kernel?

rageworx, 2019-03-08:
Dear @HermanChen 
I'm using Rock960b model from 96boards.
kernel is here : https://github.com/rockchip-linux/kernel
and you may refer to here:
https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/video/rockchip/vcodec/vcodec_service.c
Regards, Raph.