rockchip-linux/mpp #85: 请问mpp解码出来的buff有办法实现零拷贝到android的surfaceview上吗

fuckdk, 2019-02-16:
图像格式转换可以用rga，就是我的重点在零拷贝给surfaceview上显示（或者其他控件也可）

HermanChen, 2019-02-18:
可以的，在 RK 的 Android SDK 上就是这么做的，先从 native windows 申请 GraphicBuffer，再把 GraphicBuffer 里的 fd 信息 commit 给 MppBufferGroup，再把 MppBufferGroup 配置给 mpp 解码器即可。

fuckdk, 2019-02-18:
> 可以的，在 RK 的 Android SDK 上就是这么做的，先从 native windows 申请 GraphicBuffer，再把 GraphicBuffer 里的 fd 信息 commit 给 MppBufferGroup，再把 MppBufferGroup 配置给 mpp 解码器即可。

有sample可以参考吗

HermanChen, 2019-02-18:
https://github.com/c-xh/ 直接开这个去找一下
具体代码要看 RK 的 Android SDK 里 omx 实现，没有现成的 sample

fuckdk, 2019-02-18:
> https://github.com/c-xh/ 直接开这个去找一下
> 具体代码要看 RK 的 Android SDK 里 omx 实现，没有现成的 sample

RKMediaCodecDemo请问是这个吗？

fuckdk, 2019-02-18:
请问RK 的 Android SDK是公开的吗？是否要购买？

HermanChen, 2019-02-18:
公开，需要开发费，不过网上应该也能下载到

fuckdk, 2019-02-18:
> 公开，需要开发费，不过网上应该也能下载到

开发费是指技术支持吗？使用是否有限制？

HermanChen, 2019-02-19:
这些具体的我就不清楚了……需要找业务或者 FAE 问一下

fuckdk, 2019-02-27:
> 公开，需要开发费，不过网上应该也能下载到

如果是公开得话可否给我一个下载地址？

yuxiatongzhi, 2019-07-26:
> > 公开，需要开发费，不过网上应该也能下载到
> 
> 如果是公开得话可否给我一个下载地址？
> 发我邮箱也可以[epcdiy@qq.com](mailto:epcdiy@qq.com)
> 麻烦了

有找到吗？上面[https://github.com/c-xh/](https://github.com/c-xh/)的链接里都没有实现。