rockchip-linux/mpp #86: rk3288解码H264图片失败

lwdddddddd, 2019-02-20:
您好，请问在rk3288上面解码h264图片失败是什么原因？打印信息如下：

HermanChen, 2019-02-20:
估计码流有错

lwdddddddd, 2019-02-20:
我在rk3328上面也试了下，解码同一张图片是正常的，打印信息如下：
root@share:/home/vvwork/mpp/test# ./mpi_dec_test -i Tennis1080p.h264  -o 3 -w 1920 -h 1080 -t 7 -n 2
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: Tennis1080p.h264
mpi_dec_test: output file name: 3
mpi_dec_test: config file name: 
mpi_dec_test: width      : 1920
mpi_dec_test: height     : 1080
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 2
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 10786528
mpi_dec_test: mpi_dec_test decoder test start w 1920 h 1080 type 7
mpi: mpp version: 2dc830f author: sliver.chen [vp8e]: add vpu1 and vpu2 vp8e supprt.
mpp_rt: found ion allocator
mpp_rt: NOT found drm allocator
mpp_ion: vpu_service iommu_enabled 1
mpp_ion: using ion heap ION_HEAP_TYPE_SYSTEM
mpi_dec_test: decode_get_frame get info changed found
mpi_dec_test: decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
mpi_dec_test: decode_get_frame get frame 1
mpi_dec_test: decode_get_frame get frame 2
mpi_dec_test: reach max frame number 2
mpi_dec_test: test success max memory 7.97 MB

HermanChen, 2019-02-21:
内核驱动有不同，3288 和 3328 的硬件也有不同。特定的 264 码流 3288 处理起来要加额外的处理。

lwdddddd, 2019-07-29:
closed #86

closed 2019-07-29 by lwdddddddd