rockchip-linux/mpp #89: RK3399 JPEG encode failure with error status 40.

rageworx, 2019-03-06:
Now I am trying to make fastest JPEG image encoder base on your enc_test.
But it fails with these error from YUV image.
```
 linaro@linaro-alip  ~/Projects/mpp/build/test   release ● 
>  ./mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -i yuv/conv_01.yuv -o test.jpg -n 1 -d 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: yuv/conv_01.yuv
mpi_enc_test: output file name: test.jpg
mpi_enc_test: width      : 2456
mpi_enc_test: height     : 2054
mpi_enc_test: format     : 4
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 1

MPP_FMT_ARGB8888 = 65546
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp_create enter ctx 0x557c16c188 mpi 0x557c16c190
mpi: mpp version: 2dc830f0 author: sliver.chen [vp8e]: add vpu1 and vpu2 vp8e supprt.
mpi: mpp_create leave ret 0 ctx 0x557c16d5a0 mpi 0x7f9fb344c0
mpi: mpp_init enter ctx 0x557c16d5a0 type 1 coding 8
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x557c16d5a0 cmd 320003 parm 0x557c16c198
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 18917340 fps 30 gop 60
mpi: mpi_control enter ctx 0x557c16d5a0 cmd 320005 parm 0x557c16c1e4
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x557c16d5a0 cmd 320007 parm 0x557c16c21c
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x557c16d5a0 cmd 32000f parm 0x557c16ccb0
mpi: mpi_control leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x557c16d5a0 frame 0x557c172360
jpege_api: jpege: hardware return error status 40
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x557c16d5a0 packet 0x7fee3d7fa8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 0 size 631116
mpi_enc_test: test_mpp_run encode max 1 frames
mpi: mpi_reset enter ctx 0x557c16d5a0
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x557c16d5a0
mpi: mpp_destroy leave ret 0
mpi_enc_test: mpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
It generates a part of JPEG data strats with JPIF header, but cannot read all image because all right encodded datas not contained to output file by error of status 40.
Any instructions? or right way?
Regards, Raph.

sliver-chen, 2019-03-06:
just try like this.  ./mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -o test.jpg -n 1 -d 1

rageworx, 2019-03-08:
Dear @sliver-chen 
It's same with your dummy testing code.
```
 linaro@linaro-alip  ~/Projects/mpp/build/test   release 
>  ./mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -o test.jpg -n 1 -d 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name:
mpi_enc_test: output file name: test.jpg
mpi_enc_test: width      : 2456
mpi_enc_test: height     : 2054
mpi_enc_test: format     : 4
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 1
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp_create enter ctx 0x55a86061a8 mpi 0x55a86061b0
mpi: mpp version: 2dc830f0 author: sliver.chen [vp8e]: add vpu1 and vpu2 vp8e supprt.
mpi: mpp_create leave ret 0 ctx 0x55a8607340 mpi 0x7f92437720
mpi: mpp_init enter ctx 0x55a8607340 type 1 coding 8
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x55a8607340 cmd 320003 parm 0x55a86061b8
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 18917340 fps 30 gop 60
mpi: mpi_control enter ctx 0x55a8607340 cmd 320005 parm 0x55a8606204
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55a8607340 cmd 320007 parm 0x55a860623c
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55a8607340 cmd 32000f parm 0x55a8606cd0
mpi: mpi_control leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x55a8607340 frame 0x55a860b1a0
jpege_api: jpege: hardware return error status 40
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x55a8607340 packet 0x7ff70cca98
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 0 size 631116
mpi_enc_test: test_mpp_run encode max 1 frames
mpi: mpi_reset enter ctx 0x55a8607340
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x55a8607340
mpi: mpp_destroy leave ret 0
mpi_enc_test: mpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
Any solution to solve this issue ?
Regards, Raph.

HermanChen, 2019-03-08:
Is the output test.jpg correct or not?

rageworx, 2019-03-08:
Dear @HermanChen 
Dummy result opens in any graphic editor, but there's nothing, just 0x808080 contains.
I'm trying to use right source, and wanna know what is error code 40.
Should I can use RGB888 or YUV with single gray image ?
Let me know.
Regards, Raph.

HermanChen, 2019-03-09:
emm, code 40 is irq for timeout. It seems encoder has error on reading frame or writing stream.

HermanChen, 2019-03-09:
It does not happend on my Android platform. What is your kernel platform and verison?

rageworx, 2019-03-09:
Dear @HermanChen 
I'm not using Android, now on bebian-linux kernel 4.4 from 96board - rock960b model.
You can refer other issue on same problem as you know, it belong to same kernel source and same service.

qvoid, 2019-04-24:
@rageworx check out the lastest version, I think the commit da0f013bc6b422f6993f41d1e9db1e84b82c838c
should fix your problem.

rageworx, 2019-05-03:
Dear, latest commits still in error by "jpege_api: jpege: hardware return error status 40", and still result jpeg cannot read by any viewer because nothing contains inside file except JPEG tags.

Here's test log:
```
>  ./mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -i conv_01.yuv -o test.jpg -n 1 -d 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: conv_01.yuv
mpi_enc_test: output file name: test.jpg
mpi_enc_test: width      : 2456
mpi_enc_test: height     : 2054
mpi_enc_test: format     : 4
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 1
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp_create enter ctx 0x55c016d1a8 mpi 0x55c016d1b0
mpi: mpp version: da0f013b author: Johnson Ding [jpegd]: Fix vpu1 encoder problem
mpi: mpp_create leave ret 0 ctx 0x55c016e5c0 mpi 0x7f79c3f768
mpi: mpp_init enter ctx 0x55c016e5c0 type 1 coding 8
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x55c016e5c0 cmd 320003 parm 0x55c016d1b8
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 18917340 fps 30 gop 60
mpi: mpi_control enter ctx 0x55c016e5c0 cmd 320005 parm 0x55c016d204
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55c016e5c0 cmd 320007 parm 0x55c016d23c
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55c016e5c0 cmd 32000f parm 0x55c016dcd0
mpi: mpi_control leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x55c016e5c0 frame 0x55c0173420
jpege_api: jpege: hardware return error status 40
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x55c016e5c0 packet 0x7feb0cf8e8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 0 size 631116
mpi_enc_test: test_mpp_run encode max 1 frames
mpi: mpi_reset enter ctx 0x55c016e5c0
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x55c016e5c0
mpi: mpp_destroy leave ret 0
mpi_enc_test: mpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
and also, your non-source testing way ( ./mpi_enc_test -f 4 -w 2456 -h 2054 -t 8 -o test.jpg -n 1 -d 1 ) not works, same error result.

here a link contains source image and results with some info.
https://mega.nz/#!x59kHSaa!nurg8QOTONsWx6g_O9d31N5nRdXJqUuhNI7JEceFRsk

qvoid, 2019-05-08:
Are you sure that the "conv_01.yuv" file you provided is really a yuv file? I opened it in Vooya, but I saw a mess of dots, not an normal image. Beside, can you provide the kernel log when encoder failed with error code 40.
If you don't  have Vooya installed, you can open your yuv file on this [site](http://rawpixels.net/)

rageworx, 2019-06-19:
I am now came back to continue do testing this again, I will make a new code for testing image from other source ( not YUV ), can I test with RGB source to JPEG ? Anyone has example to encode JPEG (q10, less loss) encode from RGB array, or Y (gray) only?

qvoid, 2019-06-19:
RGB is supported. use *-f* option in *mpi_enc_test* to specify color format. See **MppFrameFormat** in *mpp_frame.h* for color format values. It seems that Y (gray) only is not support directly. You can have a try. If it is not working properly, try to convert grey scale to other color formats, for example, YUV420SP.

teor292, 2019-06-19:
I have the same issue (jpege_api: jpege: hardware return error status 40) on the same test example. I use ubuntu: "rockpi4b-ubuntu-bionic-minimal-20190104_2101-gpt". 
Kernel: "Linux linux 4.4.154-59-rockchip-g5e70f14 #4 SMP Fri Dec 14 20:55:41 CST 2018 aarch64 aarch64 aarch64 GNU/Linux".

rageworx, 2019-06-20:
Dear qvoid, 
I have tested trying encode RGB888 to JPEG, but encode_get_packet returns zero length.
```
 linaro@linaro-alip  ~/Projects/rkmpptest 
>  bin/rkmpptest imgs/grab_L\ 0.bmp
source : imgs/grab_L 0.bmp
output : imgs/grab_L 0.jpg
mpi_enc_test start
loading source ... size : 2456x2054 --> strided : 2464x2064
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp version: 66b140ee author: Johnson Ding [h264d]: Fix unused parameter warning
#Phase: MPP initialize.
#Phase: MPP set up.
mpi_enc_test bps 18917340 fps 30 gop 60
Configuring encoder ...
#Phase: MPP encoding.
-- mpp_frame_init
-- mpp_frame_set_xxx ...
-- encode_put_frame
-- encode_get_packet
encoded frame 0, but size is zero!
found last packet
encode max 1 framesmpi_enc_test success total frame 1 bps 0
mpp_buffer: ~MppBufferService cleaning misc group
```
I've modified mpi_enc_test to reads Windows BMP source, and encoding only for JPEG with FLTK for reading BMP.
Here what I did:
- Reads Windows bitmap as RGB888.
- Encoding source format set to MPP_FMT_RGB888.
- Calculating strides 
- Copy source bitmap arrays into strided size as like this:
```
106         width  = img->w();
107         height = img->h();
108         hor_stride = MPP_ALIGN( width, 16 );
109         ver_stride = MPP_ALIGN( height, 16 );
110
111         // Convert image to MPP_FMT_RGB888
112         unsigned imgsz = width * height;
113         unsigned bufsz = hor_stride * ver_stride;
114
115         buf = new RK_U8[ bufsz * 3 ];
116         if ( buf != NULL )
117         {
118             memset( buf, 0, bufsz );
119
120             RK_U8* rbuff = (RK_U8*)img->data()[0];
121
122             for( unsigned cnt=0; cnt<height; cnt++ )
123             {
124                 unsigned w_pos = ( cnt * hor_stride * 3 );
125                 unsigned r_pos = ( cnt * width * 3 );
126
127                 memcpy( &buf[w_pos], &rbuff[r_pos], width * 3 );
128             }
129         }
130
```
- codec_cfg set to only doing MPP_VIDEO_CodingMJPEG with codec_cfg->jpeg.quant = 10;
- codec_cfg, rc_cfg->quality set to MPP_ENC_RC_QUALITY_BEST.
- calling mpi->control() has no error returned both of MPP_ENC_SET_RC_CFG and MPP_ENC_SET_CODEC_CFG.
- works encode_get_packet() OK.
- all seems to OK but mpp_packet_get_length( packet ) returns zero.
Any requirements to be shared ?

rageworx, 2019-06-20:
Update,
I've been fixed a code mpp_frame_set_buffer() as mpp_frame_set_buffer( frame, p->frm_buf );
and now it returns some packet but problem is there's nothing inside except a header with error return 40.
```
>  bin/rkmpptest imgs/grab_L\ 0.bmp
source : imgs/grab_L 0.bmp
output : imgs/grab_L 0.jpg
mpi_enc_test start
loading source ... size : 2456x2054 --> strided : 2464x2064
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp version: 66b140ee author: Johnson Ding [h264d]: Fix unused parameter warning
#Phase: MPP initialize.
#Phase: MPP set up.
mpi_enc_test bps 18917340 fps 30 gop 60
Configuring encoder ...
#Phase: MPP encoding.
-- mpp_frame_init
-- mpp_frame_set_xxx ...
-- encode_put_frame
HAL_JPEGE_VDPU2: hal_jpege_vepu2_set_extra_info other format(65542)
jpege_api: jpege: hardware return error status 40
-- encode_get_packet
encoded frame 0 size 631116
encode max 1 framesmpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```

rageworx, 2019-06-20:
Dear, I can see an error message of "HAL_JPEGE_VDPU2: hal_jpege_vepu2_set_extra_info other format(65542)" in advanced code.
```
source : imgs/grab_L 0.bmp
output : imgs/grab_L 0.jpg
mpi_enc_test start
loading source ... size : 2456x2054 --> strided : 2464x2064
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp version: 66b140ee author: Johnson Ding [h264d]: Fix unused parameter warning
#Phase: MPP initialize.
#Phase: MPP set up.
mpi_enc_test bps 18917340 fps 30 gop 60
Configuring encoder ...
#Phase: MPP encoding.
-- mpp_frame_init
-- mpp_frame_set_xxx ...
-- encode_put_frame
HAL_JPEGE_VDPU2: hal_jpege_vepu2_set_extra_info other format(65542)
jpege_api: jpege: hardware return error status 40
-- encode_get_packet
encoded frame 0 size 631116
encode max 1 framesmpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
Is meant be "need to use other format" ?
Let me knows.

rageworx, 2019-06-20:
Now I've been tested YUV420SP ( for only Y channels, we don't use color source ) and result is same.
```
>  bin/rkmpptest imgs/grab_L\ 0.bmp
source : imgs/grab_L 0.bmp
output : imgs/grab_L 0.jpg
mpi_enc_test start
loading source ...
 write buff pos : 5058592 / 7628544 ....  size : 2456x2054 --> strided : 2464x2064
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp version: 66b140ee author: Johnson Ding [h264d]: Fix unused parameter warning
#Phase: MPP initialize.
#Phase: MPP set up.
mpi_enc_test bps 18917340 fps 30 gop 60
Configuring encoder ...
#Phase: MPP encoding.
-- mpp_frame_init
-- mpp_frame_set_xxx ...
-- encode_put_frame
jpege_api: jpege: hardware return error status 40
-- encode_get_packet
encoded frame 0 size 631116
encode max 1 framesmpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
result is still contains nothing, just zeroes except header.
Any idea to solve this issue ?
Regards, Raph.

rageworx, 2019-06-20:
Plus, I also tested my YUV420SP image on your test binary, and seems to exactly same.
```
>  mpi_enc_test -i testdump.yuv420sp -o test.jpg -w 2456 -h 2054 -f 0 -t 8 -n 1 -d 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: testdump.yuv420sp
mpi_enc_test: output file name: test.jpg
mpi_enc_test: width      : 2456
mpi_enc_test: height     : 2054
mpi_enc_test: format     : 0
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 1
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2456 h 2054 type 8
mpi: mpp_create enter ctx 0x559cb111a8 mpi 0x559cb111b0
mpi: mpp version: 66b140ee author: Johnson Ding [h264d]: Fix unused parameter warning
mpi: mpp_create leave ret 0 ctx 0x559cb125c0 mpi 0x7f8d7bd118
mpi: mpp_init enter ctx 0x559cb125c0 type 1 coding 8
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x559cb125c0 cmd 320003 parm 0x559cb111b8
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 18917340 fps 30 gop 60
mpi: mpi_control enter ctx 0x559cb125c0 cmd 320005 parm 0x559cb11204
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x559cb125c0 cmd 320007 parm 0x559cb1123c
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x559cb125c0 cmd 32000f parm 0x559cb11cd0
mpi: mpi_control leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x559cb125c0 frame 0x559cb174a0
jpege_api: jpege: hardware return error status 40
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x559cb125c0 packet 0x7fc3abc8f8
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 0 size 631116
mpi_enc_test: test_mpp_run encode max 1 frames
mpi: mpi_reset enter ctx 0x559cb125c0
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x559cb125c0
mpi: mpp_destroy leave ret 0
mpi_enc_test: mpi_enc_test success total frame 1 bps 151467840
mpp_buffer: ~MppBufferService cleaning misc group
```
And [here](https://mega.nz/#!FgMDSKhB!S-nu2hx264z-53bReGYNm8bMnRJy6ctOuispG3zlo1c) a converted file for YUV420SP.

rageworx, 2019-06-20:
And [here](https://mega.nz/#!x9lXzSxI!UhQUDRvGwZO6WD8Kb-ndIDl-It8RyTDYMuTaBrZ2pqo) is code and binary for aarch64-linaro.

rageworx, 2019-06-20:
Plus, dmesg log here when trying encode JPEG.
```
[92799.132341] rk_vcodec: vpu_service_ioctl:1875: error: unknown vpu service ioctl cmd 40086c01
[92799.141011] rk_iommu ff650800.iommu: Page fault at 0x0000000000007380 of type read
[92799.141694] rk_iommu ff650800.iommu: iova = 0x0000000000007380: dte_index: 0x0 pte_index: 0x7 page_offset: 0x380
[92799.142594] rk_iommu ff650800.iommu: mmu_dte_addr: 0x00000000eed9d000 dte@0x00000000eed9d000: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[92799.144062] rk-vcodec ff650000.vpu_service: vcodec, fault addr 0x00007380 status 4b
[92799.144732] rk-vcodec ff650000.vpu_service: no memory region mapped
[92799.145286] rk-vcodec ff650000.vpu_service: dumping enc register set:
[92799.145853] rk-vcodec ff650000.vpu_service: reg[00] 00000000
[92799.146351] rk-vcodec ff650000.vpu_service: reg[01] 00000000
[92799.146849] rk-vcodec ff650000.vpu_service: reg[02] 04050a0e
[92799.147347] rk-vcodec ff650000.vpu_service: reg[03] 04070d12
[92799.147845] rk-vcodec ff650000.vpu_service: reg[04] 02030304
[92799.148343] rk-vcodec ff650000.vpu_service: reg[05] 03040506
[92799.148841] rk-vcodec ff650000.vpu_service: reg[06] 070b1013
[92799.149339] rk-vcodec ff650000.vpu_service: reg[07] 0b0d1114
[92799.149837] rk-vcodec ff650000.vpu_service: reg[08] 0505080a
[92799.150335] rk-vcodec ff650000.vpu_service: reg[09] 080c0b11
[92799.150833] rk-vcodec ff650000.vpu_service: reg[10] 0e101516
[92799.151332] rk-vcodec ff650000.vpu_service: reg[11] 16151814
[92799.151829] rk-vcodec ff650000.vpu_service: reg[12] 0a0c0e10
[92799.152328] rk-vcodec ff650000.vpu_service: reg[13] 0c0b0b0c
[92799.152826] rk-vcodec ff650000.vpu_service: reg[14] 15171815
[92799.153324] rk-vcodec ff650000.vpu_service: reg[15] 0f121414
[92799.153822] rk-vcodec ff650000.vpu_service: reg[16] 03040509
[92799.154320] rk-vcodec ff650000.vpu_service: reg[17] 0404050d
[92799.154819] rk-vcodec ff650000.vpu_service: reg[18] 14141414
[92799.155317] rk-vcodec ff650000.vpu_service: reg[19] 14141414
[92799.155815] rk-vcodec ff650000.vpu_service: reg[20] 05050b14
[92799.156312] rk-vcodec ff650000.vpu_service: reg[21] 090d1414
[92799.156811] rk-vcodec ff650000.vpu_service: reg[22] 14141414
[92799.157309] rk-vcodec ff650000.vpu_service: reg[23] 14141414
[92799.157807] rk-vcodec ff650000.vpu_service: reg[24] 14141414
[92799.158305] rk-vcodec ff650000.vpu_service: reg[25] 14141414
[92799.158803] rk-vcodec ff650000.vpu_service: reg[26] 14141414
[92799.159301] rk-vcodec ff650000.vpu_service: reg[27] 14141414
[92799.159798] rk-vcodec ff650000.vpu_service: reg[28] 14141414
[92799.160296] rk-vcodec ff650000.vpu_service: reg[29] 14141414
[92799.160794] rk-vcodec ff650000.vpu_service: reg[30] 14141414
[92799.161293] rk-vcodec ff650000.vpu_service: reg[31] 14141414
[92799.161791] rk-vcodec ff650000.vpu_service: reg[32] 00000000
[92799.162288] rk-vcodec ff650000.vpu_service: reg[33] 00000000
[92799.162786] rk-vcodec ff650000.vpu_service: reg[34] 00000000
[92799.163284] rk-vcodec ff650000.vpu_service: reg[35] 00000000
[92799.163782] rk-vcodec ff650000.vpu_service: reg[36] 00000000
[92799.164280] rk-vcodec ff650000.vpu_service: reg[37] 00000000
[92799.164778] rk-vcodec ff650000.vpu_service: reg[38] 00000000
[92799.165276] rk-vcodec ff650000.vpu_service: reg[39] 00000000
[92799.165774] rk-vcodec ff650000.vpu_service: reg[40] 00000000
[92799.166271] rk-vcodec ff650000.vpu_service: reg[41] 00000000
[92799.166769] rk-vcodec ff650000.vpu_service: reg[42] 00000000
[92799.167267] rk-vcodec ff650000.vpu_service: reg[43] 00000000
[92799.167765] rk-vcodec ff650000.vpu_service: reg[44] 00000000
[92799.168263] rk-vcodec ff650000.vpu_service: reg[45] 00000000
[92799.168761] rk-vcodec ff650000.vpu_service: reg[46] 00000000
[92799.169259] rk-vcodec ff650000.vpu_service: reg[47] 00000000
[92799.169757] rk-vcodec ff650000.vpu_service: reg[48] 00000000
[92799.170255] rk-vcodec ff650000.vpu_service: reg[49] 004d9a00
[92799.170753] rk-vcodec ff650000.vpu_service: reg[50] 004d9a00
[92799.171250] rk-vcodec ff650000.vpu_service: reg[51] 02110311
[92799.171748] rk-vcodec ff650000.vpu_service: reg[52] 003f0000
[92799.172246] rk-vcodec ff650000.vpu_service: reg[53] 004cf721
[92799.172744] rk-vcodec ff650000.vpu_service: reg[54] 01011000
[92799.173242] rk-vcodec ff650000.vpu_service: reg[55] 00000000
[92799.173740] rk-vcodec ff650000.vpu_service: reg[56] 00000000
[92799.174238] rk-vcodec ff650000.vpu_service: reg[57] 00000000
[92799.174736] rk-vcodec ff650000.vpu_service: reg[58] 00000000
[92799.175234] rk-vcodec ff650000.vpu_service: reg[59] 00000000
[92799.175732] rk-vcodec ff650000.vpu_service: reg[60] 0038002a
[... 30 lines cut ...]
[92799.191167] rk-vcodec ff650000.vpu_service: reg[91] 00000000
[92799.191665] rk-vcodec ff650000.vpu_service: reg[92] 00000000
[92799.192163] rk-vcodec ff650000.vpu_service: reg[93] 00000000
[92799.192661] rk-vcodec ff650000.vpu_service: reg[94] 00000000
[92799.193160] rk-vcodec ff650000.vpu_service: reg[95] 962b4c85
[92799.193658] rk-vcodec ff650000.vpu_service: reg[96] 90901d50
[92799.194155] rk-vcodec ff650000.vpu_service: reg[97] 0000b694
[92799.194653] rk-vcodec ff650000.vpu_service: reg[98] 00000000
[92799.195151] rk-vcodec ff650000.vpu_service: reg[99] 00000000
[92799.195649] rk-vcodec ff650000.vpu_service: reg[100] 00000000
[92799.196159] rk-vcodec ff650000.vpu_service: reg[101] 1f522780
[92799.196667] rk-vcodec ff650000.vpu_service: reg[102] 00000000
[... cut ...]
[92799.199213] rk-vcodec ff650000.vpu_service: reg[107] 00000000
[92799.199721] rk-vcodec ff650000.vpu_service: reg[108] 00000000
[92799.200231] rk-vcodec ff650000.vpu_service: reg[109] 00001441
[92799.200740] rk-vcodec ff650000.vpu_service: reg[110] 3a122158
[92799.201249] rk-vcodec ff650000.vpu_service: reg[111] 00000000
[92799.201758] rk-vcodec ff650000.vpu_service: reg[112] 00000000
[92799.202267] rk-vcodec ff650000.vpu_service: reg[113] 00000000
[92799.202776] rk-vcodec ff650000.vpu_se 
```

rageworx, 2019-06-25:
Anyone knows this issue ?
Any additional firmware required for doing this ?

qvoid, 2019-06-25:
@rageworx I test your YUV420SP file in my RK3399 board with the command below:
```
mpi_enc_test -i ./testdump.yuv420sp.yuv -w 2464 -h 2064 -t 8 -o ./test.jpg
```
Things work fine, both 32-bit library and 64-bit library.
And the output image is:
[image]

Kernel version of my board:
```
rk3399_mid:/ # uname -a
Linux localhost 4.4.167 #4 SMP PREEMPT Mon Mar 11 08:51:05 CST 2019 aarch64
```

There is a **Page Fault** at your kernel log, It may has something to do with the wrong *with* and *height* parameters since the yuv file you provided is 2456x2064。

rageworx, 2019-06-25:
Dear @qvoid ,
I've been test with your shell input string,
```
./mpi_enc_test -i ./testdump.yuv420sp.yuv -w 2464 -h 2064 -t 8 -o ./test.jpg
```
and prints these lines -
```
>  ./mpi_enc_test -i ./testdump.yuv420sp.yuv -w 2464 -h 2064 -t 8 -o ./test.jpg
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: ./testdump.yuv420sp.yuv
mpi_enc_test: output file name: ./test.jpg
mpi_enc_test: width      : 2464
mpi_enc_test: height     : 2064
mpi_enc_test: format     : 0
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpi_enc_test: failed to open input file ./testdump.yuv420sp.yuv
mpi_enc_test: create default yuv image for test
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2464 h 2064 type 8
mpi: mpp version: 66b140ee author: Johnson Ding [h264d]: Fix unused parameter warning
mpi_enc_test: mpi_enc_test bps 19071360 fps 30 gop 60
jpege_api: jpege: hardware return error status 40
mpi_enc_test: test_mpp_run encoded frame 0 size 636250
mpi_enc_test: test_mpp_run encode max 1 frames
mpi_enc_test: mpi_enc_test success total frame 1 bps 152700000
mpp_buffer: ~MppBufferService cleaning misc group
```
Result test.jpg size is 636250, filled with 0x00 in data area.
dmesg printed as like this:
```
[107945.044431] rk_vcodec: vpu_service_ioctl:1875: error: unknown vpu service ioctl cmd 40086c01
[107945.315056] rk_iommu ff650800.iommu: Page fault at 0x0000000000007380 of type read
[107945.315771] rk_iommu ff650800.iommu: iova = 0x0000000000007380: dte_index: 0x0 pte_index: 0x7 page_offset: 0x380
[107945.316696] rk_iommu ff650800.iommu: mmu_dte_addr: 0x00000000ee14a000 dte@0x00000000ee14a000: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[107945.318179] rk-vcodec ff650000.vpu_service: vcodec, fault addr 0x00007380 status 4b
[107945.318870] rk-vcodec ff650000.vpu_service: no memory region mapped
[107945.319444] rk-vcodec ff650000.vpu_service: dumping enc register set:
[107945.320032] rk-vcodec ff650000.vpu_service: reg[00] 00000000
[107945.320550] rk-vcodec ff650000.vpu_service: reg[01] 00000000
[107945.321068] rk-vcodec ff650000.vpu_service: reg[02] 01010101
[107945.321585] rk-vcodec ff650000.vpu_service: reg[03] 01010101
[107945.322102] rk-vcodec ff650000.vpu_service: reg[04] 01010101
[107945.322620] rk-vcodec ff650000.vpu_service: reg[05] 01010101
[107945.323137] rk-vcodec ff650000.vpu_service: reg[06] 01010101
[107945.323654] rk-vcodec ff650000.vpu_service: reg[07] 01010101
[107945.324172] rk-vcodec ff650000.vpu_service: reg[08] 01010101
[107945.324689] rk-vcodec ff650000.vpu_service: reg[09] 01010101
[107945.325207] rk-vcodec ff650000.vpu_service: reg[10] 01010101
[107945.325724] rk-vcodec ff650000.vpu_service: reg[11] 01010101
[107945.326241] rk-vcodec ff650000.vpu_service: reg[12] 01010101
[107945.326759] rk-vcodec ff650000.vpu_service: reg[13] 01010101
[107945.327276] rk-vcodec ff650000.vpu_service: reg[14] 01010101
[107945.327794] rk-vcodec ff650000.vpu_service: reg[15] 01010101
[107945.328311] rk-vcodec ff650000.vpu_service: reg[16] 01010101
[107945.328828] rk-vcodec ff650000.vpu_service: reg[17] 01010101
[107945.329345] rk-vcodec ff650000.vpu_service: reg[18] 01010101
[107945.329863] rk-vcodec ff650000.vpu_service: reg[19] 01010101
[107945.330381] rk-vcodec ff650000.vpu_service: reg[20] 01010101
[107945.330898] rk-vcodec ff650000.vpu_service: reg[21] 01010101
[107945.331416] rk-vcodec ff650000.vpu_service: reg[22] 01010101
[107945.331934] rk-vcodec ff650000.vpu_service: reg[23] 01010101
[107945.332451] rk-vcodec ff650000.vpu_service: reg[24] 01010101
[107945.332969] rk-vcodec ff650000.vpu_service: reg[25] 01010101
[107945.333486] rk-vcodec ff650000.vpu_service: reg[26] 01010101
[107945.334004] rk-vcodec ff650000.vpu_service: reg[27] 01010101
[107945.334522] rk-vcodec ff650000.vpu_service: reg[28] 01010101
[107945.335040] rk-vcodec ff650000.vpu_service: reg[29] 01010101
[... 72 lines cut ...]
[107945.372889] rk-vcodec ff650000.vpu_service: reg[101] 1f522780
[107945.373418] rk-vcodec ff650000.vpu_service: reg[102] 00000000
[107945.373947] rk-vcodec ff650000.vpu_service: reg[103] 08109a60
[107945.374475] rk-vcodec ff650000.vpu_service: reg[104] 00000000
[107945.375004] rk-vcodec ff650000.vpu_service: reg[105] fc000000
[107945.375532] rk-vcodec ff650000.vpu_service: reg[106] 00000000
[107945.376060] rk-vcodec ff650000.vpu_service: reg[107] 00000000
[107945.376589] rk-vcodec ff650000.vpu_service: reg[108] 00000000
[107945.377118] rk-vcodec ff650000.vpu_service: reg[109] 00001441
[107945.377646] rk-vcodec ff650000.vpu_service: reg[110] 3a122158
[107945.378175] rk-vcodec ff650000.vpu_service: reg[111] 00000000
[107945.378704] rk-vcodec ff650000.vpu_service: reg[112] 00000000
[107945
```
I am running on debian-stretch-aarch64, Rock960B 4GB/32GB revision 1.3.
Kernel is
```
Linux linaro-alip 4.4.154 #34 SMP Mon Jun 10 18:19:17 KST 2019 aarch64 GNU/Linux
```
I can see "/dev/vpu_service" existed.

May I something missed ?
Some kind of kernel configuration need to be changed ?

Regards, Raph.K.

rageworx, 2019-06-25:
I have checked kernel config for MPP, and I can see these options are enabled ( =y ).
```
CONFIG_HAS_IOMEM=y
CONFIG_ARCH_ROCKCHIP=y
CONFIG_ROCKCHIP_MPP_SERVICE=y
CONFIG_ROCKCHIP_MPP_DEVICE=y
```
Hmm .. Am I need different kernel ?

rageworx, 2019-06-25:
And I have been tested on Sapphire Excavator board too.
Same environment, same codes ( from your repo ).
Same result.
Different kernel ( Linux linaro-alip 4.4.167 #21 SMP Tue May 14 10:39:25 KST 2019 aarch64 GNU/Linux )
And here dmesg
```
[    0.292089] rk-vcodec ff650000.vpu_service: Looking up vcodec-supply from device tree
[    0.292101] rk-vcodec ff650000.vpu_service: Looking up vcodec-supply property in node /vpu_service@ff650000 failed
[    0.292114] rk-vcodec ff650000.vpu_service: no regulator for vcodec
[    0.292947] rk-vcodec ff650000.vpu_service: probe device
[    0.293666] rk-vcodec ff650000.vpu_service: drm allocator with mmu enabled
[    0.294853] rk-vcodec ff650000.vpu_service: could not find power_model node
[    0.295523] rk-vcodec ff650000.vpu_service: init success
....
[  340.101366] rk_vcodec: vpu_service_ioctl:1890: error: unknown vpu service ioctl cmd 40086c01
[  340.136623] rk_iommu ff650800.iommu: Page fault at 0x0000000000006040 of type read
[  340.136642] rk_iommu ff650800.iommu: iova = 0x0000000000006040: dte_index: 0x0 pte_index: 0x6 page_offset: 0x40
[  340.136653] rk_iommu ff650800.iommu: mmu_dte_addr: 0x00000000f205c000 dte@0x00000000f205c000: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  340.136663] rk-vcodec ff650000.vpu_service: vcodec, fault addr 0x00006040 status 4b
[  340.136670] rk-vcodec ff650000.vpu_service: no memory region mapped
[  340.136676] rk-vcodec ff650000.vpu_service: dumping enc register set:
[  340.136684] rk-vcodec ff650000.vpu_service: reg[00] 00000000
[  340.136691] rk-vcodec ff650000.vpu_service: reg[01] 00000000
[  340.136698] rk-vcodec ff650000.vpu_service: reg[02] 01010101
[  340.136705] rk-vcodec ff650000.vpu_service: reg[03] 01010101
[  340.136712] rk-vcodec ff650000.vpu_service: reg[04] 01010101
[  340.136719] rk-vcodec ff650000.vpu_service: reg[05] 01010101
[  340.136727] rk-vcodec ff650000.vpu_service: reg[06] 01010101
[  340.136734] rk-vcodec ff650000.vpu_service: reg[07] 01010101
[  340.136741] rk-vcodec ff650000.vpu_service: reg[08] 01010101
[  340.136748] rk-vcodec ff650000.vpu_service: reg[09] 01010101
[  340.136755] rk-vcodec ff650000.vpu_service: reg[10] 01010101
[  340.136762] rk-vcodec ff650000.vpu_service: reg[11] 01010101
[  340.136769] rk-vcodec ff650000.vpu_service: reg[12] 01010101
[  340.136776] rk-vcodec ff650000.vpu_service: reg[13] 01010101
[  340.136783] rk-vcodec ff650000.vpu_service: reg[14] 01010101
[  340.136790] rk-vcodec ff650000.vpu_service: reg[15] 01010101
[  340.136797] rk-vcodec ff650000.vpu_service: reg[16] 01010101
[  340.136804] rk-vcodec ff650000.vpu_service: reg[17] 01010101
[  340.136811] rk-vcodec ff650000.vpu_service: reg[18] 01010101
[  340.136818] rk-vcodec ff650000.vpu_service: reg[19] 01010101
[  340.136825] rk-vcodec ff650000.vpu_service: reg[20] 01010101
[  340.136832] rk-vcodec ff650000.vpu_service: reg[21] 01010101
[  340.136839] rk-vcodec ff650000.vpu_service: reg[22] 01010101
[  340.136846] rk-vcodec ff650000.vpu_service: reg[23] 01010101
[  340.136853] rk-vcodec ff650000.vpu_service: reg[24] 01010101
[  340.136860] rk-vcodec ff650000.vpu_service: reg[25] 01010101
[  340.136867] rk-vcodec ff650000.vpu_service: reg[26] 01010101
[  340.136874] rk-vcodec ff650000.vpu_service: reg[27] 01010101
[  340.136881] rk-vcodec ff650000.vpu_service: reg[28] 01010101
[  340.136888] rk-vcodec ff650000.vpu_service: reg[29] 01010101
[  340.136895] rk-vcodec ff650000.vpu_service: reg[30] 01010101
[  340.136902] rk-vcodec ff650000.vpu_service: reg[31] 01010101
[  340.136909] rk-vcodec ff650000.vpu_service: reg[32] 00000000
[  340.136916] rk-vcodec ff650000.vpu_service: reg[33] 00000000
[  340.136923] rk-vcodec ff650000.vpu_service: reg[34] 00000000
[  340.136930] rk-vcodec ff650000.vpu_service: reg[35] 00000000
[  340.136937] rk-vcodec ff650000.vpu_service: reg[36] 00000000
[  340.136945] rk-vcodec ff650000.vpu_service: reg[37] 00000000
[  340.136952] rk-vcodec ff650000.vpu_service: reg[38] 00000000
[  340.136958] rk-vcodec ff650000.vpu_service: reg[39] 00000000
[  340.136965] rk-vcodec ff650000.vpu_service: reg[40] 00000000
[  340.136972] rk-vcodec ff650000.vpu_service: reg[41] 00000000
[  340.136979] rk-vcodec ff650000.vpu_service: reg[42] 00000000
[  340.136986] rk-vcodec ff650000.vpu_service: reg[43] 00000000
[  340.136993] rk-vcodec ff650000.vpu_service: reg[44] 00000000
[  340.137000] rk-vcodec ff650000.vpu_service: reg[45] 00000000
[  340.137008] rk-vcodec ff650000.vpu_service: reg[46] 00000000
[  340.137015] rk-vcodec ff650000.vpu_service: reg[47] 00000000
[  340.137022] rk-vcodec ff650000.vpu_service: reg[48] 00000000
[... 115 lines cut ...]
[  340.137838] rk-vcodec ff650000.vpu_service: reg[164] 00000000
[  340.137845] rk-vcodec ff650000.vpu_service: reg[165] 00000000
[  340.137852] rk-vcodec ff650000.vpu_service: reg[166] 00000000
[  340.137859] rk-vcodec ff650000.vpu_service: reg[167] 00000000
[  340.137866] rk-vcodec ff650000.vpu_service: reg[168] 00000000
[  340.137873] rk-vcodec ff650000.vpu_service: reg[169] 00000000
[  340.137880] rk-vcodec ff650000.vpu_service: reg[170] 00000000
[  340.137887] rk-vcodec ff650000.vpu_service: reg[171] 00000000
[  340.137894] rk-vcodec ff650000.vpu_service: reg[172] 00000000
[  340.137901] rk-vcodec ff650000.vpu_service: reg[173] 00000000
[  340.137908] rk-vcodec ff650000.vpu_service: reg[174] 00000000
[  340.137915] rk-vcodec ff650000.vpu_service: reg[175] 00000000
[... cut ...]
[  340.137950] rk-vcodec ff650000.vpu_service: reg[180] 00000000
[  340.137957] rk-vcodec ff650000.vpu_service: reg[181] 00000000
[  340.137964] rk-vcodec ff650000.vpu_service: reg[182] 00000000
[  340.137971] rk-vcodec ff650000.vpu_service: reg[183] 00000000
[  340.138170] rk-vcodec ff650000.vpu_service: resetting...
[  340.138224] rk-vcodec ff650000.vpu_service: reset done
[  340.138291] rk-vcodec ff650000.vpu_service: reset done
```
Result is same file size as 636250, filled with 0x00 (NULL).

teor292, 2019-06-25:
qvoid, can be problem in the board, not only kernel?  I have ROCK PI 4A v1.4, and the same error even on dummy image.

qvoid, 2019-06-26:
@rageworx More log info is need to analyze this problem. set the env *mpp_buffer_debug* and echo 0xffff to */sys/module/rk_vcodec/parameters/debug*.  And then re-run the test and save the complete console log and kernel log. Upload them here, Thx.

rageworx, 2019-06-26:
@qvoid 
/sys/module/rk_vcodec/paramters/debug should not be able to written with even sudo, or root account.
So I can give you result with -d 1 as below:
```
▶ ./mpi_enc_test -i ./testdump.yuv420sp.yuv -w 2464 -h 2064 -t 8 -o ./test.jpg -d 1
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: ./testdump.yuv420sp.yuv
mpi_enc_test: output file name: ./test.jpg
mpi_enc_test: width      : 2464
mpi_enc_test: height     : 2064
mpi_enc_test: format     : 0
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 1
mpi_enc_test: mpi_enc_test start
mpi_enc_test: failed to open input file ./testdump.yuv420sp.yuv
mpi_enc_test: create default yuv image for test
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 2464 h 2064 type 8
mpi: mpp_create enter ctx 0x5576241188 mpi 0x5576241190
mpi: mpp version: Without VCS info
mpi: mpp_create leave ret 0 ctx 0x5576242300 mpi 0x7fb4b01718
mpi: mpp_init enter ctx 0x5576242300 type 1 coding 8
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x5576242300 cmd 320003 parm 0x5576241198
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 19071360 fps 30 gop 60
mpi: mpi_control enter ctx 0x5576242300 cmd 320005 parm 0x55762411e4
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x5576242300 cmd 320007 parm 0x557624121c
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x5576242300 cmd 32000f parm 0x5576241cb0
mpi: mpi_control leave ret 0
mpi: mpi_encode_put_frame enter ctx 0x5576242300 frame 0x5576246200
jpege_api: jpege: hardware return error status 40
mpi: mpi_encode_put_frame leave ret 0
mpi: mpi_encode_get_packet enter ctx 0x5576242300 packet 0x7fe9230e88
mpi: mpi_encode_get_packet leave ret 0
mpi_enc_test: test_mpp_run encoded frame 0 size 636250
mpi_enc_test: test_mpp_run encode max 1 frames
mpi: mpi_reset enter ctx 0x5576242300
mpi: mpi_reset leave ret 0
mpi: mpp_destroy enter ctx 0x5576242300
mpi: mpp_destroy leave ret 0
mpi_enc_test: mpi_enc_test success total frame 1 bps 152700000
mpp_buffer: ~MppBufferService cleaning misc group
```
And kernel log here, ( removed some our MIPI CSI2 device kernel driver log )

[dmesg_1.txt](https://github.com/rockchip-linux/mpp/files/3328702/dmesg_1.txt)

qvoid, 2019-06-27:
@rageworx  Sorry, I forgot to tell you the value of **mpp_buffer_debug**. Set **mpp_buffer_debug** to **0x137**. In Linux you  need to use **export** command, or use **setprop** in Androd. And dump log again, Thanks.

rageworx, 2019-06-27:
@qvoid 
Still occured "permission denied" with root accout, so I had changed permissions with root accout for 666, then put 0xffff (65535) for debug parameter. And ran test again.
Just refer to following zip file:
[rkmpp.zip](https://github.com/rockchip-linux/mpp/files/3333364/rkmpp.zip)
Regards, Raph.

qvoid, 2019-06-27:
look like there is some thing wrong with buffer fd. It is abnormal that fd equals 0.
```
mpp_buffer: group  0 buffer  0 fd  0 ops buf create  ref_count 0 caller mpi_enc_test
mpp_buffer: group  0 buffer  0 fd  0 ops buf ref inc ref_count 0 caller mpi_enc_test
```

rageworx, 2019-06-28:
> .

Dear @qvoid , so who process mpp_buffer fd ? mpi_enc_test and mpp those are belong to here.
Any idea to solve this ?

qvoid, 2019-07-02:
@rageworx I think there is something wrong with memory mapping. But I don't know what cause that. See this log from your testing:
```
[  340.136623] rk_iommu ff650800.iommu: Page fault at 0x0000000000006040 of type read
[  340.136642] rk_iommu ff650800.iommu: iova = 0x0000000000006040: dte_index: 0x0 pte_index: 0x6 page_offset: 0x40
[  340.136653] rk_iommu ff650800.iommu: mmu_dte_addr: 0x00000000f205c000 dte@0x00000000f205c000: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  340.136663] rk-vcodec ff650000.vpu_service: vcodec, fault addr 0x00006040 status 4b
[  340.136670] rk-vcodec ff650000.vpu_service: no memory region mapped
```
There is a page fault when accessing 0x00006040. And the vpu_service show an "no memory region mapped" error.

qvoid, 2019-07-02:
> > look like there is some thing wrong with buffer fd. It is abnormal that fd equals 0.
> > ```
> > mpp_buffer: group  0 buffer  0 fd  0 ops buf create  ref_count 0 caller mpi_enc_test
> > mpp_buffer: group  0 buffer  0 fd  0 ops buf ref inc ref_count 0 caller mpi_enc_test
> > ```
> 
> Dear @qvoid , so who process mpp_buffer fd ? mpi_enc_test and mpp those are belong to here.
> Any idea to solve this ?

mpp_buffer is accessed both by MPP (call by user space application, in this case, it is the *mpi_enc_test*) and kernel.

rageworx, 2019-07-02:
> Dear @qvoid , so who process mpp_buffer fd ? mpi_enc_test and mpp those are belong to here.
> Any idea to solve this ?

So, what's next ? need to change something from rockchip mpp sources ?
Let me know,

rageworx, 2019-07-03:
@qvoid , Or do you have any contact point to solve this issue in platform level ? - as like something configuration or kernel level.
I believe we can solve this !

qvoid, 2019-07-04:
@rageworx  you can try to post this  issue in [RockChip Linux Kernel](https://github.com/rockchip-linux/kernel) or in [Rockchip redmine](https://redmine.rockchip.com.cn/)

rageworx, 2019-07-04:
Dear @qvoid, I commited an issue [here](https://github.com/rockchip-linux/kernel/issues/172).
I will share something new solution availed from kernel team.