rockchip-linux/mpp #90: RK3399 mpi_enc_test 使用android-ndk-r17c 进行编译， 运行时报错

khuang024, 2019-03-11:
host 主机系统 ubuntu 16.04

使用开发手册v3中的步骤使用android-ndk-r10d进行编译时 mpi_enc_test 运行正常。
使用android-ndk-r17c 进行编译成功， 但是运行时报错。

编译设置:
在 mpp/build/android/arm/ 路径下运行
cmake  -DCMAKE_SYSTEM_NAME=Android   -DCMAKE_SYSTEM_VERSION=21   -DCMAKE_ANDROID_ARCH_ABI="armeabi-v7a"    -DCMAKE_ANDROID_NDK=/media/perceptin/Data/Downloads/android-ndk-r17c/   -DCMAKE_ANDROID_STL_TYPE=system ../../../ 

之后运行
make
将mpi_enc_test push 到3399上 运行
./mpi_enc_test_17  -w 1024 -h 576 -t 7 -f 4 -i ./TextInMotion-VideoSample-576p.yuv   -o ./out.h264

Log 如下

03-11 01:04:54.738  1778  1778 I mpi_enc_test: cmd parse result:
03-11 01:04:54.738  1778  1778 I mpi_enc_test: input  file name: ./TextInMotion-VideoSample-576p.yuv
03-11 01:04:54.738  1778  1778 I mpi_enc_test: output file name: ./out.h264
03-11 01:04:54.738  1778  1778 I mpi_enc_test: width      : 1024
03-11 01:04:54.738  1778  1778 I mpi_enc_test: height     : 576
03-11 01:04:54.738  1778  1778 I mpi_enc_test: format     : 4
03-11 01:04:54.739  1778  1778 I mpi_enc_test: type       : 7
03-11 01:04:54.739  1778  1778 I mpi_enc_test: debug flag : 0
03-11 01:04:54.739  1778  1778 I mpi_enc_test: mpi_enc_test start
03-11 01:04:54.740  1778  1778 I mpp_rt  : NOT found ion allocator
03-11 01:04:54.740  1778  1778 I mpp_rt  : found drm allocator
03-11 01:04:54.740  1778  1778 I mpi_enc_test: mpi_enc_test encoder test start w 1024 h 576 type 7
03-11 01:04:54.740  1778  1778 I mpi     : mpp version: 2dc830f author: sliver.chen [vp8e]: add vpu1 and vpu2 vp8e supprt.
03-11 01:04:54.745  1778  1778 I mpi_enc_test: mpi_enc_test bps 2211840 fps 30 gop 60
03-11 01:04:54.745  1778  1778 I h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 2211840 [2073600 : 2350080]
**03-11 01:04:54.752  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.752  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.752  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14**
03-11 01:04:54.752  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 0 size 0
03-11 01:04:54.756  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.756  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.756  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.756  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 1 size 0
03-11 01:04:54.760  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.760  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.760  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.760  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 2 size 0
03-11 01:04:54.764  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.765  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.765  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.765  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 3 size 0
03-11 01:04:54.768  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.768  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.769  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.769  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 4 size 0
03-11 01:04:54.772  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.772  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.772  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.772  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 5 size 0
03-11 01:04:54.776  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.776  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.776  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.779  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.779  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.779  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14
03-11 01:04:54.779  1778  1778 I mpi_enc_test: test_mpp_run encoded frame 6 size 0
03-11 01:04:54.784  1778  1779 E mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
03-11 01:04:54.785  1778  1779 E hal_h264e_vepu2: mpp_device_send_reg failed ret 14
03-11 01:04:54.785  1778  1779 E mpp_enc : mpp 0x7e2bf17000 hal_hw_start failed return 14

HermanChen, 2019-03-11:
高版本 ndk 建议使用 ndk 里的 toolchain.cmake

khuang024, 2019-03-13:
已解决，谢谢

closed 2019-03-13 by khuang024