rockchip-linux/mpp #92: 请问有没什么指令或办法可查到RK3288或RK3328是否已经启用的硬解码了？

liyungithub, 2019-04-11:
现在我做了一个小demo来测试RK3328的硬解码能力，发现CPU占用资源非常高（解1080p@25bps），而且也看到的解码帧率显示很卡，不知道是解码不过来还是CPU渲染不过来。现在我想确认的是：
1、RK3328能否流畅地解码1080p@25bps视频？
2、用什么办法可以查到RK3288或RK3328是否已经启用的硬解码了？

liyungithub, 2019-04-12:
哪位大神能回复下？

HermanChen, 2019-04-12:
是什么格式的视频？
用 mpi_dec_test 来测试下解码性能比较好。
看是否硬解，把内核的硬件运行时间打开看下
echo 0x100 > /sys/module/rk_vcodec/parameters/debug

liyungithub, 2019-04-12:
h264格式的，1080p@25bps，画面是阿凡达电影，在RK3288测试是没问题的，但在RK3328跑就卡了，RK3328是能畅地解码H264，1080p@25bps视频的吧？也有可能是画面渲染不过来问题，还有怎么看硬件运行时间？已经执行echo 0x100 > /sys/module/rk_vcodec/parameters/debug了。
[image]

HermanChen, 2019-04-12:
echo 完了再支持测试程序

liyungithub, 2019-04-12:
[image]
是这个打印？

HermanChen, 2019-04-12:
是的，这个就是硬件的解码时间

liyungithub, 2019-04-12:
好的，谢谢了！

closed 2019-04-12 by liyungithub

mmabluesky, 2019-04-12:
@liyungithub 
你怎么运行那个测试命令的 我怎么一直没有vpu_dec这种打印 ，只能看到这个
[image]

mmabluesky, 2019-04-15:
@HermanChen 
怎样才能启用硬解码 ，我编了一个固件，用vlc播，貌似默认是软解

HermanChen, 2019-04-15:
这样的开放式问题我没法回答……