rockchip-linux/mpp #94: mpp_buffer_create reach group size limit

oggcgg, 2019-04-23:
Hi RK folks,

I'm currently using mpp on RK1108, in my application, the decode frame size changes sometime, from 240x135 to 240x180, but, somehow, I frequently face the error that  mpp_buffer_create reach group size limit, error log would be following:

**decode_get_frame get info changed found
decoder require buffer w:h [240:134] stride [240:144] size 69120
mpp_buffer: mpp_buffer_create required size 92160 reach group size limit 69120
mpp_buffer: mpp_buffer_create required size 92160 reach group size limit 69120
mpp_buffer: mpp_buffer_create required size 92160 reach group size limit 69120
decode_get_frame get info changed found
decoder require buffer w:h [240:180] stride [240:192] size 92160
mpp_buf_slot: found info change ready set without internal info change**

then the app will get stuck until to restart app, and this issue could be reproduced by test/mpi_dec_test.c  (with some modifications)

Is there any way to change decode size correctly?  Thanks!

oggcgg, 2019-04-23:
One more thing, the video streaming will be stopped randomly when changing resolution, so i may put_packet with an uncompleted-frame, is that got something to do with this? If so, how could I to deal with it?

HermanChen, 2019-04-24:
Is it jpeg stream or h.264 stream? Can you provide the sample stream you are using?
The info change flow in mpi_dec_test works for the case on Android.

oggcgg, 2019-04-24:
It's h264 stream, following is the stream I used. The app will read the h264 files by fixed size, like 1024 bytes each time, then put to packet, let encoder split frame. App will randomly stop reading/decoding one file and switch to the other one by user's request.

[video_stream.zip](https://github.com/rockchip-linux/mpp/files/3110274/video_stream.zip)

qvoid, 2019-04-24:
@oggcgg  Check out to see if buffers for each decoded frame is recreated and the size is enough. Once "info changed" is encountered, the output buffer should be recreated because the resolution of the incoming packet is already changed.

oggcgg, 2019-04-29:
I found something weird. Every time I stop putting packet to decoder then change resolution, put the new resolution packet to decoder, I'll always get few wrong size raw frames from decode buffer. (ex: resolution changes to 240x135, then will get some 240x180 frames)  

I'm not sure if it's something to do with that we set **mpp_buffer_group_limit_config** after **decode_get_frame** get info changed found, then we get a larger frame from buffer, leads to facing the error "**mpp_buffer: mpp_buffer_create required size 92160 reach group size limit 69120**"

qvoid, 2019-05-30:
@oggcgg  There must be something wrong when getting decoded frame. From the log that you provided, the group size limit is configure twice in one *info_changed* event. Try this patch :
```
diff --git a/mpp/codec/mpp_dec.cpp b/mpp/codec/mpp_dec.cpp
index ed13f58..40f173b 100644
--- a/mpp/codec/mpp_dec.cpp
+++ b/mpp/codec/mpp_dec.cpp
@@ -368,7 +368,17 @@ static void mpp_dec_push_display(Mpp *mpp, HalDecTaskFlag flags)
     MppBufSlots frame_slots = dec->frame_slots;
     RK_U32 eos = flags.eos;
     HalDecTaskFlag tmp = flags;
     tmp.eos = 0;
+    tmp.info_change = 0;
 
     mpp->mThreadHal->lock(THREAD_OUTPUT);
     while (MPP_OK == mpp_buf_slot_dequeue(frame_slots, &index, QUEUE_DISPLAY)) {
```

oggcgg, 2019-07-17:
@qvoid Sorry for being to reply, I did not notice that you had replied me! lol...
For my issue, I was trying to enlarge the buffer size. The buffer size I used was depending on frame size. However, I tried to apply the max frame size (240x180) when configing mpp buffer group. Then, this issue gone. I think we could close this issue. Thank you guys!

`- ret = mpp_buffer_group_limit_config(pH264Decoder->frm_grp, frame_size, 24);`
`+ ret = mpp_buffer_group_limit_config(pH264Decoder->frm_grp, MAX_BUFFER_SIZE, 24);`

closed 2019-07-17 by oggcgg