rockchip-linux/mpp #95: rk3328/rk1808 h.264 encoding speed

jonsmirl, 2019-04-26:
Is the h.264 encoder hardware the same on the RK1808 and the RK3228? Both are limited to 1080P30? Since RK1808 supports 4K sensors a 1080P30 encoder seems quite slow.

If the hardware is the same/similar, I can write my code on RK3228 while I wait for someone to make RK1808 dev board.

In the RK1808 datasheet it says:
Video Encoder
 Support H.264 video encoder at BP/MP/HP@level4.2
 1920p@30FPS
 1x1080p@30fps or 2x720p@30fps encoding

What does 1920p@30FPS mean? Does it mean 1920x3414 @ 30 FPS?

sliver-chen, 2019-05-16:
1. in my impression, there are two kind types of rockchip vpu, vpu1 and vpu2 respectively.
but mpp has unified the difference brought by hardware, so you can develop application first in RK3228 and then move it to RK1808 board.
2. 1920p@30fps means that 1920x1080 with 30 fps encoding ability.

HermanChen, 2019-05-17:
level4.2 is an error. Level 4.2 is 1080p@60fps while level 4.1 is 1080p@30fps
1920p is also an error. I think it will be 1920x1080@30fps.
If you add RK1808 support to mpp_vpu_version with HAVE_VPU2 flag it can support RK1808.
But RK1808 only have H.264/jpeg encoder/decoder.

jonsmirl, 2019-05-17:
Is there a mode when the rk1808 h.264 encoder supports 1080P@60FPS? That is a critical feature for the product we are designing. If the encoder supports 1080P@60FPS then it should also be able to do 4K@15FPS.

HermanChen, 2019-05-17:
Emm...
The hardware encoding has limit that unable to support larger resolution than its design.
For internal memory and logic design limit 1080p hardware encoder can not support larger size like 4K. But smaller resolution with higher framerate is OK.

jonsmirl, 2019-05-17:
Is there a mode that supports 1080P60?

HermanChen, 2019-05-17:
You can try raise the vpu frequency. But it will effect stability and power consumption.

jonsmirl, 2019-05-17:
I see, the hardware on the RK1808 is Level 4.1 so it is limited to 1080P30. The above reply was not clear in its meaning.

kira2016, 2019-08-07:
RK3328 datasheet   HEVC:  1x1080p@30fps or 2x720p@30fps encoding,
是否支持 4路 编码？如果可以，要降低分辨率，要到多少？@
@HermanChen