rockchip-linux/mpp #96: RK3288 mpp硬件解码报错VPU_IOC_SET_REG failed。

lwdddddddd, 2019-05-06:
您好，我在RK3288上面使用mpp硬件解码，报如下错误：

haimo-free, 2019-05-10:
目前解决了么？如何解决的？

niujiabenben, 2019-05-26:
我在RK3399上遇到过这个问题, 当时我在一个新建的账户下运行mpi_dec_test, 报的错和上述的一模一样. 切换到主账户就正常了. 似乎是权限的问题.

zhuboMr, 2019-06-21:
请问解决了吗？ 如何解决的？

HermanChen, 2019-06-21:
有完整 log 么？

lwdddddd, 2019-07-20:
抱歉，事情比较多，这么久才登录回来。

lwdddddd, 2019-07-20:
抱歉，事情比较多，这么久才登录回来。
我用系统自带的librockchip_mpp.so库不会报这个错误，可以正常解码。
我的linux系统是：Linux linaro-alip 4.4.154 #94 SMP Wed Mar 27 14:41:58 CST 2019 armv7l GNU/Linux
但是我使用在这里下载的mpp源码编译出来的librockchip_mpp.so库进行解码的时候就会报这个错误，完整的log信息是：
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi: mpp version: Without VCS info
jpegd_parser: input img maybe png format,check it
jpegd_parser: input img maybe png format,check it
mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_start mpp_device_send_reg Failed!!!
mpp_device: mpp_device_wait_reg ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp_log: error: decoder_get_frame get err info:1 discard:0.
mpp_log: error: fl_decode_advanced failed.
mpp_task_impl: mpp_task_queue_deinit idx 0 task 0x80c3e860 status 3 meta size 0
mpp_task_impl: Assertion p->tasks[i].status == MPP_INPUT_PORT || p->tasks[i].status == MPP_INPUT_HOLD failed at mpp_task_queue_deinit:445

lwdddddd, 2019-07-20:
我编译测试程序的时候已经先移除了系统自带的mpp相关库，库应该没有出现混用情况。

HermanChen, 2019-07-22:
看起来上层和内核不匹配

lwdddddd, 2019-07-29:
CloseIssueInput!

lwdddddd, 2019-07-29:
input (CloseIssueInput!)

closed 2019-07-29 by lwdddddddd

lwdddddddd, 2019-08-16:
以上报错的原因找到了，cmake编译的时侯需要加上选项-DHAVE_DRM=ON

coflery, 2021-04-21:
> 以上报错的原因找到了，cmake编译的时侯需要加上选项-DHAVE_DRM=ON

谢谢，同样问题，已解决