rockchip-linux/mpp #98: mpp 解码h264出现马赛克

zhuboMr, 2019-05-27:
使用ffmpeg 将mp4视频转换 h264文件。 用mpp对其解码. rga转换 drm显示，显示时出现了马赛克，花屏现象。
h264文件 在vlc上播放并未出现任何问题。我无法确定原因。

qvoid, 2019-05-27:
可以开启调试开关把解码的yuv数据dump出来看看是否正常：设置环境变量mpp_debug的值为0x400。然后再执行解码操作，会在 */data* 目录下生成 *mpp_dec_out_<tid>.bin* 。这是个颜色格式为NV12的 yuv文件，可以用yuv查看器或者ffplay打开。使用ffplay播放的话命令如下：
```
ffplay -f rawvideo -video_size <width>x<height> -pix_fmt nv12 <yuv_file>
```

zhuboMr, 2019-05-28:
@qvoid 感谢你的回答
我使用了 mpp/test/mpi_dec_test.c 做了相关实验,确定是 mpp解码后yuv数据产生了花屏，及马赛克。并非是rga转换和drm显示的原因。 使用命令如下
./main -t 7 -i /mnt/usb/video/222.h264 -o /mnt/usb/video/222.yuv -w 1920 -h 1080 -n 300

我怀疑是在解码B帧的时候出现了问题，但rk3288 是支持B帧的。
我实验了2个mp4，通过ffmpeg 转换h264，都出现了花屏现象。使用命令如下： 
ffmpeg -i xxx.mp4 -codec copy -bsf h264_mp4toannexb -f h264 xx.264

qvoid, 2019-05-28:
@zhuboMr  请问你使用的什么芯片？什么OS？内核版本多少？mpp的版本是多少？能否把有问题的片源也发出来下，我这边分析看下是什么原因。

zhuboMr, 2019-05-28:
@qvoid  相关信息 Linux rk3288 4.4.154 #1 SMP Tue May 7 13:14:19 CST 2019 armv7l GNU/Linux 

相关的h264 视频文件
链接: https://pan.baidu.com/s/17ZxAwR1Wwl7ml6Unm3qKRQ 提取码: xz92

qvoid, 2019-05-28:
@zhuboMr  我这里用RK3288 Android 7.1.2, Kerenl: Linux localhost 4.4.143 #43 SMP PREEMPT Thu Apr 11 10:34:01 CST 2019 armv7l, Mpp的代码用的是最新的，验证你提供的片源，没有看到马赛克。

qvoid, 2019-05-28:
@zhuboMr Mpp的代码是最新的吗？

zhuboMr, 2019-05-28:
请问应该如何获取 mpp 版本号，或者 mpp/test/mpi_dec_test.c 是否应该有其他的设置。

zhuboMr, 2019-05-28:
@qvoid  请问应该如何获取 mpp 版本号，或者 mpp/test/mpi_dec_test.c 是否应该有其他的设置。

qvoid, 2019-05-28:
在你执行mpi_dec_test之后，logcat中应该会有＂mpi version＂的打印，或者用strings 命令直接从 so库中获取：
```
strings libmpp.so | grep author
```

zhuboMr, 2019-05-28:
@qvoid  我这里 没有libmpp.so 只有 librockchip_mpp.so  
strings librockchip_mpp.so | grep author 无法获取版本号
mpi: mpp version: Without VCS info

zhuboMr, 2019-05-28:
@qvoid  我会重新编译库文件，再次实验，非常感谢您的帮助。

qvoid, 2019-05-28:
你的应该是Linux版本的，所以编译出来的是librockchip_mpp.so，但是不知道为何版本相关的信息没了。你可以使用最新的Mpp代码编译试试，看看还会不会马赛克。

qvoid, 2019-05-30:
@zhuboMr 最新的代码是否能正常？

zhuboMr, 2019-05-31:
抱歉这几天在弄音频播放。我尝试了编译最新的库，但是运行时出现了问题。 mpi_dec_test 测序程序也无法运行了。 我还在查找原因 
[root@rk3288:~/nuklear/demo]# ./mpi_dec_test -t 7 -i /mnt/usb/video/222.h264 
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: /mnt/usb/video/222.h264
mpi_dec_test: output file name: 
mpi_dec_test: config file name: 
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 98982267
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: Without VCS info
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
h264d_dpb: dpb_size error.
mpi_dec_test: decode_get_frame get info changed found
mpi_dec_test: decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 1
mpi_dec_test: decoder_get_frame get err info:0 discard:1.
mpi_dec_test: decode_get_frame get frame 2
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 3
mpi_dec_test: decoder_get_frame get err info:0 discard:1.
mpi_dec_test: decode_get_frame get frame 4

err_info = mpp_frame_get_errinfo(frame) | mpp_frame_get_discard(frame);
 if (err_info) {
          mpp_log("decoder_get_frame get err info:%d discard:%d.\n",
          mpp_frame_get_errinfo(frame), mpp_frame_get_discard(frame));
 }

[image]

zhuboMr, 2019-06-11:
新版库文件可以使用，谢谢

closed 2019-06-11 by zhuboMr

HermanChen, 2019-06-11:
感谢反馈~