rockchip-linux/mpp #100: MPP_BUFFER_TYPE_NORMAL出现错误

huliang4github, 2019-06-12:
mpi_dec_mt_test.c 中我把MPP_BUFFER_TYPE_ION改为MPP_BUFFER_TYPE_NORMAL，运行的时候获取的每一个frame都带有errinfo，并且驱动也提示了错误，请问是什么问题？

[ 1782.229207] rk_iommu ff360480.iommu: Page fault at 0x0000000000151800 of type write
[ 1782.230003] rk_iommu ff360480.iommu: iova = 0x0000000000151800: dte_index: 0x0 pte_index: 0x151 page_offset: 0x800
[ 1782.231018] rk_iommu ff360480.iommu: mmu_dte_addr: 0x0000000037d7b000 dte@0x0000000037d7b000: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[ 1782.232631] rk-vcodec ff360000.rkvdec: vcodec, fault addr 0x00151800 status 6b
[ 1782.233367] rk-vcodec ff360000.rkvdec: vcodec, reg[  4] mem region [00] 0xffc00000 200000
[ 1782.234183] rk-vcodec ff360000.rkvdec: vcodec, reg[  6] mem region [01] 0xffffe000 1000
[ 1782.234987] rk-vcodec ff360000.rkvdec: vcodec, reg[ 41] mem region [02] 0xffc00000 200000
[ 1782.235802] rk-vcodec ff360000.rkvdec: vcodec, reg[ -1] mem region [03] 0xffffd000 1000
[ 1782.236605] rk-vcodec ff360000.rkvdec: vcodec, reg[ 42] mem region [04] 0xffff8000 3000
[ 1782.237409] rk-vcodec ff360000.rkvdec: vcodec, reg[ 43] mem region [05] 0xffffc000 1000
[ 1782.238213] rk-vcodec ff360000.rkvdec: vcodec, reg[ 75] mem region [06] 0xfffc0000 24000
[ 1782.239013] rk-vcodec ff360000.rkvdec: dumping dec register set:
[ 1782.239597] rk-vcodec ff360000.rkvdec: reg[00] 34101f00
[ 1782.240111] rk-vcodec ff360000.rkvdec: reg[01] 00000063
[ 1782.240625] rk-vcodec ff360000.rkvdec: reg[02] 00100000
[ 1782.241139] rk-vcodec ff360000.rkvdec: reg[03] ffe50850
[ 1782.241653] rk-vcodec ff360000.rkvdec: reg[04] ffc00000
[ 1782.242166] rk-vcodec ff360000.rkvdec: reg[05] 00005be0
[ 1782.242680] rk-vcodec ff360000.rkvdec: reg[06] ffffe000
[ 1782.243194] rk-vcodec ff360000.rkvdec: reg[07] 00000000
[ 1782.243707] rk-vcodec ff360000.rkvdec: reg[08] 0000e100
[ 1782.244221] rk-vcodec ff360000.rkvdec: reg[09] 00015180
[ 1782.244735] rk-vcodec ff360000.rkvdec: reg[10] 00000000
[ 1782.245249] rk-vcodec ff360000.rkvdec: reg[11] 00000000
[ 1782.245763] rk-vcodec ff360000.rkvdec: reg[12] 00000000
[ 1782.246277] rk-vcodec ff360000.rkvdec: reg[13] 00000000
[ 1782.246790] rk-vcodec ff360000.rkvdec: reg[14] 00000000
[ 1782.247304] rk-vcodec ff360000.rkvdec: reg[15] 00000000
[ 1782.247818] rk-vcodec ff360000.rkvdec: reg[16] 00000000
[ 1782.248331] rk-vcodec ff360000.rkvdec: reg[17] 00000000
[ 1782.248845] rk-vcodec ff360000.rkvdec: reg[18] 00000000
[ 1782.249359] rk-vcodec ff360000.rkvdec: reg[19] 00000000
[ 1782.249872] rk-vcodec ff360000.rkvdec: reg[20] 00000000
[ 1782.250386] rk-vcodec ff360000.rkvdec: reg[21] 00000000
[ 1782.250900] rk-vcodec ff360000.rkvdec: reg[22] 00000000
[ 1782.251414] rk-vcodec ff360000.rkvdec: reg[23] 00000000
[ 1782.251928] rk-vcodec ff360000.rkvdec: reg[24] 00000000
[ 1782.252441] rk-vcodec ff360000.rkvdec: reg[25] 00000000
[ 1782.252955] rk-vcodec ff360000.rkvdec: reg[26] 00000000
[ 1782.253468] rk-vcodec ff360000.rkvdec: reg[27] 00000000
[ 1782.253982] rk-vcodec ff360000.rkvdec: reg[28] 00000000
[ 1782.254495] rk-vcodec ff360000.rkvdec: reg[29] 00000000
[ 1782.255009] rk-vcodec ff360000.rkvdec: reg[30] 00000000
[ 1782.255523] rk-vcodec ff360000.rkvdec: reg[31] 00000000
[ 1782.256037] rk-vcodec ff360000.rkvdec: reg[32] 00000000
[ 1782.256550] rk-vcodec ff360000.rkvdec: reg[33] 00000000
[ 1782.257064] rk-vcodec ff360000.rkvdec: reg[34] 00000000
[ 1782.257578] rk-vcodec ff360000.rkvdec: reg[35] 00000000
[ 1782.258092] rk-vcodec ff360000.rkvdec: reg[36] 00000000
[ 1782.258606] rk-vcodec ff360000.rkvdec: reg[37] 00000000
[ 1782.259119] rk-vcodec ff360000.rkvdec: reg[38] 00000000
[ 1782.259633] rk-vcodec ff360000.rkvdec: reg[39] 00000000
[ 1782.260147] rk-vcodec ff360000.rkvdec: reg[40] 00000000
[ 1782.260660] rk-vcodec ff360000.rkvdec: reg[41] ffc00000
[ 1782.261174] rk-vcodec ff360000.rkvdec: reg[42] ffff8000
[ 1782.261688] rk-vcodec ff360000.rkvdec: reg[43] ffffc000
[ 1782.262202] rk-vcodec ff360000.rkvdec: reg[44] fffffff7
[ 1782.262716] rk-vcodec ff360000.rkvdec: reg[45] 00000000
[ 1782.263229] rk-vcodec ff360000.rkvdec: reg[46] 00000000
[ 1782.263743] rk-vcodec ff360000.rkvdec: reg[47] 00000000
[ 1782.264257] rk-vcodec ff360000.rkvdec: reg[48] 00000000
[ 1782.264771] rk-vcodec ff360000.rkvdec: reg[49] 00000000
[ 1782.265284] rk-vcodec ff360000.rkvdec: reg[50] 00000000
[ 1782.265798] rk-vcodec ff360000.rkvdec: reg[51] 00000000
[ 1782.266311] rk-vcodec ff360000.rkvdec: reg[52] 00000000
[ 1782.266824] rk-vcodec ff360000.rkvdec: reg[53] 00000000
[ 1782.267338] rk-vcodec ff360000.rkvdec: reg[54] 00000000
[ 1782.267851] rk-vcodec ff360000.rkvdec: reg[55] 00000000
[ 1782.268365] rk-vcodec ff360000.rkvdec: reg[56] 00000000
[ 1782.268878] rk-vcodec ff360000.rkvdec: reg[57] 00000000
[ 1782.269392] rk-vcodec ff360000.rkvdec: reg[58] 00000000
[ 1782.269906] rk-vcodec ff360000.rkvdec: reg[59] 00000000
[ 1782.270419] rk-vcodec ff360000.rkvdec: reg[60] 00000000
[ 1782.270933] rk-vcodec ff360000.rkvdec: reg[61] 00000000
[ 1782.271447] rk-vcodec ff360000.rkvdec: reg[62] 00000000
[ 1782.271961] rk-vcodec ff360000.rkvdec: reg[63] 00000000
[ 1782.272474] rk-vcodec ff360000.rkvdec: reg[64] 00000000
[ 1782.272988] rk-vcodec ff360000.rkvdec: reg[65] 00000000
[ 1782.273502] rk-vcodec ff360000.rkvdec: reg[66] 00000000
[ 1782.274015] rk-vcodec ff360000.rkvdec: reg[67] 00000000
[ 1782.274529] rk-vcodec ff360000.rkvdec: reg[68] 00000000
[ 1782.275042] rk-vcodec ff360000.rkvdec: reg[69] 00000000
[ 1782.275556] rk-vcodec ff360000.rkvdec: reg[70] 00000000
[ 1782.276069] rk-vcodec ff360000.rkvdec: reg[71] 00000000
[ 1782.276584] rk-vcodec ff360000.rkvdec: reg[72] 00000000
[ 1782.277097] rk-vcodec ff360000.rkvdec: reg[73] 00000000
[ 1782.277611] rk-vcodec ff360000.rkvdec: reg[74] 00000000
[ 1782.278124] rk-vcodec ff360000.rkvdec: reg[75] fffc0000
[ 1782.278637] rk-vcodec ff360000.rkvdec: reg[76] 00000000
[ 1782.279151] rk-vcodec ff360000.rkvdec: reg[77] 3fffffff
[ 1782.279869] rk-vcodec ff360000.rkvdec: resetting...
[ 1782.280978] rk-vcodec ff360000.rkvdec: reset done
[ 1782.281505] rk-vcodec ff360000.rkvdec: reset done

HermanChen, 2019-06-12:
硬件无法使用 normal 类型的 buffer，nornal 类型的 buffer 只在调试时使用，后面应该会去掉

huliang4github, 2019-06-12:
了解，谢谢！

closed 2019-06-12 by huliang4github