rockchip-linux/mpp #103: [root@rockchip:~]# mpi_dec_test -t 7 -i /root/Tennis1080p.h264 -n 10 mpi_dec_test: cmd parse result: mpi_dec_test: input file name: /root/Tennis1080p.h264 mpi_dec_test: output file name: mpi_dec_test: width : 0 mpi_dec_test: height : 0 mpi_dec_test: type : 7 mpi_dec_test: debug flag : 0 mpi_dec_test: max frames : 10 mpi_dec_test: mpi_dec_test start mpi_dec_test: input file size 10786528 mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7 mpi: mpp version: Without VCS info hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:119 hal_h264d_api: hal_h264d_init hard mode error, value=0 hal_h264d_api: Assertion 0 failed at hal_h264d_init:169 mpp_device: mpp_device_init failed to find device for coding 7 type 0 hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In mpp_rt: NOT found ion allocator mpp_rt: found drm allocator Segmentation fault

yourepresious, 2019-06-26:
用RK3399pro做解码测试的时候报错。
[root@rockchip:~]# mpi_dec_test -t 7 -i /root/Tennis1080p.h264 -n 10
mpi_dec_test: cmd parse result:
mpi_dec_test: input file name: /root/Tennis1080p.h264
mpi_dec_test: output file name:
mpi_dec_test: width : 0
mpi_dec_test: height : 0
mpi_dec_test: type : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 10
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 10786528
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: Without VCS info
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:119
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:169
mpp_device: mpp_device_init failed to find device for coding 7 type 0
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Segmentation fault

参考了之前的#52的issue，但是看不出来是如何解决的。请问能详细的说一下吗。

HermanChen, 2019-06-26:
mpp_device: mpp_device_init failed to find device for coding 7 type 0
内核 dts 没有配置上解码器设备，代码是开源的，搜下 log 就直接是哪里的问题

yourepresious, 2019-06-26:
是的。现在就是不知道dts怎么配置解码器设备。初次使用mpp很多地方不是很懂。

HermanChen, 2019-06-26:
这个到 RK 的 redmine 上提个问题吧，这问题不属于这边能回答的了……

yourepresious, 2019-06-26:
好的，能麻烦给个链接吗。

HermanChen, 2019-06-26:
http://redmine.rockchip.com.cn/
先去找找 FAE

yourepresious, 2019-06-26:
这个网站 我一周前注册过，一直提示 “等待管理员的审核。”是现在没人维护了吗~

HermanChen, 2019-06-27:
需要审核，要报上公司信息，在公司客户列表里的，才会通过吧