rockchip-linux/mpp #104: mpph264enc real speed?

doovirus, 2019-07-16:
What is the real speed of h264 encoding on rk3288?

The rk3288 datasheets show the fllowing:

>  Maximum frame rate is up to 30fps@1920x1080③

However, the result of testing it was 21fps.
Who ever seen 30fps of h264 encoding??

All tests done on aio3288j board.
next below is test commands.
**First, the input was confirmed to be 60fps using rkcamsrc.**

```
root@firefly:~# gst-launch-1.0 -v rkcamsrc device=/dev/video1 ! video/x-raw,format=NV12,width=1920,height=1088,framerate=60/1 ! fpsdisplaysink name=fpssink text-overlay=false video-sink=fakesink signal-fps-measurements=true sync=false
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRKCamSrc:rkcamsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
requesting 4 MMAP buffers/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 31, dropped: 0, current: 61.95, average: 61.95
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 61, dropped: 0, current: 59.94, average: 60.95
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 91, dropped: 0, current: 59.86, average: 60.58
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 122, dropped: 0, current: 60.02, average: 60.44
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 152, dropped: 0, current: 59.94, average: 60.34
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 182, dropped: 0, current: 59.85, average: 60.26
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 213, dropped: 0, current: 59.98, average: 60.22
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 243, dropped: 0, current: 59.99, average: 60.19
```

**Second, adding mpph264enc**

```
root@firefly:~# gst-launch-1.0 -v rkcamsrc device=/dev/video1 ! video/x-raw,format=NV12,width=1920,height=1088,framerate=60/1 ! mpph264enc ! fpsdisplaysink name=fpssink text-overlay=false video-sink=fakesink signal-fps-measurements=true sync=false
Setting pipeline to PAUSED ...
mpi: mpp version: a3bdfde0 author: Herman Chen [hal_h264e]: Add gop ref implement to vepu2
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRKCamSrc:rkcamsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 15667200 [14688000 : 16646400]
/GstPipeline:pipeline0/GstMppH264Enc:mpph264enc0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
requesting 2 MMAP buffers/GstPipeline:pipeline0/GstMppH264Enc:mpph264enc0.GstPad:src: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 12, dropped: 0, current: 22.73, average: 22.73
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 24, dropped: 0, current: 21.19, average: 21.93
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 34, dropped: 0, current: 19.44, average: 21.14
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 46, dropped: 0, current: 23.33, average: 21.67
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 57, dropped: 0, current: 21.01, average: 21.54

```
**Last, io-mode of rkcamsrc option changed to dmabuf.**

```
root@firefly:~# gst-launch-1.0 -v rkcamsrc device=/dev/video1 io-mode=dmabuf ! video/x-raw,format=NV12,width=1920,height=1088,framerate=60/1 ! mpph264enc ! fpsdisplaysink name=fpssink text-overlay=false video-sink=fakesink signal-fps-measurements=true sync=false
[... 2 lines cut ...]
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRKCamSrc:rkcamsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 15667200 [14688000 : 16646400]
[... cut ...]
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 48, dropped: 0, current: 23.05, average: 22.89
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 60, dropped: 0, current: 23.15, average: 22.94
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 71, dropped: 0, current: 20.78, average: 22.57
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 83, dropped: 0, current: 22.71, average: 22.59
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 94, dropped: 0, current: 21.70, average: 22.49
```

HermanChen, 2019-07-16:
RK3288's VPU can encoder 1080p@30fps.
Please check VPU's clock frequency.

Use command:
echo 0x100 > /sys/modules/vcodec_service/parameters/debug
to enable kernel driver time log then run the encoder again and check the hardware timing.

doovirus, 2019-07-16:
thanks for reply

i tired the next below command.

```
root@firefly:~# echo 0x100 > /sys/module/rk_vcodec/parameters/debug
root@firefly:~# cat /sys/module/rk_vcodec/parameters/debug
256
```
and i tried the encoder.
so dmesg result is next.
```
[  279.580380] tc35874x 1-000f: subdevice name tc35874x 1-000f
[  279.681660] tc35874x 1-000f: __isp_pipeline_s_isp_clk out
[  279.938001] rk_vcodec: vpu_enc task: 29 ms
[  279.986826] rk_vcodec: vpu_enc task: 21 ms
[  280.040708] rk_vcodec: vpu_enc task: 28 ms
[  280.065371] rk_vcodec: vpu_enc task: 24 ms
[  280.090139] rk_vcodec: vpu_enc task: 24 ms
[  280.144942] rk_vcodec: vpu_enc task: 25 ms
[  280.187423] rk_vcodec: vpu_enc task: 21 ms
[  280.232686] rk_vcodec: vpu_enc task: 21 ms
[  280.274266] rk_vcodec: vpu_enc task: 21 ms
[  280.317153] rk_vcodec: vpu_enc task: 21 ms
[  280.362733] rk_vcodec: vpu_enc task: 21 ms
[  280.405522] rk_vcodec: vpu_enc task: 21 ms
[  280.448790] rk_vcodec: vpu_enc task: 21 ms
[  280.493399] rk_vcodec: vpu_enc task: 21 ms
[  280.533283] rk_vcodec: vpu_enc task: 21 ms
[  280.581342] rk_vcodec: vpu_enc task: 22 ms
[  280.621496] rk_vcodec: vpu_enc task: 21 ms
[  280.666047] rk_vcodec: vpu_enc task: 21 ms
[  280.707587] rk_vcodec: vpu_enc task: 21 ms
[  280.749985] rk_vcodec: vpu_enc task: 21 ms
[  280.796155] rk_vcodec: vpu_enc task: 21 ms
[  280.839238] rk_vcodec: vpu_enc task: 21 ms
[  280.879707] rk_vcodec: vpu_enc task: 21 ms
[  280.921285] rk_vcodec: vpu_enc task: 21 ms
[  280.963775] rk_vcodec: vpu_enc task: 21 ms
[  281.007326] rk_vcodec: vpu_enc task: 21 ms
[  281.057281] rk_vcodec: vpu_enc task: 30 ms
[  281.079537] rk_vcodec: vpu_enc task: 22 ms
[  281.125619] rk_vcodec: vpu_enc task: 22 ms
[  281.170822] rk_vcodec: vpu_enc task: 21 ms
[  281.217860] rk_vcodec: vpu_enc task: 21 ms
[  281.260948] rk_vcodec: vpu_enc task: 21 ms
[  281.305618] rk_vcodec: vpu_enc task: 21 ms
[  281.363015] rk_vcodec: vpu_enc task: 33 ms
[  281.385629] rk_vcodec: vpu_enc task: 22 ms
[  281.431707] rk_vcodec: vpu_enc task: 21 ms
[  281.474653] rk_vcodec: vpu_enc task: 21 ms
[  281.517604] rk_vcodec: vpu_enc task: 21 ms
[  281.566063] rk_vcodec: vpu_enc task: 21 ms
[  281.613023] rk_vcodec: vpu_enc task: 23 ms
[  281.656731] rk_vcodec: vpu_enc task: 22 ms
[  281.699702] rk_vcodec: vpu_enc task: 21 ms
[  281.741401] rk_vcodec: vpu_enc task: 21 ms
[  281.785337] rk_vcodec: vpu_enc task: 21 ms
[  281.833780] rk_vcodec: vpu_enc task: 21 ms
[  281.875466] rk_vcodec: vpu_enc task: 21 ms
[  281.914867] rk_vcodec: vpu_enc task: 21 ms
[  281.968505] rk_vcodec: vpu_enc task: 21 ms
[  282.011660] rk_vcodec: vpu_enc task: 21 ms
[  282.054340] rk_vcodec: vpu_enc task: 21 ms
[  282.107196] rk_vcodec: vpu_enc task: 31 ms
[  282.129368] rk_vcodec: vpu_enc task: 21 ms
[  282.176042] rk_vcodec: vpu_enc task: 22 ms
[  282.216531] rk_vcodec: vpu_enc task: 22 ms
[  282.267488] rk_vcodec: vpu_enc task: 21 ms
[  282.311571] rk_vcodec: vpu_enc task: 21 ms
[... 102 lines cut ...]
[  286.850625] rk_vcodec: vpu_enc task: 22 ms
[  286.893548] rk_vcodec: vpu_enc task: 21 ms
[  286.935757] rk_vcodec: vpu_enc task: 21 ms
[  286.982263] rk_vcodec: vpu_enc task: 25 ms
[  287.023595] rk_vcodec: vpu_enc task: 21 ms
[  287.077291] rk_vcodec: vpu_enc task: 21 ms
[  287.117367] rk_vcodec: vpu_enc task: 21 ms
[  287.160266] rk_vcodec: vpu_enc task: 21 ms
[  287.203960] rk_vcodec: vpu_enc task: 21 ms
[  287.247865] rk_vcodec: vpu_enc task: 21 ms
[  287.291028] rk_vcodec: vpu_enc task: 21 ms
[  287.339641] rk_vcodec: vpu_enc task: 22 ms
[  287.394344] rk_vcodec: vpu_enc task: 31 ms
[  287.419075] rk_vcodec: vpu_enc task: 24 ms
[  287.471346] rk_vcodec: vpu_enc task: 21 ms
[  287.515944] rk_vcodec: vpu_enc task: 21 ms
[  287.559072] rk_vcodec: vpu_enc task: 22 ms
[  287.604383] rk_vcodec: vpu_enc task: 21 ms
[  287.648312] rk_vcodec: vpu_enc task: 21 ms
[  287.699755] tc35874x 1-000f: subdevice name tc35874x 1-000f
```

Does the above result mean that the encoder uses 21ms to encode 1 frame?
The kernel time tag interval looks like an average of 45ms.
In addition to the encoding time of 21ms, I don't know where 24ms is used.

HermanChen, 2019-07-16:
Yes, this is the hardware working time from kernel side.
So the performance limitation will be in userspace driver. Maybe frame copy limit the overall performance.

doovirus, 2019-07-17:
Thanks for reply.
Finally, does mpp support dmabuf of io-mode?
using dmabuf of io-mode in rkcamsrc did not improve performance.

HermanChen, 2019-07-17:
Yes, mpp support dmabuf as its input with no extra copy.

doovirus, 2019-07-17:
Thanks! 
Your reply was a great help to me.
Have a nice day.

HermanChen, 2019-07-17:
You are welcome.
Have a nice day.