rockchip-linux/mpp #105: mpp循环解码mpp_buf_slot: mpp_buf_slot_get_unused failed to get a unused slot问题

lwdddddddd, 2019-07-29:
你好，我写了个测试程序循环读文件进行解码，当解码到第四帧的时候会报如下错误：
**mpp_buf_slot: mpp_buf_slot_get_unused failed to get a unused slot**
循环部分代码为：
`	MppCtx ctx  		= demo->ctx;
        MppApi *mpi 		= demo->mpi;
	MppFrame frame 	= demo->frame;
	MppPacket packet	= NULL;
	
	if (NULL == demo->packet_buffer) {
		if (mpp_buffer_get(demo->packet_group, &demo->packet_buffer, data_size)) {
			printf("error: failed to get packet buffer\n");
			return -1;
		}

		if (mpp_packet_init_with_buffer(&demo->packet, demo->packet_buffer)) {
			printf("error: init packet with buffer failed.\n");
			return -1;
		}
	}

	packet	= demo->packet;

	mpp_buffer_write(demo->packet_buffer, 0, data, data_size);
	mpp_packet_set_pos(packet, mpp_buffer_get_ptr(demo->packet_buffer));
	mpp_packet_set_length(packet, data_size);
	mpp_packet_set_eos(packet);

	MppTask task = NULL;
	if (mpi->poll(ctx, MPP_PORT_INPUT, MPP_POLL_BLOCK)) 
		return -1;
	
	/* input queue */
	if (mpi->dequeue(ctx, MPP_PORT_INPUT, &task) || !task) 
		return -1;
	mpp_assert(task);

	mpp_task_meta_set_packet(task, KEY_INPUT_PACKET, packet);
	mpp_task_meta_set_frame (task, KEY_OUTPUT_FRAME, frame);
	if (mpi->enqueue(ctx, MPP_PORT_INPUT, task)) 
		return -1;	
	
	/* poll and wait here */
	if (mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK)) {
		return -1;
	}

	/* output queue */
	if (mpi->dequeue(ctx, MPP_PORT_OUTPUT, &task) || !task)
		return -1;

	mpp_assert(task);
	if (!task) 
		return -1;

	MppFrame frame_out = NULL;
	if (mpp_task_meta_get_frame(task, KEY_OUTPUT_FRAME, &frame_out) || !frame_out)
		return -1;

	if(mpp_frame_get_eos(frame_out)) {
		msleep(1);
		return -1;
	}
	
	if (mpp_frame_get_errinfo(frame_out) | mpp_frame_get_discard(frame_out)) {
		msleep(1);
		return -1;
	}

	/*dump MppFrame*/
	MppBuffer mbuffer   = mpp_frame_get_buffer(frame_out);
        if (NULL == mbuffer)
    	    return -1;
	mpp_buffer_read(mbuffer, 0, demo->out_frame.data[0], demo->out_frame.data_size);

	/* output queue */
	mpi->enqueue(ctx, MPP_PORT_OUTPUT, task);`
请问，是需要重新初始化MPP实例吗？

HermanChen, 2019-07-30:
初始化一次就可以了

lwdddddddd, 2019-07-31:
好的，谢谢！

closed 2019-07-31 by lwdddddddd