rockchip-linux/mpp #106: rk3399编译mpp后执行mpi_dec_test，错误如下:CANNOT LINK EXECUTABLE "mpi_dec_test": cannot locate symbol "mpp_buffer_group_usage" referenced by "/system/bin/mpi_dec_test"... Aborted,有人遇到过吗?

lindefoe, 2019-08-02:
使用的是firefly的rk3399板子，配合android7.1.1使用
用官方给出的编译方法，使用ndk14的24platform去编译。
使用以下指令将库和执行程序推到板子里
adb push ./mpp/libmpp.so /system/lib/
adb push ./mpp/legacy/libvpu.so /system/lib/
adb push ./test/mpi_dec_test /system/bin/
运行mpi_dec_test时提示以下错误
CANNOT LINK EXECUTABLE "mpi_dec_test": cannot locate symbol "mpp_buffer_group_usage" referenced by "/system/bin/mpi_dec_test"... Aborted,有人遇到过吗?

也使用过挂载的方法，直接在挂载的地址里直接运行，还是报这个错误，请问整么解决呢

HermanChen, 2019-08-05:
3399 是 64 位平台，估计是只更新了 32 位的库和可执行程序，实际运行的库可能还是 64 位的，没有更新，所以运行出错了。
分析下是哪里出问题吧。

lindefoe, 2019-08-05:
> 3399 是 64 位平台，估计是只更新了 32 位的库和可执行程序，实际运行的库可能还是 64 位的，没有更新，所以运行出错了。
> 分析下是哪里出问题吧。

lancoo@lancoo:~/Downloads/mpp-release/build/android/aarch64/mpp$ objdump -a libmpp.so 
libmpp.so：     文件格式 elf64-littleaarch64
libmpp.so
lancoo@lancoo:~/Downloads/mpp-release/build/android/aarch64/mpp$ cd legacy/
lancoo@lancoo:~/Downloads/mpp-release/build/android/aarch64/mpp/legacy$ objdump -a libvpu.so 
libvpu.so：     文件格式 elf64-littleaarch64
libvpu.so

lancoo@lancoo:~/Downloads/mpp-release/build/android/aarch64/test$ file mpi_dec_test 
mpi_dec_test: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /system/, not stripped

###rk3399是linux内核，使用aarch64框架
rk3399_firefly_aio_box:/mnt/sdcard/Download/mpp-release/build/android/arm $ uname -a
Linux localhost 4.4.83 #1 SMP PREEMPT Thu Jul 18 21:48:56 CST 2019 aarch64
我编出来的程序都是64位的，板子的系统也是64位的。库和程序推到板子里或者挂载直接运行不会变成32位的吧？请问是哪里出了问题呢？

HermanChen, 2019-08-05:
64位库要推到 lib64 下面

lindefoe, 2019-08-05:
> 64位库要推到 lib64 下面

恩，我推导64位库中，并把32位lib中的库还原为原来的内容（32位推上去以后就无法正常重启了）
这样重新运行mpi_dec_test和mpi_enc_test都没有输出了，请问这是因为我交叉编译配置的参数问题了吗？
如下（输入命令后直接进入下个命令行，没有数据反馈）
lancoo@lancoo:~/Downloads/mpp-release/build/android/aarch64$ adb shell
rk3399_firefly_aio_lvds_box:/ # mpi_dec_test
1|rk3399_firefly_aio_lvds_box:/ # mpi_enc_test
1|rk3399_firefly_aio_lvds_box:/ # mpi_dec_test -help
1|rk3399_firefly_aio_lvds_box:/ # 

我的系统信息是安卓7.1.1，aarch64架构的
Linux localhost 4.4.83 #1 SMP PREEMPT Thu Jul 18 21:48:56 CST 2019 aarch64

使用的是mpp-release/build/android/aarch64的make-Android.bash
改后的内容如下
#!/bin/bash
# Run this from within a bash shell
HOST_IP=`hostname --all-ip-addresses`
if [ ${HOST_IP} == "10.10.10.65" ] || [ ${HOST_IP} == "10.10.10.67" ]; then
    ANDROID_NDK=/home/lancoo/EnvAndroid/NDK/android-ndk-r14b/
else
    ANDROID_NDK=/home/lancoo/EnvAndroid/NDK/android-ndk-r14b/
fi

PLATFORM=$ANDROID_NDK/platforms/android-24/arch-arm64

cmake -DCMAKE_TOOLCHAIN_FILE=../android.toolchain.cmake                     \
      -DCMAKE_BUILD_TYPE=Release                                            \
      -DANDROID_FORCE_ARM_BUILD=ON                                          \
      -DANDROID_NDK=${ANDROID_NDK}                                          \
      -DANDROID_SYSROOT=${PLATFORM}                                         \
      -DANDROID_ABI="arm64-v8a"                                             \
      -DANDROID_TOOLCHAIN_NAME="aarch64-linux-android-4.9"                  \
      -DANDROID_NATIVE_API_LEVEL=android-24                                 \
      -DANDROID_STL=system                                                  \
      -DRKPLATFORM=ON                                                       \
      -DHAVE_DRM=ON                                                         \
      ../../../

# ----------------------------------------------------------------------------
# usefull cmake debug flag
# ----------------------------------------------------------------------------
      #-DCMAKE_BUILD_TYPE=Debug                                              \
      #-DCMAKE_VERBOSE_MAKEFILE=true                                         \
      #--trace                                                               \
      #--debug-output                                                        \

#cmake --build . --clean-first -- V=1
cmake --build .

# ----------------------------------------------------------------------------
# test script
# ----------------------------------------------------------------------------
#adb push osal/test/rk_log_test /system/bin/
#adb push osal/test/rk_thread_test /system/bin/
#adb shell sync
#adb shell logcat -c
#adb shell rk_log_test
#adb shell rk_thread_test
                                                                                                                                     1,1          顶端
万分感谢！！！！！

lindefoe, 2019-08-06:
> 64位库要推到 lib64 下面

您好，请问您知道这个怎么解决吗？

closed 2019-08-07 by lindefoe

jutyy, 2021-05-29:
> 64位库要推到 lib64 下面

请问下mmp有没有 32位的库