rockchip-linux/mpp #112: hal_h264e_api: vcodec type 00000000 can not find H.264 encoder device

grkblood13, 2019-09-13:
I opened this same issue in the gstreamer-rockchip repo, but realized this is probably a better place.

OS: Manjaro-ARM 19.09
kernel: Linux localhost 5.3.0-rc4-1-udl #8 SMP Mon Sep 2 16:29:02 EDT 2019 aarch64 GNU/Linux

First, I build mpp with the following:

```
cmake -DRKPLATFORM=ON -DHAVE_DRM=ON -DCMAKE_INSTALL_PREFIX=/usr
make -j4
sudo make install
```

I can then install gstreamer and build and gstreamer-rockhip from source, but when I try to run mpph264 against an input it fails. I can verify a successful compilation of the encoder via the following:

```
[rock64@localhost tmp]$ gst-inspect-1.0 mpph264enc
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                Rockchip Mpp H264 Encoder
  Klass                    Codec/Encoder/Video
  Description              Encode video streams via Rockchip Mpp
  Author                   Randy Li <randy.li@rock-chips.com>

Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.12.0
  License                  GPL
  Source module            gstreamer-rockchip
  Binary package           GStreamer Plug-ins source release
  Origin URL               Unknown package origin
```

When attempting to run the mpph264enc plugin against a verified working Hauppauge WinTV-HVR-950Q USB tuner I get the following.

```
[rock64@localhost tmp]$ gst-launch-1.0 -v dvbsrc frequency=177000000 delsys=atsc modulation=8vsb pids=49:52:1 ! decodebin ! queue ! mpph264enc ! filesink location=test.mkv
Setting pipeline to PAUSED ...
mpi: mpp version: 451ae59 author: syzby [cmake]: Fix aarch linux cmake compile error
hal_h264e_api: vcodec type 00000000 can not find H.264 encoder device
mpp_hal: mpp_hal_init hal h264e_rkvenc init failed ret -1
mpp_hal: mpp_hal_init could not found coding type 7
mpp_enc: mpp_enc_init could not init hal
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpp: error found on mpp initialization
```

Could someone help me resolve this?

HermanChen, 2019-09-16:
This is a kernel driver issue. rkvenc is only on RV1108 chipset with /dev/rkvenc device in kernel.
The mpp can not be found in this platfrom. Please check the platfrom you are using.

grkblood13, 2019-09-17:
I'm running a Pine Rock64. When using ayufan's kernel (4.4.190-1233-rockchip-ayufan-gd3f1be0ed310) I get /dev/rkvdec but no /dev/rkvenc.

WainDing, 2019-09-17:
/dev/rkvenc  only used in soc rv1108， check whether have /dev/vepu ?

grkblood13, 2019-09-17:
Yes, /dev/vepu is present

grkblood13, 2019-09-18:
I captured the attached sample file using:
```
$ gst-launch-1.0 -v dvbsrc frequency=177000000 delsys=atsc modulation=8vsb pids=49:52:1 ! filesink location=test.raw
```
download link: https://gofile.io/?c=YgrqVF

To play file:
```
$ ffplay test.raw
```

Output of ffplay:
```
$ ffplay test.raw 
ffplay version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2003-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
[mpeg2video @ 0x7f50b8003fc0] Invalid frame dimensions 0x0. f=0/0   
    Last message repeated 5 times
Input #0, mpegts, from 'test.raw': 0KB vq=    0KB sq=    0B f=0/0   
  Duration: 00:00:11.71, start: 63428.952322, bitrate: 11815 kb/s
    Stream #0:0[0x31]: Video: mpeg2video (Main), yuv420p(tv, bt709, top first), 1920x1080 [SAR 1:1 DAR 16:9], Closed Captions, 29.97 fps, 29.97 tbr, 90k tbn, 59.94 tbc
    Stream #0:1[0x34]: Audio: ac3, 48000 Hz, 5.1(side), fltp, 384 kb/s
63435.66 A-V: -0.009 fd=   0 aq=   51KB vq= 2798KB sq=    0B f=0/0   

```

When trying to convert to h264 using mpph264enc the following occurs:
```
gst-launch-1.0 filesrc location=/tmp/test.raw ! decodebin ! queue ! mpph264enc ! fakesink silent=false
Setting pipeline to PAUSED ...
mpi: mpp version: 451ae59 author: syzby [cmake]: Fix aarch linux cmake compile error
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
0:00:00.088600355  3466   0x559a57fb80 WARN                 basesrc gstbasesrc.c:3600:gst_base_src_start_complete:<filesrc0> pad not activated yet
Pipeline is PREROLLING ...
0:00:00.144065627  3466   0x7f9c0550f0 WARN              mpegtsbase mpegtsbase.c:1641:mpegts_base_loop:<tsdemux0> error: Internal data stream error.
0:00:00.144181417  3466   0x7f9c0550f0 WARN              mpegtsbase mpegtsbase.c:1641:mpegts_base_loop:<tsdemux0> error: No program activated before EOS
ERROR: from element /GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTSDemux:tsdemux0: Internal data stream error.
Additional debug info:
../gst-plugins-bad/gst/mpegtsdemux/mpegtsbase.c(1641): mpegts_base_loop (): /GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTSDemux:tsdemux0:
No program activated before EOS
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
```

AndreVallestero, 2019-12-08:
I'm not sure if this is relevant but usually ffplay/ffmpeg is ompiled with the --rkmpp flag to support MPP. It looks like your ffplay was not compiled with this flag. Also, you usually have to specify -vcodec h264_rkmpp with ffmpeg or ffplay to specify decoding using mpp h264.