rockchip-linux/mpp #113: color distortion with rockchip mpp vp8 hw decoder

naseebpanghal, 2019-10-10:
I am using mpp library to decode vp8 data, color distortion starts happening around the static part in the frame and then spread through the whole frame.

H264 part works perfect.
Is there anyone who has seen this kind of behaviour with VP8? Is there any specific setting needs to be done for VP8 in mpp?

qvoid, 2019-10-10:
Could you upload the VP8 stream? So we can analyze it.

qvoid, 2019-10-11:
The file you provided can't be played by FFmpeg. Maybe it is lack of webm headers. 
If you have some problem saving the VP8 stream, you can try the command below before start playing VP8 stream:
```
setprop mpp_debug 0x200
```
This will save the input stream providing to mpp under */data*. It is named as *mpp_dec_in_<pid>.bin*.

naseebpanghal, 2019-10-11:
> setprop mpp_debug 0x200

I am sorry, i forgot to mention i am using ubuntu on rk3399 machine for my testing.
what is alternative for setprop on ubuntu? If i can set mpp_debug in mpp library, i can recompile mpp?

naseebpanghal, 2019-10-11:
I think, ffmpeg on my machine is not able to understand webm stream. i found below finding
$ ffmpeg -formats | grep webm
 D  matroska,webm   Matroska / WebM
  E webm            WebM
  E webm_chunk      WebM Chunk Muxer
 DE webm_dash_manifest WebM DASH Manifest

Means default install doesn't provide webm decoder but encoder.
I am downloading ffmpeg source and check if i can enable webm decoder through configuration.

qvoid, 2019-10-11:
Use cmd : 
```
export mpp_debug=0x200
```
 instead of **setprop** on Ubuntu. 
Sorry, this won't work for VP8 decoding. You may have to add webm header before dumping into files. And the ffmpeg you install is fine with webm demuxing, since **D** means *demux*

naseebpanghal, 2019-10-11:
There are two more thing to share:
1. Just before feeding stream to mpp decoder, if i redirect to intel decoder(another machine), there same stream works perfect.
2. If i use software decoder on rk3399 then also it works perfect. Color distortion happens with mpp.

As you suggested, I will try to dump data into file with webm headers.

naseebpanghal, 2019-10-11:
> 2\. Seems like those decoder store information internally which mpp decoder expects with each frame or at regular interval.

I have faced similar kind of issue with H264 mpp decoder.
I have some H264 encoders which send SPS/PPS only once just before first IDR, hence i parsed h264 stream and fed SPS/PPS to mpp decoder before every IDR frame. If i feed stream as it is, mpp decoder was giving some issue(there isssue was different than color distotion. i don't remember exactly, if you need i can recheck).

naseebpanghal, 2019-10-15:
@qvoid i have dumped the vp8 stream into proper webm file using ffmpeg av_write_frame().
ffplay/vlc are able to play this file properly without any color distortion, but mpp_decoder shows color distortion.
(Note: just not lead to any confusion, i have dumped data into this file just before feeding to mpp_decoder. So ffplay/vlc or mpp_decoder gets same data. mpp_decoder distort color but other decoder doesn't). 
[dump.zip](https://github.com/rockchip-linux/mpp/files/3727809/dump.zip)

qvoid, 2019-10-17:
I test the webm file you provided on my RK3328 board. The decoder works normally. 
Try to feed MPP VP8 decoder with complete one frame of data each time, since MPP VP8 decoder expects that each packet is a complete frame.

naseebpanghal, 2019-10-17:
> Try to feed MPP VP8 decoder with complete one frame of data each time, since MPP VP8 decoder expects that each packet is a complete frame.

Thanks for your prompt response.
I feed data as you suggest.

I checked it on my another board. i.e RK3288. Thankfully there VP8 works normal way.
when i compared mpp version on my both the boards, i found:
RK3399: color distortion
mpi: mpp version: 0b701fe author: Lawrence-Tang DEBIAN: prepare for release_20171218-3
RK3288: color normal
mpi: mpp version: 598cae3 author: Jacob Chen DEBIAN: update rules for release_20171218-2

Now i am trying to get same version as RK3288 on RK3399 and hope it should work there.

naseebpanghal, 2019-10-18:
i have used same mpp version 451ae593 on both the boards but RK3399 still giving color distortion and RK3288 working fine.

1. i have checked my application thoroughly, there is no path deviation in 32-bit or 64-bit compilation.
2. Now userspace librockchip_mpp.so library is same as well.

Please can someone shed some light what can be the reason. 
1. can it be kernel space issue?

@qvoid , please try, webm file i shared, on RK3399 board as well if possible.

qvoid, 2019-10-24:
Sorry for doing this test late.
I do test is on My RK3399 Android 9.0 board. Things work fine. 
If the mpp library and player applications are the same between your RK3399 board and RK3288 board. Maybe it is the kernel space issue. You can try to compare files under *kerne/drive/video/rockchip/vcodec*.

naseebpanghal, 2019-10-31:
> Sorry for doing this test late.
> I do test is on My RK3399 Android 9.0 board. Things work fine.

Thanks for doing the test.
At least i am relaxed as issue is not related to hardware.
i will focus on kernel side now to identify the root cause.

naseebpanghal, 2019-10-31:
> I do test is on My RK3399 Android 9.0 board. Things work fine.

@qvoid , may i know which player you used to play webm file on Android RK3399?

qvoid, 2019-11-04:
> > I do test is on My RK3399 Android 9.0 board. Things work fine.
> 
> @qvoid , may i know which player you used to play webm file on Android RK3399?

The built-in media player. And also tested with APP [jkMediaplyer](https://github.com/bilibili/ijkplayer) using AndroidMediaPlayer (change the default one to AndroidMediaPlayer in settings)

naseebpanghal, 2019-11-04:
@qvoid , i tried the .webm file on the same board with Android 8 and it shows distorted image here as well.
1. I am using FriendlyElec NanoPC-T4(RK3399) board. May i know which board(vendor) you used to test .webm file?
2. Did you built Android 9.0 image yourself for RK3399?

naseebpanghal, 2019-11-04:
I have tried to play .webm sample on another vendor board as well i.e Orange Pi RK3399 with Ubuntu image. There as well i face distortion.

qvoid, 2019-11-05:
The below commit fixed a bug on VP8 decoding.

> commit 662b64c4eed0e41ab746a25ea8cea60e300d415e
Author: Johnson Ding <johnson.ding@rock-chips.com>
Date:   Mon Jul 8 16:41:44 2019 +0800

    [vp8d]: Fix assigning wrong DCT address problem
    
    Refer to this issue:
    https://bugs.chromium.org/p/chromium/issues/detail?id=979091

I don't know if this has something to do with problem you are facing. But you can try to checkout the  newest mpp source code and compile it. You may need to push both libmpp.so and libvpu.so to the correct directories. And restart media service to make them take effect.
On Android platform, you can try this:
```
setprop use_mpp_mode 0
setprop use_original 1
```
and play your webm file. This make the decoder calling the old apis.
and switch back to mpp api using:
```
setprop use_mpp_mode 1
setprop use_original 0
```
If things are the same, capture a screen shot and post it here. also:
```
echo 0xffff > /sys/module/rk_vcodec/parameters/debug
```
and play your webm file and post kernel log here.
Besides, going to your kernel source code, and post your here branch and the recent commit log about *drivers/video/rockchip/vcodec*, you can use this command:
```
git log --pretty=oneline kernel/drivers/video/rockchip/vcodec
```

naseebpanghal, 2019-11-05:
> The below commit fixed a bug on VP8 decoding.
> 
I tried mpp library with commit you mentioned by replacing mpp.so and vpu.so but still color distorts. 

> On Android platform, you can try this:
> 
> ```
> setprop use_mpp_mode 0
> setprop use_original 1
> ```
I don't find any setprop utility on my Android board. Tried to install using pkg install but couldn't find package containing this utility.

> ```
> echo 0xffff > /sys/module/rk_vcodec/parameters/debug
> ```
> 
> and play your webm file and post kernel log here.

I couldn't update file /sys/module/rk_vcodec/parameters/debug, Error Permission denied even with sudo.
I tried it on both Android and Ubuntu.

> Besides, going to your kernel source code, and post your here branch and the recent commit log about _drivers/video/rockchip/vcodec_, you can use this command:
> 
> ```
> git log --pretty=oneline kernel/drivers/video/rockchip/vcodec
> ```
I am using image provided by Vendor, i don't have build tree.

I am attaching screenshot of videoplayer from Android.
You can see Green color(top right corner) near Close button of VLC. and Color distortion on status bar below(whole strip color is distorted.)

naseebpanghal, 2019-11-07:
> @qvoid , I am sorry if i am bothering you with so many questions. I need to resolve this issue.
> 
> I tried the .webm file on the same board with Android 8 and it shows distorted image here as well.
> 
> 1. I am using FriendlyElec NanoPC-T4(RK3399) board. May i know which board(vendor) you used to test .webm file?
> 2. Did you built Android 9.0 image yourself for RK3399?

naseebpanghal, 2019-11-07:
closed by mistake.

naseebpanghal, 2019-11-07:
> @qvoid , I am sorry if i am bothering you with so many questions. I need to resolve this issue.
>
> I tried the .webm file on the same board with Android 8 and it shows distorted image here as well. [...]

@qvoid , please let me know which vendor board(RK3399) you are using. I will try it on that board as well.

qvoid, 2019-11-18:
Sorry. It's my fault that did not watch the output image of player carefully. This problem does happen in my RK3399. This is [the fix to this problem](https://github.com/rockchip-linux/mpp/commit/27bbe9ff79e56b15944d9ec931a01a8359c0f59f).
Thank you for your report and patience.

naseebpanghal, 2019-11-20:
@qvoid  I tried the fix at my board and it worked. 
Thanks for your help in solving the issue. 

By the way, the day you provided the fix, it happened to be my B'day and i consider it best b'day gift :)

qvoid, 2019-11-20:
@naseebpanghal It's my pleasure.

closed 2019-11-21 by naseebpanghal