rockchip-linux/mpp #121: CMakeLists.txt looks for "armv8-a" CMAKE_SYSTEM_PROCESSOR, should look for "aarch64"

ydirson, 2019-11-26:
Although I could not find a definitive reference about well-known CMAKE_SYSTEM_PROCESSOR values:
- yocto sets CMAKE_SYSTEM_PROCESSOR to "aarch64"
- google finds tons of matches for "CMAKE_SYSTEM_PROCESSOR aarch64", whereas matches for "CMAKE_SYSTEM_PROCESSOR armv8-a" does not even seem to show a single pertinent match on first page

AndreVallestero, 2019-12-08:
I've encountered the same issue. I've been working around it using the following:

sed -i 's/${SYSPROC} STREQUAL "armv8-a"/${SYSPROC} STREQUAL "armv8-a" OR ${SYSPROC} STREQUAL "aarch64"/g' CMakeLists.txt

qvoid, 2019-12-10:
@ydirson @AndreVallestero  Refer to *android-ndk-r16b/build/cmake/android.toolchain.cmake*, When **ANDROID_ABI** equals to "arm64-v8a", "CMAKE_SYSTEM_PROCESOR" should be set as "aarch64" :
```
elseif(ANDROID_ABI STREQUAL arm64-v8a)
  set(ANDROID_SYSROOT_ABI arm64)
  set(CMAKE_SYSTEM_PROCESSOR aarch64)
  set(ANDROID_TOOLCHAIN_NAME aarch64-linux-android)
  set(ANDROID_TOOLCHAIN_ROOT ${ANDROID_TOOLCHAIN_NAME})
  set(ANDROID_LLVM_TRIPLE aarch64-none-linux-android)
  set(ANDROID_HEADER_TRIPLE aarch64-linux-android)
```
"aarch64" should be the right.
The patch will be uploaded soon.