rockchip-linux/mpp #124: RK3399 H.264 Encode Failure

ram95014, 2019-12-03:
I am running "Ubuntu 18.04.1 LTS" with "Linux linux 4.4.154-59-rockchip-g5e70f14 #4 SMP Fri Dec 14 20:55:41 CST 2018 aarch64 aarch64 aarch64 GNU/Linux".
On a simple h.264 encode test with a set of nv12 frame, I get the following error:

> mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
> hal_h264e_vepu2: mpp_device_send_reg failed ret 14
> mpp_enc: mpp 0x55809fb880 hal_hw_start failed return 14

with this command:

> ./mpi_enc_test -i ~/out.nv12 -o hurr.264 -w 1280 -h 720 -f 0 -t 7 
> mpi_enc_test: cmd parse result:
> mpi_enc_test: input  file name: /home/rock/out.nv12
> mpi_enc_test: output file name: hurr.264
> mpi_enc_test: width      : 1280
> mpi_enc_test: height     : 720
> mpi_enc_test: format     : 0
> mpi_enc_test: type       : 7
> mpi_enc_test: debug flag : 0
> mpi_enc_test: mpi_enc_test start
> mpp_rt: NOT found ion allocator
> mpp_rt: found drm allocator
> mpi_enc_test: mpi_enc_test encoder test start w 1280 h 720 type 7
> mpi: mpp version: b395f5fa author: Herman Chen [build]: Add path detection for different ndk
> mpi_enc_test: mpi_enc_test bps 3456000 fps 30 gop 60
>

AndreVallestero, 2019-12-08:
Does decode work fine for you? I'm getting s similar error with rk3399 h264 decoding.
A vcodec assertion error followed by a ioctl segmentation fault.

ram95014, 2019-12-11:
Sorry, didn't test the decoder.

HermanChen, 2020-03-06:
Could you please update mpp to latest version and test again?

jveitchmichaelis, 2020-03-25:
I get this inside Docker, but not outside. Both running Debian Stretch, RK3399.

The versions of mpp are clearly different though. Installed with:

```
RUN git clone https://github.com/rockchip-linux/mpp.git
RUN cd mpp/build && cmake .. && make -j6 && make install
RUN echo "LD_LIBRARY_PATH=/usr/local/lib/aarch64-linux-gnu:/usr/local/lib:$LD_LIBRARY_PATH" >> /root/.bashrc
```
Within the container (latest develop):

```
mpi_enc_test -i demo.mkv -o demo.264 -w 1280 -h 720 -f 0 -t 7 
mpp[513]: mpi_enc_test: cmd parse result:
mpp[513]: mpi_enc_test: cmd parse result:

mpp[513]: mpi_enc_test: input  file name: demo.mkv
mpp[513]: mpi_enc_test: input  file name: demo.mkv

mpp[513]: mpi_enc_test: output file name: demo.264
mpp[513]: mpi_enc_test: output file name: demo.264

mpp[513]: mpi_enc_test: width      : 1280
mpp[513]: mpi_enc_test: width      : 1280

mpp[513]: mpi_enc_test: height     : 720
mpp[513]: mpi_enc_test: height     : 720

mpp[513]: mpi_enc_test: format     : 0
mpp[513]: mpi_enc_test: format     : 0

mpp[513]: mpi_enc_test: type       : 7
mpp[513]: mpi_enc_test: type       : 7

mpp[513]: mpi_enc_test: debug flag : 0
mpp[513]: mpi_enc_test: debug flag : 0

mpp[513]: mpi_enc_test: mpi_enc_test start
mpp[513]: mpi_enc_test: mpi_enc_test start

mpp[513]: mpi_enc_test: failed to open input file demo.mkv
mpp[513]: mpi_enc_test: failed to open input file demo.mkv

mpp[513]: mpi_enc_test: create default yuv image for test
mpp[513]: mpi_enc_test: create default yuv image for test

mpp[513]: mpp_rt: NOT found ion allocator
mpp[513]: mpp_rt: NOT found ion allocator

mpp[513]: mpp_rt: found drm allocator
mpp[513]: mpp_rt: found drm allocator

mpp[513]: mpi_enc_test: mpi_enc_test encoder test start w 1280 h 720 type 7
mpp[513]: mpi_enc_test: mpi_enc_test encoder test start w 1280 h 720 type 7

mpp[513]: mpp_info: mpp version: e9c5eae5 author: Johnson Ding  2020-03-23 [rc]: abort on debug mode if GOP is negative
mpp[513]: mpp_info: mpp version: e9c5eae5 author: Johnson Ding  2020-03-23 [rc]: abort on debug mode if GOP is negative

mpp[513]: mpi_enc_test: mpi_enc_test bps 3456000 fps 30 gop 60
mpp[513]: mpi_enc_test: mpi_enc_test bps 3456000 fps 30 gop 60

mpp[513]: h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 3456000 [3240000 : 3672000]
mpp[513]: h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 3456000 [3240000 : 3672000]

mpp[513]: mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[513]: mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address

```

with latest release branch:
[... 70 lines cut ...]
mpi_enc_test: test_mpp_run encoded frame 4 size 11512
mpi_enc_test: test_mpp_run encoded frame 5 size 17820
mpi_enc_test: test_mpp_run encoded frame 6 size 11473
mpi_enc_test: test_mpp_run encoded frame 7 size 14995
mpi_enc_test: test_mpp_run encoded frame 8 size 12984
mpi_enc_test: test_mpp_run encoded frame 9 size 14723
mpi_enc_test: test_mpp_run encoded frame 10 size 13770
mpi_enc_test: test_mpp_run encoded frame 11 size 13822
mpi_enc_test: test_mpp_run encoded frame 12 size 12451
mpi_enc_test: test_mpp_run encoded frame 13 size 14422
mpi_enc_test: test_mpp_run encoded frame 14 size 14935
mpi_enc_test: test_mpp_run encoded frame 15 size 16329
mpi_enc_test: test_mpp_run encoded frame 16 size 14263
mpi_enc_test: test_mpp_run encoded frame 17 size 14268
mpi_enc_test: test_mpp_run encoded frame 18 size 13878
mpi_enc_test: test_mpp_run encoded frame 19 size 13389
mpi_enc_test: test_mpp_run encoded frame 20 size 12669
mpi_enc_test: test_mpp_run encoded frame 21 size 12600
mpi_enc_test: test_mpp_run encoded frame 22 size 11644
mpi_enc_test: test_mpp_run encoded frame 23 size 11408
mpi_enc_test: test_mpp_run encoded frame 24 size 10323
mpi_enc_test: test_mpp_run encoded frame 25 size 10433
mpi_enc_test: test_mpp_run encoded frame 26 size 10723
mpi_enc_test: test_mpp_run encoded frame 27 size 10095
```

HermanChen, 2020-03-26:
mpp required some kernel device to run the encoding / decoding process.
It is the point?

jveitchmichaelis, 2020-03-26:
Small update. On the host system, those packages are installed like this: https://github.com/rockchip-linux/rk-rootfs-build/blob/0cddc6e041d13d7ad91463a3b152952cc4ddcff3/mk-rootfs-stretch.sh#L107

I've replicated that inside my container, and it works:

```
docker run --rm -it --privileged 45809b54972b
root@2b5974c4bc80:/# mpi_enc_test -o demo.264 -w 1280 -h 720 -f 0 -t 7
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: 
mpi_enc_test: output file name: demo.264
mpi_enc_test: width      : 1280
mpi_enc_test: height     : 720
mpi_enc_test: format     : 0
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 1280 h 720 type 7
mpi: mpp version: 1614c5d4 author: Caesar Wang debian: add rules for mpp with 20191031
mpi_enc_test: mpi_enc_test bps 3456000 fps 30 gop 60
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 3456000 [3240000 : 3672000]
mpi_enc_test: test_mpp_run encoded frame 0 size 23399
mpi_enc_test: test_mpp_run encoded frame 1 size 9314
mpi_enc_test: test_mpp_run encoded frame 2 size 11413
mpi_enc_test: test_mpp_run encoded frame 3 size 15133
mpi_enc_test: test_mpp_run encoded frame 4 size 11512
mpi_enc_test: test_mpp_run encoded frame 5 size 17820
mpi_enc_test: test_mpp_run encoded frame 6 size 11473
mpi_enc_test: test_mpp_run encoded frame 7 size 14995
mpi_enc_test: test_mpp_run encoded frame 8 size 12984
mpi_enc_test: test_mpp_run encoded frame 9 size 14723
```

If i then clone this repo and build the develop branch it also works.

For reference:

```
RUN git clone https://github.com/rockchip-linux/rk-rootfs-build

RUN mkdir /usr/include/rga
RUN cp rk-rootfs-build/packages/arm64/rga/include/* /usr/include/rga/
RUN cp rk-rootfs-build/packages/arm64/rga/lib/librga.so /usr/lib/

RUN apt-get update --fix-missing && apt-get -y install gstreamer1.0-plugins-base gstreamer1.0-tools gstreamer1.0-plugins-base-apps
RUN dpkg -i rk-rootfs-build/packages/arm64/gst-base/gstreamer1.0-x_1.10.4-1_arm64.deb
RUN dpkg -i rk-rootfs-build/packages/arm64/video/mpp/*.deb
RUN dpkg -i rk-rootfs-build/packages/arm64/gst-rkmpp/*.deb
#RUN dpkg -i rk-rootfs-build/packages/arm64/ffmpeg/*.deb
RUN apt-get update --fix-missing && apt-get -y install ffmpeg
RUN git clone https://github.com/rockchip-linux/mpp.git
RUN cd mpp/build && cmake .. && make -j6 && make install && ldconfig
```

and then ensure to run the container as `--privileged`

@ram95014 have a look in https://github.com/rockchip-linux/rk-rootfs-build in the arm64 packages and try installing the debfiles in /video/mpp folder

jveitchmichaelis, 2020-03-27:
It definitely doesn't work if I build from source, actually. I get the same error, but the older package provided by rockchip works. I think there's possibly a regression somewhere.

HermanChen, 2020-03-28:
mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
There should be some error return from kernel.
Try dmesg to show the kernel log?
I will find some kernel colleagues to help me.

jveitchmichaelis, 2020-03-28:
Thanks! Here's the output from dmesg:

```
[  939.940483] rk_vcodec: vpu_service_ioctl:1875: error: unknown vpu service ioctl cmd 40086c01
[  939.979613] rk_vcodec: reg_init:1248: error: translate reg address failed, dumping regs
[  939.980342] rk-vcodec ff650000.vpu_service: reg[00]: 00000000
[  939.980864] rk-vcodec ff650000.vpu_service: reg[01]: 00000000
[  939.981384] rk-vcodec ff650000.vpu_service: reg[02]: 00000000
[  939.981904] rk-vcodec ff650000.vpu_service: reg[03]: 00000000
[  939.982622] rk-vcodec ff650000.vpu_service: reg[04]: 00000000
[  939.983159] rk-vcodec ff650000.vpu_service: reg[05]: 00000000
[  939.983673] rk-vcodec ff650000.vpu_service: reg[06]: 00000000
[  939.984194] rk-vcodec ff650000.vpu_service: reg[07]: 00000000
[  939.984715] rk-vcodec ff650000.vpu_service: reg[08]: 00000000
[  939.985239] rk-vcodec ff650000.vpu_service: reg[09]: 00000000
[  939.985760] rk-vcodec ff650000.vpu_service: reg[10]: 00000000
[  939.986325] rk-vcodec ff650000.vpu_service: reg[11]: 00000000
[  939.986846] rk-vcodec ff650000.vpu_service: reg[12]: 00000000
[  939.987366] rk-vcodec ff650000.vpu_service: reg[13]: 00000000
[  939.987887] rk-vcodec ff650000.vpu_service: reg[14]: 00000000
[  939.988397] rk-vcodec ff650000.vpu_service: reg[15]: 00000000
[  939.988918] rk-vcodec ff650000.vpu_service: reg[16]: 00000000
[  939.989438] rk-vcodec ff650000.vpu_service: reg[17]: 00000000
[  939.990042] rk-vcodec ff650000.vpu_service: reg[18]: 00000000
[  939.990564] rk-vcodec ff650000.vpu_service: reg[19]: 00000000
[  939.991084] rk-vcodec ff650000.vpu_service: reg[20]: 00000000
[  939.991604] rk-vcodec ff650000.vpu_service: reg[21]: 00000000
[  939.992124] rk-vcodec ff650000.vpu_service: reg[22]: 00000000
[  939.992650] rk-vcodec ff650000.vpu_service: reg[23]: 00000000
[  939.993160] rk-vcodec ff650000.vpu_service: reg[24]: 00000000
[  939.993681] rk-vcodec ff650000.vpu_service: reg[25]: 00000000
[  939.994240] rk-vcodec ff650000.vpu_service: reg[26]: 00000000
[  939.994762] rk-vcodec ff650000.vpu_service: reg[27]: 00000000
[  939.995282] rk-vcodec ff650000.vpu_service: reg[28]: 00000000
[  939.995802] rk-vcodec ff650000.vpu_service: reg[29]: 00000000
[  939.996321] rk-vcodec ff650000.vpu_service: reg[30]: 00000000
[  939.996841] rk-vcodec ff650000.vpu_service: reg[31]: 00000000
[  939.997361] rk-vcodec ff650000.vpu_service: reg[32]: 00000000
[  939.997880] rk-vcodec ff650000.vpu_service: reg[33]: 00000000
[  939.998450] rk-vcodec ff650000.vpu_service: reg[34]: 00000000
[  939.998971] rk-vcodec ff650000.vpu_service: reg[35]: 00000000
[  939.999491] rk-vcodec ff650000.vpu_service: reg[36]: 00000000
[  940.000011] rk-vcodec ff650000.vpu_service: reg[37]: 00000000
[  940.000531] rk-vcodec ff650000.vpu_service: reg[38]: 00000000
[  940.001052] rk-vcodec ff650000.vpu_service: reg[39]: 00000000
[  940.001573] rk-vcodec ff650000.vpu_service: reg[40]: 00000000
[  940.002130] rk-vcodec ff650000.vpu_service: reg[41]: 00000000
[  940.002652] rk-vcodec ff650000.vpu_service: reg[42]: 00000000
[  940.003173] rk-vcodec ff650000.vpu_service: reg[43]: 00000000
[  940.003694] rk-vcodec ff650000.vpu_service: reg[44]: 00000000
[  940.004203] rk-vcodec ff650000.vpu_service: reg[45]: 00000000
[  940.004724] rk-vcodec ff650000.vpu_service: reg[46]: 44447878
[  940.005245] rk-vcodec ff650000.vpu_service: reg[47]: 00000000
[  940.005755] rk-vcodec ff650000.vpu_service: reg[48]: 00000000
[  940.006315] rk-vcodec ff650000.vpu_service: reg[49]: 7f800000
[  940.006836] rk-vcodec ff650000.vpu_service: reg[50]: 00000000
[  940.007356] rk-vcodec ff650000.vpu_service: reg[51]: 00000000
[  940.007877] rk-vcodec ff650000.vpu_service: reg[52]: 00000000
[  940.008397] rk-vcodec ff650000.vpu_service: reg[53]: 0003fc00
[  940.008918] rk-vcodec ff650000.vpu_service: reg[54]: 00001000
[  940.009438] rk-vcodec ff650000.vpu_service: reg[55]: 00000002
[  940.010028] rk-vcodec ff650000.vpu_service: reg[56]: 00000003
[  940.010549] rk-vcodec ff650000.vpu_service: reg[57]: 7f800003
[  940.011059] rk-vcodec ff650000.vpu_service: reg[58]: 00000000
[  940.011579] rk-vcodec ff650000.vpu_service: reg[59]: 10130000
[  940.012098] rk-vcodec ff650000.vpu_service: reg[60]: 0000ff00
[  940.012618] rk-vcodec ff650000.vpu_service: reg[61]: 00000780
[  940.013138] rk-vcodec ff650000.vpu_service: reg[62]: 00000000
[  940.013658] rk-vcodec ff650000.vpu_service: reg[63]: 00000002
[  940.014230] rk-vcodec ff650000.vpu_service: reg[64]: 7f800002
[... 95 lines cut ...]
[  940.063991] rk-vcodec ff650000.vpu_service: reg[160]: 0d0d0d0d
[  940.064511] rk-vcodec ff650000.vpu_service: reg[161]: 0d0d0d0d
[  940.065030] rk-vcodec ff650000.vpu_service: reg[162]: 0d0d0d0d
[  940.065549] rk-vcodec ff650000.vpu_service: reg[163]: 0d0d0d0d
[  940.066096] rk-vcodec ff650000.vpu_service: reg[164]: 0d0d0d0d
[  940.066617] rk-vcodec ff650000.vpu_service: reg[165]: 0d0d0d0d
[  940.067136] rk-vcodec ff650000.vpu_service: reg[166]: 0d0d0d0d
[  940.067656] rk-vcodec ff650000.vpu_service: reg[167]: 0d0d0d0d
[  940.068176] rk-vcodec ff650000.vpu_service: reg[168]: 0f0f0f0f
[  940.068696] rk-vcodec ff650000.vpu_service: reg[169]: 0f0f0f0f
[  940.069215] rk-vcodec ff650000.vpu_service: reg[170]: 0f0f0f0f
[  940.069735] rk-vcodec ff650000.vpu_service: reg[171]: 0f0f0f0f
[  940.070273] rk-vcodec ff650000.vpu_service: reg[172]: 0f0f0f0f
[  940.070794] rk-vcodec ff650000.vpu_service: reg[173]: 0f0f0f0f
[  940.071313] rk-vcodec ff650000.vpu_service: reg[174]: 0f0f0f0f
[... cut ...]
[  940.072353] rk-vcodec ff650000.vpu_service: reg[176]: 0f0f0f0f
[  940.072873] rk-vcodec ff650000.vpu_service: reg[177]: 0f0f0f0f
[  940.073392] rk-vcodec ff650000.vpu_service: reg[178]: 0f0f0f0f
[  940.073930] rk-vcodec ff650000.vpu_service: reg[179]: 0f0f0f0f
[  940.074451] rk-vcodec ff650000.vpu_service: reg[180]: 0f0f0f0f
[  940.074971] rk-vcodec ff650000.vpu_service: reg[181]: 0f0f0f0f
[  940.075490] rk-vcodec ff650000.vpu_service: reg[182]: 0f0f0f0f
[  940.076009] rk-vcodec ff650000.vpu_service: reg[183]: 0f0f0f0f
```