rockchip-linux/mpp #125: H264 Encoder Fails on RK3288

jbloomfield2008, 2020-02-21:
The current version of MPP fails when encoding H264 on a RK3288. The errors returned by MPP are as follows:

hal_h264e_vepu2: hardware returns error:-1
mpp_enc: mpp 0x5598c69000 hal_hw_wait failed return -1009

The error was introduced post commit a9d6b3a994662411d1ac4cad1fbbd6e4a0831fec. Up to this version H264 encoding works as expected.

HermanChen, 2020-02-21:
The commit a9d6b3a has not relation with encoder. So it maybe some other commit?

HermanChen, 2020-02-21:
Can you try my branch https://github.com/HermanChen/mpp/ to check this branch has the same error or not?
Alsocan you provide the full log?

jbloomfield2008, 2020-02-21:
Agreed the commit is not related to the encoded and building mpp at that revision works, so it's a commit just after a9d6b3a994662411d1ac4cad1fbbd6e4a0831fec that introduced the error. The kernel logs are:

Feb 21 01:47:28 localhost kernel: [ 4372.609556] rk_vcodec: vpu_service_ioctl:1890: error: unknown vpu service ioctl cmd 40086c01
Feb 21 01:47:28 localhost kernel: [ 4372.633423] usb 5-1: 3:8: cannot get freq at ep 0x86
Feb 21 01:47:28 localhost kernel: [ 4372.670748] uvcvideo: Non-zero status (-71) in video completion handler.
Feb 21 01:47:30 localhost kernel: [ 4374.772292] rk_vcodec: vpu_service_ioctl:1831: error: pid 7909 wait 1 task done timeout
Feb 21 01:47:30 localhost kernel: [ 4374.772309] rk-vcodec vpu_combo: 1 task is running but not return, reset hardware...
Feb 21 01:47:30 localhost kernel: [ 4374.772317] rk-vcodec vpu_combo: resetting...
Feb 21 01:47:30 localhost kernel: [ 4374.772380] ------------[ cut here ]------------
Feb 21 01:47:30 localhost kernel: [ 4374.772386] WARNING: at drivers/video/rockchip/vcodec/vcodec_service.c:693
Feb 21 01:47:30 localhost kernel: [ 4374.772391] Modules linked in: af_packet lz4hc lz4hc_compress zlib ftdi_sio rk_vcodec lzo zram ip_tables x_tables autofs4
Feb 21 01:47:30 localhost kernel: [ 4374.772416]
Feb 21 01:47:30 localhost kernel: [ 4374.772426] CPU: 1 PID: 7912 Comm: mpp_enc_ctrl Tainted: G        W       4.4.192-rockchip64 #1
Feb 21 01:47:30 localhost kernel: [ 4374.772431] Hardware name: Firefly ROC-RK3328-CC (DT)
Feb 21 01:47:30 localhost kernel: [ 4374.772437] task: ffffffc0762f0e00 task.stack: ffffffc068294000
Feb 21 01:47:30 localhost kernel: [ 4374.772466] PC is at vpu_reset+0x34/0x230 [rk_vcodec]
Feb 21 01:47:30 localhost kernel: [ 4374.772491] LR is at vpu_reset+0x2c/0x230 [rk_vcodec]
Feb 21 01:47:30 localhost kernel: [ 4374.772497] pc : [<ffffff8000c014f4>] lr : [<ffffff8000c014ec>] pstate: 40000145
Feb 21 01:47:30 localhost kernel: [ 4374.772501] sp : ffffffc068297d20
Feb 21 01:47:30 localhost kernel: [ 4374.772506] x29: ffffffc068297d20 x28: ffffffc0762f0e00
Feb 21 01:47:30 localhost kernel: [ 4374.772513] x27: ffffff8008bc2000 x26: 0000000000000001
Feb 21 01:47:30 localhost kernel: [ 4374.772520] x25: ffffffc065f60148 x24: ffffffc065fc5a88
Feb 21 01:47:30 localhost kernel: [ 4374.772526] x23: ffffffc065e0c428 x22: 0000000000000001
Feb 21 01:47:30 localhost kernel: [ 4374.772533] x21: ffffffc065fc5828 x20: ffffffc065e0c428
Feb 21 01:47:30 localhost kernel: [ 4374.772539] x19: ffffffc065fc5828 x18: 0000000000000005
Feb 21 01:47:30 localhost kernel: [ 4374.772546] x17: 0000007f9fa1fb10 x16: ffffff8008206c1c
Feb 21 01:47:30 localhost kernel: [ 4374.772553] x15: 0000000000000000 x14: 000000000000000a
Feb 21 01:47:30 localhost kernel: [ 4374.772559] x13: ffffff808928ba9b x12: 0000000000000030
Feb 21 01:47:30 localhost kernel: [ 4374.772566] x11: 0000000000000006 x10: ffffff800928baa3
Feb 21 01:47:30 localhost kernel: [ 4374.772572] x9 : 0000000005f5e0ff x8 : 6974746573657220
Feb 21 01:47:30 localhost kernel: [ 4374.772579] x7 : 3a6f626d6f635f75 x6 : ffffff800928bacc
Feb 21 01:47:30 localhost kernel: [ 4374.772586] x5 : ffffffc07ff4eb38 x4 : 0000000000000001
Feb 21 01:47:30 localhost kernel: [ 4374.772592] x3 : 0000000000000007 x2 : 0000000000000007
Feb 21 01:47:30 localhost kernel: [ 4374.772599] x1 : 0000000000000006 x0 : ffffffc0743f0c00
Feb 21 01:47:30 localhost kernel: [ 4374.772606]
Feb 21 01:47:30 localhost kernel: [ 4374.772606] PC: 0xffffff8000c01474:
Feb 21 01:47:30 localhost kernel: [ 4374.772611] 1474  ffffff80 00c07a5b ffffff80 00c07a87 ffffff80 00c07a9b ffffff80 00c07ab8
Feb 21 01:47:30 localhost kernel: [ 4374.772630] 1494  ffffff80 00c07ad3 ffffff80 00c07af5 ffffff80 00c07b11 ffffff80 00c07b58
Feb 21 01:47:30 localhost kernel: [ 4374.772647] 14b4  ffffff80 00c0725f ffffff80 a9bb7bfd 910003fd a90153f3 a9025bf5 aa0003f4
Feb 21 01:47:30 localhost kernel: [ 4374.772663] 14d4  aa1e03e0 d503201f 58000fa1 f9404693 f942ba60 95eb8f4a f941de60 b4000040
Feb 21 01:47:30 localhost kernel: [ 4374.772679] 14f4  d4210000 f941e260 b4000040 d4210000 f941e660 b4000040 d4210000 f942ba60
Feb 21 01:47:30 localhost kernel: [ 4374.772696] 1514  f901de7f f901e27f f901e67f 95e7f272 f9433e60 f9401001 58000d60 eb00003f
Feb 21 01:47:30 localhost kernel: [ 4374.772712] 1534  54000521 910de275 aa1503e0 95fe9f79 9100c3a8 52800002 52800001 52800000
Feb 21 01:47:30 localhost kernel: [ 4374.772729] 1554  95f4d48a aa1503e0 95fe9fa5 f942ba60 95e7f27b 58000bd5 f942ba60 aa1503e1
Feb 21 01:47:30 localhost kernel: [ 4374.772746]
Feb 21 01:47:30 localhost kernel: [ 4374.772746] LR: 0xffffff8000c0146c:
Feb 21 01:47:30 localhost kernel: [ 4374.772752] 146c  ffffff80 00c07a3a ffffff80 00c07a5b ffffff80 00c07a87 ffffff80 00c07a9b
Feb 21 01:47:30 localhost kernel: [ 4374.772768] 148c  ffffff80 00c07ab8 ffffff80 00c07ad3 ffffff80 00c07af5 ffffff80 00c07b11
Feb 21 01:47:30 localhost kernel: [ 4374.772784] 14ac  ffffff80 00c07b58 ffffff80 00c0725f ffffff80 a9bb7bfd 910003fd a90153f3
Feb 21 01:47:30 localhost kernel: [ 4374.772801] 14cc  a9025bf5 aa0003f4 aa1e03e0 d503201f 58000fa1 f9404693 f942ba60 95eb8f4a
Feb 21 01:47:30 localhost kernel: [ 4374.772817] 14ec  f941de60 b4000040 d4210000 f941e260 b4000040 d4210000 f941e660 b4000040
Feb 21 01:47:30 localhost kernel: [ 4374.772834] 150c  d4210000 f942ba60 f901de7f f901e27f f901e67f 95e7f272 f9433e60 f9401001
[... cut ...]
Feb 21 01:47:32 localhost kernel: [ 4376.783352] [<ffffff8008206774>] do_vfs_ioctl+0x9c/0x544
Feb 21 01:47:32 localhost kernel: [ 4376.783359] [<ffffff8008206c84>] SyS_ioctl+0x68/0x98
Feb 21 01:47:32 localhost kernel: [ 4376.783367] [<ffffff8008082f30>] el0_svc_naked+0x24/0x28
Feb 21 01:47:32 localhost kernel: [ 4376.783427] rk-vcodec vpu_combo: reset done
Feb 21 01:47:32 localhost kernel: [ 4376.783474] rk-vcodec vpu_combo: reset done
Feb 21 01:47:32 localhost kernel: [ 4376.802330] uvcvideo: Failed to resubmit video URB (-1).

jbloomfield2008, 2020-02-21:
Sorry one more thing... Your branch https://github.com/HermanChen/mpp/ works just fine!