rockchip-linux/mpp #126: rk3328 android encode h264 issue,  ffplay playback log "no frame"

xfirefly, 2020-02-29:
encode command: 
./mpi_enc_test -t 7 -f 4  -w 480 -h 272 -i ds_480x272.yuv -o f.h264  

play command:
 ffplay.exe -f h264  f.h264

yuv file: 
https://github.com/leixiaohua1020/simplest_ffmpeg_video_encoder/blob/master/ds_480x272.yuv

encode log:
02-29 23:37:53.400  2086  2086 E vpu_api : dlopen vpu lib failed
02-29 23:37:53.400  2086  2086 I mpi_enc_test: cmd parse result:
02-29 23:37:53.400  2086  2086 I mpi_enc_test: input  file name: ds_480x272.yuv
02-29 23:37:53.400  2086  2086 I mpi_enc_test: output file name: f.h264
02-29 23:37:53.400  2086  2086 I mpi_enc_test: width      : 480
02-29 23:37:53.400  2086  2086 I mpi_enc_test: height     : 272
02-29 23:37:53.400  2086  2086 I mpi_enc_test: type       : 7
02-29 23:37:53.400  2086  2086 I mpi_enc_test: debug flag : 0
02-29 23:37:53.401  2086  2086 I mpi_enc_test: mpi_enc_test start
02-29 23:37:53.402  2086  2086 I mpp_ion : vpu_service iommu_enabled 1
02-29 23:37:53.402  2086  2086 I mpp_ion : using ion heap ION_HEAP_TYPE_SYSTEM
02-29 23:37:53.403  2086  2086 I mpi_enc_test: mpi_enc_test encoder test start w 480 h 272 type 7
02-29 23:37:53.403  2086  2086 I mpi     : mpp version: 1d3291b author: Herman Chen [vp8d]: Fix vp8d error on multiple dct partition
02-29 23:37:53.404  2086  2086 I h264e_api: Set rate control: qp 24 [24, 24] bps  7833600
02-29 23:37:53.404  2086  2086 I h264e_api: pic 0 mb 0 skip 0 hrd 0 cpbSize 7833600 gopLen 60
02-29 23:37:53.425  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 0 size 2040
02-29 23:37:53.447  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 1 size 2040
02-29 23:37:53.468  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 2 size 2040
02-29 23:37:53.490  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 3 size 2040
02-29 23:37:53.511  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 4 size 2040
02-29 23:37:53.533  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 5 size 2040
02-29 23:37:53.554  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 6 size 2040
02-29 23:37:53.576  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 7 size 2040
02-29 23:37:53.597  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 8 size 2040
02-29 23:37:53.619  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 9 size 2040
02-29 23:37:53.640  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 10 size 2040
02-29 23:37:53.661  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 11 size 2040
02-29 23:37:53.683  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 12 size 2040
02-29 23:37:53.704  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 13 size 2040
02-29 23:37:53.726  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 14 size 2040
02-29 23:37:53.747  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 15 size 2040
02-29 23:37:53.762  1357  1432 I MulticastReceiver: Could not read datagram: Could not parse headers: java.lang.ArrayIndexOutOfBoundsException: length=1; index=2
02-29 23:37:53.765  1357  1432 I MulticastReceiver: Could not read datagram: Could not parse headers: java.lang.ArrayIndexOutOfBoundsException: length=1; index=2
02-29 23:37:53.769  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 16 size 2040
02-29 23:37:53.790  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 17 size 2040
02-29 23:37:53.812  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 18 size 2040
02-29 23:37:53.834  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 19 size 2040
02-29 23:37:53.855  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 20 size 2040
02-29 23:37:53.877  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 21 size 2040
02-29 23:37:53.898  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 22 size 2040
02-29 23:37:53.920  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 23 size 2040
02-29 23:37:53.941  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 24 size 2040
02-29 23:37:53.963  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 25 size 2040
02-29 23:37:53.984  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 26 size 2040
02-29 23:37:54.006  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 27 size 2040
02-29 23:37:54.027  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 28 size 2040
02-29 23:37:54.049  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 29 size 2040
02-29 23:37:54.070  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 30 size 2040
02-29 23:37:54.091  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 31 size 2040
02-29 23:37:54.113  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 32 size 2040
02-29 23:37:54.134  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 33 size 2040
02-29 23:37:54.156  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 34 size 2040
02-29 23:37:54.177  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 35 size 2040
02-29 23:37:54.199  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 36 size 2040
02-29 23:37:54.220  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 37 size 2040
02-29 23:37:54.241  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 38 size 2040
02-29 23:37:54.263  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 39 size 2040
02-29 23:37:54.284  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 40 size 2040
02-29 23:37:54.306  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 41 size 2040
02-29 23:37:54.327  2086  2086 I mpi_enc_test: mpi_enc_test encoded frame 42 size 2040
[... 49 lines cut ...]
02-29 23:37:55.338  2086  2086 I mpi_enc_test: mpi_enc_test success total frame 90 bps 489600
 

ffplay log:
ffplay version 4.2.1 Copyright (c) 2003-2019 the FFmpeg developers
  built with gcc 9.1.1 (GCC) 20190807
  configuration: --disable-static --enable-shared --enable-gpl --enable-version3 --enable-sdl2 --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libdav1d --enable-libbluray --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --enable-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-libvo-amrwbenc --enable-libmysofa --enable-libspeex --enable-libxvid --enable-libaom --enable-libmfx --enable-amf --enable-ffnvcodec --enable-cuvid --enable-d3d11va --enable-nvenc --enable-nvdec --enable-dxva2 --enable-avisynth --enable-libopenmpt
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
[h264 @ 02b6b5c0] missing picture in access unit with size 183636
[h264 @ 02b6b5c0] no frame!
[h264 @ 02b62000] decoding for stream 0 failed
[h264 @ 02b62000] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size
Consider increasing the value for the 'analyzeduration' and 'probesize' options
Input #0, h264, from 'f.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264, none, 25 tbr, 1200k tbn, 50 tbc
[h264 @ 02b77580] no frame!
    nan M-V:    nan fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0

HermanChen, 2020-03-02:
Can you update the mpp source code and try again?

closed 2020-03-26 by xfirefly

HermanChen, 2020-03-27:
Is the issue fixed?