rockchip-linux/mpp #127: rk3399 h264 encode fails (Debian Stretch)

brainstormi, 2020-03-02:
When trying to execute the following command (mpp master compiled today), it fails with the error below:

`mpi_enc_test -i demo.mkv -o demo.264 -w 1280 -h 720 -f 0 -t 7
`

```
mpi: mpp version: Without VCS, under bleeding
mpp: Assertion cmd < MPP_CMD_END failed at control:545
mpp: command 200007 param 0x7fc6dda4dc ret -1
mpi_enc_test: mpi control set output timeout -1 ret -1
mpi_enc_test: mpi_enc_test failed ret -1
*** Error in `mpi_enc_test': munmap_chunk(): invalid pointer: 0x000000559082d330   ***

```

HermanChen, 2020-03-03:
Mpp encoder can only accept yuv file as its input. it does not support mkv file input.
Mpp is different from ffmpeg.

HermanChen, 2020-03-03:
Please try this cmd:
mpi_enc_test -o demo.264 -w 1280 -h 720 -f 0 -t 7

brainstormi, 2020-03-09:
Same error:

```
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: 
mpi_enc_test: output file name: demo.264
mpi_enc_test: width      : 1280
mpi_enc_test: height     : 720
mpi_enc_test: format     : 0
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpi_enc_test: mpi_enc_test encoder test start w 1280 h 720 type 7
mpi: mpp version: Without VCS, under bleeding
mpp: Assertion cmd < MPP_CMD_END failed at control:545
mpp: command 200007 param 0x7fdb94e54c ret -1
mpi_enc_test: mpi control set output timeout -1 ret -1
mpi_enc_test: mpi_enc_test failed ret -1
*** Error in `mpi_enc_test': munmap_chunk(): invalid pointer: 0x000000558737f330 ***
```

HermanChen, 2020-03-10:
It seems a environment problem.
Could you please run the mpp_platform_test and mpp_runtime_test and show me the log?

brainstormi, 2020-03-12:
**mpp_platform_test** output:

```
mpp_plat_test: chip name: pine64,rockpro64 rockchip,rk3399
mpp_plat_test:
mpp_plat_test: chip vcodec type 00040202
mpp_plat_test: found vpu2 codec
mpp_plat_test: found rkvdec decoder
mpp_plat_test: found vpu2 encoder
mpp_plat_test:
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp_plat_test: avs   decoder: /dev/vpu_service
mpp_plat_test:
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder: /dev/vpu_service
mpp_plat_test: H.265 encoder: /dev/vpu_service
mpp_plat_test: MJPEG encoder: /dev/vpu_service
mpp_plat_test: mpp platform test done
```

**mpp_runtime_test** output:

```
mpp_rt: found ion allocator
mpp_rt: found drm allocator
mpp_rt: found drm allocator in dts
mpp_rt_test: mpp found ion buffer is invalid
mpp_rt_test: mpp found drm buffer is valid
```

HermanChen, 2020-03-12:
mpp: Assertion cmd < MPP_CMD_END failed at control:545
mpp: command 200007 param 0x7fdb94e54c ret -1

This log seems a software error between library and binary. Please check the library which the binary linked to.